openwrt/openwrt #10804: Invalid preferred life times in router advertisements

jorti, 2022-09-24:
Hi,

After upgrading to OpenWrt 22.03, I've noticed a weird thing. My Linux clients are logging these error messages:

~~~
kernel: IPv6: addrconf: prefix option has invalid lifetime
~~~

I have a DHCP configuration like shown below. I'm using a life time of 30min for the DHCP leases and the option `ra_useleasetime` should make the `leasetime` to be also used as limit and preferred lifetime of the IPv6 prefix.

~~~
# uci show dhcp.lab
dhcp.lab=dhcp
dhcp.lab.interface='lab'
dhcp.lab.start='100'
dhcp.lab.limit='150'
dhcp.lab.ra='server'
dhcp.lab.dhcpv6='server'
dhcp.lab.dhcp_option='6,192.168.5.1'
dhcp.lab.ra_flags='managed-config' 'other-config'
dhcp.lab.ra_slaac='0'
dhcp.lab.ra_useleasetime='1'
dhcp.lab.leasetime='30m'
~~~

However, I'm getting invalid preferred lifetimes for the advertised prefixes:

~~~
18:42:08.550221 IP6 (flowlabel 0xe72ec, hlim 255, next-header ICMPv6 (58) payload length: 176) fe80::ea9f:80ff:fe5d:3d6e > fe80::5054:ff:fe9a:e: [icmp6 sum ok] ICMP6, router advertisement, length 176
	hop limit 64, Flags [managed, other stateful], pref medium, router lifetime 1800s, reachable time 0ms, retrans timer 0ms
	  source link-address option (1), length 8 (1): e8:9f:80:5d:3d:6e
	  mtu option (5), length 8 (1):  1492
	  prefix info option (3), length 32 (4): 2a00:abcd:abcd:1405::/64, Flags [onlink], valid time 1800s, pref. time 43200s
	  prefix info option (3), length 32 (4): fddc:f797:78ef:5::/64, Flags [onlink], valid time 1800s, pref. time 43200s
	  route info option (24), length 24 (3):  2a00:abcd:abcd:1400::/56, pref=medium, lifetime=1800s
	  route info option (24), length 24 (3):  fddc:f797:78ef::/48, pref=medium, lifetime=1800s
	  rdnss option (25), length 24 (3):  lifetime 1800s, addr: fddc:f797:78ef:5::1
	  advertisement interval option (7), length 8 (1):  600000ms
~~~

The preferred life time has to be always equal or less than the valid life time and the documented behaviour of `ra_useleasetime` is not honored, so this looks like a bug.

dave14305, 2022-09-24:
Try setting `preferred_lifetime` to match `leasetime`.

jorti, 2022-09-24:
> Try setting `preferred_lifetime` to match `leasetime`.

That worked. Thanks.

However, the documented behaviour of [`ra_useleasetime`](https://openwrt.org/docs/techref/odhcpd) says:

~~~
If set, the configured DHCPv4 leasetime is also used as limit and preferred lifetime of the IPv6 prefix. 
~~~

So either the documentation has to be updated or `ra_useleasetime` should also set the preferred lifetime.

dave14305, 2022-09-24:
There was some discussion of the behavior inconsistencies here in the commit:

https://github.com/openwrt/odhcpd/commit/3bda90079ec5574ef469e2a7804808302f17769d

renne, 2024-05-24:
It seems setting the `preferred_lifetime` to `0m` is ignored in openwrt 23.05 (OpenWrt 23.05.3 (r23809-234f1a2efa):
```
config dhcp 'wan_static'
        option interface 'wan_static'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option ra 'server'
        option dns_service '0'
        option preferred_lifetime '0m'
        option dynamicdhcp '0'
```
```
root@docker:~# ip -6 addr show enp6s18
2: enp6s18: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    inet6 2a03:4000:20:XXXX::3/64 scope global dynamic mngtmpaddr 
       valid_lft forever preferred_lft forever
```
IPv6 uses [Address_scopes](https://en.wikipedia.org/wiki/IPv6_address#Address_scopes) and`preferred_lifetime` to determine the IPv6 prefix/address for new outgoing connections.
A wrong `preferred_lifetime` will cause the selection of the wrong IPv6 prefix/address for outgoing connections.

A `preferred_lifetime` of `0` and a `valid_lifetime` of `infinite` is the special case in which incoming connections are allowed via an IPv6 prefix/address but new outgoing connections never use this IPv6 prefix/address.

whao, 2025-07-16:
@renne Hi, there. I am encountering exact same problem here. I have to IPv6-PD upstreams from two WANs and I only want to use one of them for new outbound connection from LAN. Setting `preferred_lifetime` to `0` does not change the advertisement on LAN. Have you solve this problem?

renne, 2025-07-18:
@TCB13 My case was a combination of a static IPv6-prefix for incoming connections and a dynamic IPv6-prefix for new outgoing connections. The static IPv6-prefix shall never be used for outgoing connections. So “Router Preference” is not reliable enough,

@whao Sorry, no.

TCB13, 2025-07-18:
> [@TCB13](https://github.com/TCB13) My case was a combination of a static IPv6-prefix for incoming connections and a dynamic IPv6-prefix for new outgoing connections. The static IPv6-prefix shall [...]

What behavior do you see with “Router Preference”? I’m curious now.