openwrt/openwrt #10819: External USB Hard Disk initialization takes too long (20 min)

Konstantin-Glukhov, 2022-09-26:
kmod-usb-storage-uas

When an external USB drive is plugged it takes around 20 min to get it online.

dmesg shows these repeating fragments:
```
usb 2-1: reset SuperSpeed Gen 1 USB device number 3 using xhci-hcd
scsi host1: uas_eh_device_reset_handler success
scsi host1: uas_eh_device_reset_handler start
sd 1:0:0:0: tag#8 uas_zap_pending 0 uas-tag 1 inflight:
sd 1:0:0:0: tag#8 CDB: opcode=0xa3, sa=0xc a3 0c 01 41 00 00 00 00
usb 2-1: reset SuperSpeed Gen 1 USB device number 3 using xhci-hcd
```
Finally 20 min later the device is initialized with the following message:
```
scsi host1: uas_eh_device_reset_handler success
sd 1:0:0:0: [sda] Attached SCSI disk
```

If the device is left attached and the router is rebooted, it takes 20 min to boo. Basically the boot continues only after the device is initialized.

Konstantin-Glukhov, 2022-09-29:
I would like to get to the bottom of this issue, but I don't know where to start. Can someone give me some pointers how to debug the slow initialization issue?

brada4, 2022-09-30:
First would be to try without UAS module

Konstantin-Glukhov, 2022-09-30:
Without UAS module the disk is not recognized at all.

brada4, 2022-09-30:
Cannot happen - do you see it in lsusb?

Konstantin-Glukhov, 2022-09-30:
I see it via lsusb, but it does not appear as /dev/sd*

Konstantin-Glukhov, 2022-09-30:
I want USB3 functionality. The USB2 speed is too slow for my needs.

Konstantin-Glukhov, 2022-09-30:
Are there debug flags that can be enabled to see more info in the log?

brada4, 2022-09-30:
Please install usb-storage module and check if the device works at all and try to pass on as accurate info as possible when asked.

Konstantin-Glukhov, 2022-10-01:
The router is **TP-Link AD7200 Talon**.
The hard disk is **LaCie Rugged THB USB3 1TB**
I doubt it is USB power issue because there is no problem with the stock firmware.

My debugging steps:
```
rmmod uas
```
attach HD
```
lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=, Driver=uas, 5000M
```
With UAS module unloaded **lsusb** reports the HD, but it is not initialized as scsi device. **ls /dev/sd*** shows no devices
```
insmod uas
```
After **insmod uas** the following appears in **dmesg**:
```
scsi host1: uas
usbcore: registered new interface driver uas
scsi 1:0:0:0: Direct-Access     LaCie    Rugged THB USB3  051E PQ: 0 ANSI: 6
sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
sd 1:0:0:0: [sdb] 4096-byte physical blocks
sd 1:0:0:0: [sdb] Write Protect is off
sd 1:0:0:0: [sdb] Mode Sense: 43 00 00 00
sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
sd 1:0:0:0: tag#8 uas_eh_abort_handler 0 uas-tag 1 inflight: IN
sd 1:0:0:0: tag#8 CDB: opcode=0xa3, sa=0xc a3 0c 01 12 00 00 00 00 02 00 00 00
scsi host1: uas_eh_device_reset_handler start
usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
scsi host1: uas_eh_device_reset_handler success
scsi host1: uas_eh_device_reset_handler start
sd 1:0:0:0: tag#8 uas_zap_pending 0 uas-tag 1 inflight:
sd 1:0:0:0: tag#8 CDB: opcode=0xa3, sa=0xc a3 0c 01 12 00 00 00 00 02 00 00 00
usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
scsi host1: uas_eh_device_reset_handler success
```
The last 6 lines in the log keep repeating for about 18 min until scsi device device is finally initialized:
```
logread -e "usb 2-1"
Sat Oct  1 15:10:30 2022 kern.info kernel: [ 1955.367634] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Oct  1 15:26:03 2022 kern.info kernel: [ 2888.438835] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
...
Sat Oct  1 15:43:58 2022 kern.info kernel: [ 3963.650195] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
```
Notes:
15:10:30 - HD is physically attached
15:26:03 - insmod uas
15:43:58 - HD is initialized as scsi device (dmesg output: **[ 3963.682766] sd 1:0:0:0: [sdb] Attached SCSI disk**)

brada4, 2022-10-01:
Device reset loop is not exactly normal operation.
2 yes/no tries to attempt -
Try with 2.4GHz wifi off - obviously without losing network connectivity... - it interferes with USB3 if in very close proximity. Like cheap cable not made to spec, like with unspecified power ranking. Sometimes it is crossed on SoC and you cannot do anything.
 Please try with kmod-usb-storage-uas uninstalled.

Konstantin-Glukhov, 2022-10-01:
I already tried all the suggestions. My 2.4GHz is always off, and initially I started without the UAS module installed.

If it was a problem with a physical component (electrical problem, interference, etc.), the HD will not work at all, but it does work, it just takes 20 minutes to come online.

There must be some logic in the device driver that causes so many retries. I tried to search for "**_uas_eh_device_reset_handler start_**" in the source code, but nothing comes up. I am a little bit puzzled about where these error log messages come from.

brada4, 2022-10-01:
Another is USB power issues 1/ powersaving 2/ spinup draw of 5W for spinup moment, where regulator may reset the port when excess is drawn - e.g - can you fast-charge mobile phone on that router port? Do you have a powered USB2 or 3 hub?

I am not an expert further but disable all:
https://wiki.archlinux.org/title/Power_management#USB_autosuspend
udev.d maps to ->
/sys/bus/<subsys>/<devname>/parameters/power/control
If device resets (disconnects/reconnects) you might not be able to catch it by hand.

The idea is that port (USB on PCI device) enters sleep state while there are sleeping USB devices still needing minimal but power at all times. Say ones without moving parts dont care, but spinning up disk spindle may...

Konstantin-Glukhov, 2022-10-02:
TP-Link AD7200 Talon router has only two USB3 ports.
I am afraid ARM version of linux does not implement /etc/udev functionality.

brada4, 2022-10-02:
OpenWRT analog of udev is /etc/hotplug.d , but first thing is to see if powersave is enabled somewhere in USB drive driver chain. Arch linux has single page with everything that could save power...
According to specification USB 3.0 provides 4.5W, LaCie says their drives consume 4W max, what I suspect that some powersave option makes it less for the moment of hard drive spin-up. Under formal specifications it works just fine, but not for you.