openwrt/openwrt #10821: rt4230w_rev6: issues with partitioning

Lanchon, 2022-09-26:
hi @dangowrt @mans0n @lmore377,

this openwrt rt4230w_rev6 fw is also being used by rev10 devices: rev6 have 512MB flash while rev10 have 256MB. keep this in mind.

the stock fw for 512MB devices has these partitions (end of list):
```
...
[    1.960826] 0x000001ea0000-0x000001f00000 : "0:BOOTCONFIG1"
[    1.967041] 0x000001f00000-0x000002400000 : "0:APPSBL"
[    1.982228] 0x000002400000-0x00000a400000 : "rootfs_1"
[    2.231782] 0x00000a400000-0x000012400000 : "rootfs"
[    2.551668] 0x000012400000-0x00001c400000 : "ubifs"
             ( 0x00001c400000-0x000020000000 : plus 60 MiB of unused flash )
```
(stock layout for 256MB is unknown to me.)

openwrt's 512MB layout ends with:
```
...
[    1.287970] 0x000001ea0000-0x000001f00000 : "BOOTCONFIG1"
[    1.289644] 0x000001f00000-0x000002400000 : "APPSBL"
[    1.302174] 0x000002400000-0x00001c400000 : "ubi"
             ( 0x00001c400000-0x000020000000 : plus 60 MiB of unused flash )
```
but when the same openwrt fw is booted on 256MB devices, the layout becomes:
```
...
[    1.287970] 0x000001ea0000-0x000001f00000 : "BOOTCONFIG1"
[    1.289644] 0x000001f00000-0x000002400000 : "APPSBL"
[    1.302174] 0x000002400000-0x000010000000 : "ubi"
```

turns out that openwrt truncates the partition size instead of failing to boot, but UBOOT does not. so to account for this difference, the developer of the port provided different UBOOT settings for the 2 flash sizes.

so the first problem are those 60MB gone unused on 512MB devices. it would have been great if the dev would have specified the "ubi" partition as "from here to the end of the flash device" in both openwrt's kernel and the UBOOT config. this way both configs would have been correct and unified across flash sizes. but he did not, and left this 60MB unused area in the large flash devices.

extending the partition size now seems not doable. the problem is: the first openwrt update to the new layout will work. but subsequent updates may sooner or later write part of the kernel volume in those added 60MB and UBOOT will not be able to find it and the device will brick.

1) so extending the partition requires synchronized changes in openwrt and UBOOT config. i guess this cannot be done safely. is there a way to do this? is it worth the complication?

(i really wished openwrt defaulted to use the partition layout handed over from the bootloader so that everyone is free to partition their devices as they see fit. i know not all BLs can do this, but IMHO openwrt should take advantage of this whenever possible.)

2) if not possible, id like to add an "extra" partition with the extra 60MB so at least people can use them if needed without a custom kernel. i did the same [here](https://github.com/openwrt/openwrt/blob/4ed90e84f87e8284cb98d0ed99de041a8a4cde35/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-tr4400-v2.dts#L301-L304) (and the install instructions use that partition to store and alternate initramfs recovery fw that boots if for any reason the main fw does not). i'd also like to reintroduce the deleted stock partitions as writable at their original mtd numbers with names prefixed by "stock_" for rollback-to-stock purposes (as i did [here](https://github.com/openwrt/openwrt/blob/4ed90e84f87e8284cb98d0ed99de041a8a4cde35/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-tr4400-v2.dts#L252-L271)), as some of these devices are rented. this would change the mtd number of the current "ubi" partition, but this change does not affect openwrt. would you approve of these changes?

3) the new "extra" partition would lie completely outside flash on 256MB devices (both start and end, not just end). what would openwrt do while booting? just skip adding the partition or would it cause some failure?

thanks!!

mans0n, 2022-09-26:
> this openwrt rt4230w_rev6 fw is also being used by rev10 devices: rev6 have 512MB flash while rev10 have 256MB.

Err, why don't we just split them? Just add rev10 support as a separate device with slightly different partition map in dts.

> 1) so extending the partition requires synchronized changes in openwrt and UBOOT config. i guess this cannot be done safely. is there a way to do this? is it worth the complication?

It sounds like it can be accomplished via `DEVICE_COMPAT_VERSION`. Major version bump will prevent users from automatic sysupgrade. It will instead display a warning message and you can describe the required procedure for manual update in `DEVICE_COMPAT_MESSAGE`.

> i'd also like to reintroduce the deleted stock partitions as writable at their original mtd numbers with names prefixed by "stock_" for rollback-to-stock purposes (as i did [...]

Sure, go ahead. Personally I'm not a strong fan of keeping the stock partition map (instead I tend to create a stock image that fits the openwrt partition map, right before a revert) but I agree that it has its own benefits.

> 3) the new "extra" partition would lie completely outside flash on 256MB devices (both start and end, not just end). what would openwrt do while booting? just skip adding the partition or would it cause some failure?

Try it yourself and let me know the result. I guess the kernel will just ignore the invalid partition node.

Lanchon, 2022-09-26:
@mans0n

hi again, thanks for your quick reply!

> why don't we just split them?

as i understand it, they are really the same except for the flash. IMHO making separate builds just wastes time, space and energy if it is not needed, plus we get more code to maintain. (besides all my devices are rev6 and i can't test on real rev10.)

i can (and will) amend the wiki clarifying this and updating install instructions because right now the instructions brick devices if they are rev10 (256MB) with RAC stock firmware. (instructions for rev10 SAC fw devices are fine.)

> ... `DEVICE_COMPAT_VERSION`

interesting, so it can be done but it becomes a hassle for users...

right now "ubi" is 416 MiB on affected devices. i think the benefit of additional 60 MiB is not enough to bother users with this. i could slap an "extra" partition, but then we REALLY shouldn't grow "ubi" later. or i could leave it as is... but then nobody will really grow ubi ever, i think. what do you favor?

> I tend to create a stock image that fits the openwrt partition map

i'm not aware of an easy way to do this. is there? it's easy enough on NOR, but NAND can have bad blocks. for all i know, the driver stack will let you start writing a MTD partition on an offset, but the offset would be logical (not counting bad blocks) and we need a physical offset to hit the start of a nested partition.

> [partition lying completely outside flash] Try it yourself, report

sure, i will if i attempt any changes to the device at all.

Lanchon, 2022-09-27:
i found a stock 256MB partitioning scheme online: https://github.com/coolsnowwolf/lede/issues/7582#issuecomment-890669560

same 3 partitions but smaller. (linux mtd20 is still "rootfs_1" although the BL is listing them in a different order.)

Lanchon, 2022-09-27:
@mans0n 

i am setting up a guide to back up stock and install openwrt via ssh without TFTP and with a recovery initramfs image.

UBOOT config is unified for flash sizes and looks like this:
```
echo "setting up u-boot environment"
fw_setenv -s - <<"EOF"

boot_custom test -n "$boot_pre" && run boot_pre; run boot_main_512M; run boot_recovery_512M; run boot_main_256M; run boot_recovery_256M; run boot_tftp

boot_main_512M set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:416M@0x2400000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 kernel && bootm
boot_main_256M set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:220M@0x2400000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 kernel && bootm

boot_recovery_512M set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:416M@0x2400000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 recovery && bootm
boot_recovery_256M set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:220M@0x2400000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 recovery && bootm

boot_tftp test -n "$ipaddr" || set ipaddr 192.168.1.1; test -n "$serverip" || set serverip 192.168.1.2; tftpboot recovery.bin && bootm

bootcmd run boot_custom
bootdelay 2

EOF
```

the i realized official instructions all boot the kernel like this:
```
ubi read 0x44000000 kernel 0x6e0000 && bootm
```
why is kernel size limited to 6.875 MiB (0x6e0000)? is there something i am missing?

right now the official kernel is about 3 MB, so not a problem for official openwrt users.

i need to update the wiki's install instructions as they brick RAC fw rev10 devices (happened to a friend). i don't know what to do about this kernel size limit.

Lanchon, 2022-09-30:
@dangowrt Daniel maybe you can help with my question in the previous message?

lmore377, 2022-09-30:
I just noticed this issue so I thought I'd give some input.

> this openwrt rt4230w_rev6 fw is also being used by rev10 devices: rev6 have 512MB flash while rev10 have 256MB. keep this in mind.

This is already known (under the 'Different revisions' section on the device page)  

> so the first problem are those 60MB gone unused on 512MB devices.

I just went with that because that's where the stock fw has the flash end and I assumed there was some reason for that. 

> 2\. as some of these devices are rented.

Imho it's a bad idea to install this on rented devices and I even advise against it in the device page. Even if/when someone comes up with a way to restore the router back to stock there's still a chance of stuff going wrong and I just don't want people to end up in a situation where they have to explain to spectrum that their router somehow magically bricked itself and they need a replacement.

> linux mtd20 is still "rootfs_1" although the BL is listing them in a different order

The reason for the differently ordered partitions is that this router has 2 different boot slots (kinda similar to [Android's A/B partition layout](https://source.android.com/docs/core/ota/ab#slots)) so rootfs and rootfs_1 may be different depending on what slot is currently booted. As far as I know it's the same behavior across all rt4230w devices

> why is kernel size limited to 6.875 MiB (0x6e0000)? is there something i am missing?

Honestly there's no specific reason this is just what happened to work when trying to get openwrt booted

> i need to update the wiki's install instructions as they brick RAC fw rev10 devices

I'll take care of this, this is honestly my bad. When I first wrote the instructions I just assumed all RAC devices were rev6 and all SAC were rev10 and later on in [the forum thread](https://forum.openwrt.org/t/askey-rac2v1k-rt4230w-rev6-support/15830) we eventually figured out that isn't the case and either model can be either revision but I never updated the instructions. I never had a rev10 device that wasn't rented so I only ever went as far as booting from ram and I just relied on others to test if openwrt could be installed on them.

Just an extra note, openwrt overrides the stock rootfs, rootfs_1 and ubifs partitions and merges them to make one big ubifs partition that the kernel is stored inside of. The two rootfs partitions together take up ~200mb of space only leaving ~168mb of free space for the user (even less on rev10) so by merging them all in the end there's around 354mb of usable free space (I think ~150mb on rev10).

lmore377, 2022-09-30:
I completely forgot about this issue you opened at my repo lol. I'm going to close that one since not much discussion happened over there and it might be better to move it here especially since the router is part of the release builds now.

https://github.com/lmore377/openwrt-rt4230w/issues/5

Lanchon, 2022-10-01:
hi @lmore377 , long time

> it's a bad idea to install this on rented devices [...] if/when someone comes up with a way to restore the router back to stock there's still a chance of stuff going wrong and I just don't want people to end up

i've done that already. i've got an alternate install procedure that i'll publish that is easier and faster: it doesn't require setting up a TFTP server, you just paste some lines via SSH. it produces full backups of your device, installs a "recovery" volume with an initramfs image that boots if the "main" kernel is corrupt somehow, and includes a unified UBOOT configuration that works with 512 and 256 MB devices and doesn't require the user to check the memory size and possibly get it wrong. this install procedure will be linked as optional from the wiki, but its UBOOT configuration is the same i want to update the standard wiki with.

the UBOOT config is pasted on this thread earlier. if you want to look at similar backup/install scripts, peek at the official [TR4400 v2](https://openwrt.org/inbox/toh/arris/tr4400_v2) install procedure.

i agree that people shouldn't hack rented devices, but still no reason not to provide backup and back-to-stock procedures.

> > why is kernel size limited to 6.875 MiB

> Honestly there's no specific reason

cool, it's good to know that there is no hidden reason, thanks!

> > i need to update the wiki's install instructions as they brick RAC fw rev10 devices

> I'll take care of this, this is honestly my bad.

it's not, these devices hadn't been spotted yet and maybe didn't even exist when you wrote that procedure.

let me do the wiki update. i was about to update the instructions with unified (512/256) UBOOT (similar to what's pasted here but simpler), but first i needed confirmation that eliminating the kernel size limit would not have unintended consequences.

> Just an extra note

yes i know. the TR4400 v2 (same soc) is much worse: there are 3 reusable non-contiguous areas separated with bootloader stuff and unique device config (mac addresses, calibration, etc). the install script triple checks everything and relocates the non-BL stuff to make 70% of the space contiguous for a big ubi, and provides an 'extra' partition with the other 30%. it uses extra to install 'recovery' and also a dumps a backup of relocated stuff there for good measure.

the kernel supports mtd concat that can join several non-contiguous partitions into a big mtd device, which can then be used as a ubi partition. UBOOT also, but it has to be a recent UBOOT, and that's what forced me to relocate stuff instead.

it would be great if the ubi kernel guys could add a per-volume setting that forced some volumes to reside completely in the first N blocks of the ubi partition. we could use this for the kernel volume so that UBOOT finds it while we join all available space into a big ubi. this would also solve the issue with this device: the 60 MB unused space at the end of 512 MB devices. we'd just ask the ubi layer to keep the kernel volume away from the last 60 MB.

but reading on kernel.org i saw that the guys there are not keen on making ubi's on-flash layout more complex or supporting openwrt: this is why ubi is forked on the openwrt kernel with some hacks active while attaching a partition. so i didn't even bother bringing this up with them, they would hate it lol.

Lanchon, 2022-10-01:
(come to think about it, the solution is not hacking the kernel and extending the ubi layout. the solution is to add a 2nd stage open-sourced UBOOT with its separate config, and then we could do anything we want. but who would want to go through that trouble!)

Lanchon, 2022-12-06:
well... life got in my way

FYI, i finally updated the wiki. i included unified uboot configs that autodetect the flash size, so people are less likely to make mistakes and brick. also added a simpler install method that only requires SSH and forgoes TFTP. added backup scripts, a recovery OS (initramfs), and fallback to TFTP if all fails.

https://openwrt.org/inbox/toh/askey/rt4230w_rev6
https://github.com/Lanchon/openwrt-rt4230w-rev6

cheers!