openwrt/openwrt #10822: Imagebuilder packages no longer contain profiles

eduperez, 2022-09-26:
Starting with 21.02.03 (and still happening at 22.03.0), the `imagebuilder` package can no longer build images (at least on some platforms):

```
[eperez@delfin openwrt-imagebuilder-22.03.0-mvebu-cortexa9.Linux-x86_64]$ make info
Current Target: "mvebu/cortexa9"
Current Architecture: "arm"
Current Revision: "r19523-bfd070e7fa"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-wolfssl logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp uboot-envtools kmod-gpio-button-hotplug dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

```

For some reason, the `profiles` folder seems to be empty for most platforms now:

```
[eperez@delfin openwrt-imagebuilder-22.03c5-mvebu-cortexa9.Linux-x86_64]$ ls -l target/linux/mvebu/cortexa9/
total 8
drwxr-xr-x. 1 eperez eperez  20 Jul  6 21:09 base-files
-rw-r--r--. 1 eperez eperez 111 Jul  6 21:09 config-5.10
-rw-r--r--. 1 eperez eperez 221 Jul  6 21:09 target.mk
```

Is this intentional? Has the Image Builder procedure changed recently?

dave14305, 2022-09-26:
It seems it would likely be an issue with perl on your build host. I can see the profiles on a Debian Bullseye 11.5 host.
```
$ make info
Current Target: "mvebu/cortexa9"
Current Architecture: "arm"
Current Revision: "r19685-512e76967f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-wolfssl logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp uboot-envtools kmod-gpio-button-hotplug dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

buffalo_ls421de:
    Buffalo LinkStation LS421DE
    Packages: kmod-rtc-rs5c372a kmod-hwmon-gpiofan kmod-hwmon-drivetemp kmod-usb3 kmod-linkstation-poweroff kmod-md-raid0 kmod-md-raid1 kmod-md-mod kmod-fs-xfs mkf2fs e2fsprogs partx-utils
    hasImageMetadata: 1
    SupportedDevices: buffalo,ls421de
ctera_c200-v2:
    Ctera C200 V2
    Packages: kmod-gpio-button-hotplug kmod-hwmon-drivetemp kmod-hwmon-nct7802 kmod-rtc-s35390a kmod-usb3 kmod-usb-ledtrig-usbport
    hasImageMetadata: 1
    SupportedDevices: ctera,c200-v2
cznic_turris-omnia:
    CZ.NIC Turris Omnia
    Packages: mkf2fs e2fsprogs kmod-fs-vfat kmod-nls-cp437 kmod-nls-iso8859-1 wpad-basic-wolfssl kmod-ath9k kmod-ath10k-ct ath10k-firmware-qca988x-ct partx-utils kmod-i2c-mux-pca954x kmod-leds-turris-omnia iwinfo
    hasImageMetadata: 1
    SupportedDevices: cznic,turris-omnia armada-385-turris-omnia
globalscale_mirabox:
    Globalscale Mirabox
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: globalscale,mirabox mirabox
iptime_nas1dual:
    ipTIME NAS1dual
    Packages: kmod-hwmon-drivetemp kmod-hwmon-gpiofan kmod-usb3
    hasImageMetadata: 1
    SupportedDevices: iptime,nas1dual
kobol_helios4:
    Kobol Helios4
    Packages: mkf2fs e2fsprogs partx-utils
    hasImageMetadata: 1
    SupportedDevices: kobol,helios4
linksys_wrt1200ac:
    Linksys WRT1200AC
    Packages: kmod-mwlwifi wpad-basic-wolfssl mwlwifi-firmware-88w8864 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt1200ac armada-385-linksys-caiman linksys,caiman
linksys_wrt1900acs:
    Linksys WRT1900ACS v1
    Packages: kmod-mwlwifi wpad-basic-wolfssl mwlwifi-firmware-88w8864 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt1900acs armada-385-linksys-shelby linksys,shelby
linksys_wrt1900ac-v1:
    Linksys WRT1900AC v1
    Packages: kmod-mwlwifi wpad-basic-wolfssl mwlwifi-firmware-88w8864 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt1900ac-v1 armada-xp-linksys-mamba linksys,mamba
linksys_wrt1900ac-v2:
    Linksys WRT1900AC v2
    Packages: kmod-mwlwifi wpad-basic-wolfssl mwlwifi-firmware-88w8864 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt1900ac-v2 armada-385-linksys-cobra linksys,cobra
linksys_wrt3200acm:
    Linksys WRT3200ACM
    Packages: kmod-mwlwifi wpad-basic-wolfssl kmod-btmrvl kmod-mwifiex-sdio mwlwifi-firmware-88w8964 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt3200acm armada-385-linksys-rango linksys,rango
linksys_wrt32x:
    Linksys WRT32X
    Packages: kmod-mwlwifi wpad-basic-wolfssl kmod-btmrvl kmod-mwifiex-sdio mwlwifi-firmware-88w8964 iwinfo
    hasImageMetadata: 1
    SupportedDevices: linksys,wrt32x armada-385-linksys-venom linksys,venom
marvell_a370-db:
    Marvell Armada 370 Development Board (DB-88F6710-BP-DDR3)
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: marvell,a370-db armada-370-db
marvell_a370-rd:
    Marvell Armada 370 RD (RD-88F6710-A1)
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: marvell,a370-rd armada-370-rd
marvell_a385-db-ap:
    Marvell Armada 385 Development Board AP (DB-88F6820-AP)
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: marvell,a385-db-ap armada-385-db-ap
marvell_a388-rd:
    Marvell Armada 388 RD (RD-88F6820-AP)
    Packages: 
    hasImageMetadata: 0
    SupportedDevices: armada-388-rd marvell,a385-rd
marvell_axp-db:
    Marvell Armada XP Development Board (DB-78460-BP)
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: marvell,axp-db armada-xp-db
marvell_axp-gp:
    Marvell Armada Armada XP GP (DB-MV784MP-GP)
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: marvell,axp-gp armada-xp-gp
plathome_openblocks-ax3-4:
    Plat'Home OpenBlocks AX3 4 ports
    Packages: 
    hasImageMetadata: 1
    SupportedDevices: plathome,openblocks-ax3-4 openblocks-ax3-4
solidrun_clearfog-base-a1:
    SolidRun ClearFog Base
    Packages: mkf2fs e2fsprogs partx-utils
    hasImageMetadata: 1
    SupportedDevices: solidrun,clearfog-base-a1 armada-388-clearfog-base
solidrun_clearfog-pro-a1:
    SolidRun ClearFog Pro
    Packages: mkf2fs e2fsprogs partx-utils
    hasImageMetadata: 1
    SupportedDevices: solidrun,clearfog-pro-a1 armada-388-clearfog armada-388-clearfog-pro
```
```
$ sudo apt list --installed perl*
Listing... Done
perl-base/stable,now 5.32.1-4+deb11u2 amd64 [installed]
perl-modules-5.32/stable,stable,now 5.32.1-4+deb11u2 all [installed,automatic]
perl-openssl-defaults/stable,now 5 amd64 [installed,automatic]
perl-tk/stable,now 1:804.035-0.1+b1 amd64 [installed,automatic]
perl/stable,now 5.32.1-4+deb11u2 amd64 [installed]
```

eduperez, 2022-09-28:
The imagebuilder directory did contain a "`.profiles.mk`" file, but it was empty.

I un-packaged the imagebuilder again, and noticed that it does not contain any "`.profiles.mk`" file at all. Then run "`make info`" again, and noticed it fails and produces a meaningful error message, but only the first time it is run:
```
[eperez@delfin openwrt-imagebuilder-22.03.0-mvebu-cortexa9.Linux-x86_64]$ make info
Can't locate FindBin.pm in @INC (you may need to install the FindBin module) (@INC contains: /usr/local/lib64/perl5/5.34 /usr/local/share/perl5/5.34 /usr/lib64/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib64/perl5 /usr/share/perl5) at /home/eperez/Downloads/OpenWrt/openwrt-imagebuilder-22.03.0-mvebu-cortexa9.Linux-x86_64/scripts/target-metadata.pl line 2.
BEGIN failed--compilation aborted at /home/eperez/Downloads/OpenWrt/openwrt-imagebuilder-22.03.0-mvebu-cortexa9.Linux-x86_64/scripts/target-metadata.pl line 2.
Current Target: "mvebu/cortexa9"
Current Architecture: "arm"
Current Revision: "r19685-512e76967f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-wolfssl logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp uboot-envtools kmod-gpio-button-hotplug dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

[eperez@delfin openwrt-imagebuilder-22.03.0-mvebu-cortexa9.Linux-x86_64]$ make info
Current Target: "mvebu/cortexa9"
Current Architecture: "arm"
Current Revision: "r19685-512e76967f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-wolfssl logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp uboot-envtools kmod-gpio-button-hotplug dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

```

After installing the missing "`perl-FindBinary`" package, it now works as expected.

closed 2022-09-28 by eduperez