openwrt/openwrt #10828: BT Home Hub 5A (HH5a) - 22.03.0 - performance poor since upgrade

whiskerp, 2022-09-27:
I upgraded by BT HomeHub 5A (Lantiq xrx200) from 21.02 to 22.03, carefully recording and reproducing all settings. It works OK but the performance has been hammered. Upload WLAN (iperf3) is as expected at 160mbps but download is below 30mbps (previously >120mbps). Routing to broadband (80mbps PPPoE) used to give an iperf3 of > 76mbps but maxes out at below 60mbps now. The VDSL link comes up at 79.999 as before so it is the router that has lost performance. I have tried with software-offload (the previous setting) and without but it is not possible to get anywhere near the performance with 22.03 so, sadly, I have reverted this router to 21.02 for now. SamKnows graph shows what I mean!
[image]

(also reported on forum : https://forum.openwrt.org/t/openwrt-22-03-0-first-stable-release/136252/80 )

abajk, 2022-09-27:
@whiskerp Maybe Software Flow Offload is not working for PPPoE. Do you see a difference in performance between SFO enabled and disabled?

Most of the time I was testing IPoE. In this scenario on 22.03, NAT performance is 5x better than on 21.02.

whiskerp, 2022-10-23:
Hi

I've checked this again against 22.03.2 and it's still much slower. CPU
still has ~45% free but speedtest is 56mbps versus 73mbps on 21.02.5.

The SFO switch (which was on) seems to do little and enabling packet
steering slows things down.

Peter

whiskerp, 2022-10-24:
Linking to #10224

closed 2022-10-24 by whiskerp