openwrt/openwrt #10830: ath9k 5GHz tx power limit too low for some countries

tim-seoss, 2022-09-27:
I have a couple of ath9k devices:

Netgear WNDAP660
TP-Link TL-WDR4900

Both seem to limit their 5GHz txpower to lower values than I would expect.

I will focus on the WNDAP660 since this runs the latest stable firmware.  The UK regulatory DB limits tx power like this (output of `iw reg get` edited to include 5GHz band only):
```
country GB: DFS-ETSI
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
```

With OpenWRT 22.03.0 if configured to use channel 60 with a 40 Mhz bandwidth (5290 to 5330 MHz), with country set to GB, the firmware limits the txpower to 15dBm .  I'd expect a limit of 20dBm to be applied.

```
root@wifi_660_1:~# iwinfo phy1 info ; iwinfo phy1 txpower
phy1      ESSID: unknown
          Access Point: 84:1B:5E:xx:xx:xx
          Mode: Master  Channel: 60 (5.300 GHz)
          Center Channel 1: 62 2: unknown
          Tx-Power: 15 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11abgn
          Hardware: 168C:0030 168C:3112 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
* 15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
```

It is possible to lower the transmit power with e.g. `iw phy phy1 set txpower limit 1200` but if a limit above 1500 is set, then the reported tx power will not increase beyond 15 dBm.

I changed the configured country code to JP,  and kept the same 5290 to 5330 MHz channel configuration.  This resulted in txpower being set to 20 dBm as I'd expect - the JP rules covering this band being:

    (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW

vs. the similar GB rule:

    (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW

Looking at the driver source, I think this behaviour is due to the driver applying the lowest intersection of ETSI rules and GB rules for txpower, when the country code GB is configured.

I don't think this behaviour is correct, but I may be missing something.

CodeFetch, 2022-09-28:
Please take a dump of the ART partition: https://github.com/CodeFetch/art-collection
Considering the possibility of it including the antenna gain, 15 dBm might be correct (if it has a 5 dBi antenna).

tim-seoss, 2022-09-28:
Thanks for the reply @CodeFetch !  The radios for the WNDAP660 are on internal mini-pcie cards, so I assume the info will be on EEPROM on the cards themselves, but I'll check both.

As a quick check, I've also tried to configure the WNDAP660 to use channel 100 (40MHz so 5490 - 5530 MHz).  The regdb gives this limit as 26dBm, but the card still reports that it using a txpower of 15dBm (and once again can't be raised above this).

```
root@wifi_660_1:~# iwinfo phy1 info ; iwinfo phy1 txpower
phy1      ESSID: unknown
          Access Point: 84:1B:5E:5D:B9:B2
          Mode: Master  Channel: 100 (5.500 GHz)
          Center Channel 1: 102 2: unknown
          Tx-Power: 15 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11abgn
          Hardware: 168C:0030 168C:3112 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
* 15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
```

... this would suggest the limit is not being imposed because a fixed antenna gain is being subtracted from the regdb values?

tim-seoss, 2022-09-28:
The TP-Link WNDR-4900 v1 has:

```
dev:    size   erasesize  name
mtd0: 00050000 00010000 "u-boot"
mtd1: 00010000 00010000 "dtb"
mtd2: 00f80000 00010000 "firmware"
mtd3: 0026e80c 00010000 "kernel"
mtd4: 00d117f4 00010000 "rootfs"
mtd5: 00990000 00010000 "rootfs_data"
mtd6: 00010000 00010000 "config"
mtd7: 00010000 00010000 "caldata"
```
and I've added caldata to a fork here: https://github.com/tim-seoss/art-collection/tree/tl-wdr4900  (I'll open the router up to get board rev details when I can and then rename the files and create a PR).

The Netgear WNDAP660 has:

```
dev:    size   erasesize  name
mtd0: 00100000 00004000 "u-boot"
mtd1: 00010000 00004000 "u-boot-env"
mtd2: 01ac0000 00004000 "firmware"
mtd3: 005f0000 00004000 "kernel"
mtd4: 014d0000 00004000 "ubi"
mtd5: 00010000 00004000 "device-tree"
mtd6: 00400000 00004000 "var"
mtd7: 00020000 00004000 "manudata"
```

`manudata` seems to only include serial number and ethernet MAC.  I 've taken copies of `/sys/kernel/debug/ieee80211/phy*/ath9k/eeprom`, but `tools/ar9300_eeprom` doesn't support the PCIe eeprom dumps.  I can use https://github.com/rsa9000/atheepmgr on it instead if you like (otherwise I can see how easy or otherwise it might be to adapt `ar9300_eeprom` )?

tim-seoss, 2022-10-03:
I've opened a PR for the TP-Link WDR4900 rev 1.3.  Any thoughts on how to proceed with the Netgear WNDAP660 ?

CodeFetch, 2022-10-05:
@tim-seoss As in other TP-Link ART dumps, they haven't set the antenna gain field, but put the antenna gain into the "Tx gain" field (3 dBi on 5GHz, 0 on 2.4) which likely results in a maximum TX power of 17 dBm on 5 GHz. Is that correct?

@s-2 You've recently fiddled with ART on external EEPROM, haven't you?

tim-seoss, 2022-10-06:
@CodeFetch On the WDR4900, I'm seeing:
18 dBm on 2.4 GHz
16 dBm  on 5 GHz

These can be lowered, but not raised.

local regulatory limits according to the regulatory db are:
20 dBm on 2.4GHz (all channels)
23 dBm 5150 - 5250 MHz (channels 36 to 48)
20 dBm 5250 - 5350 MHz (channels 52 to 64)
26 dBm 5470 - 5730 MHz (channels 100 to 144)
23 dBm 5725 - 5850 MHz (channels 149 to 165)

... given what you've said, I'd expect that it would allow 17 dBm on channels 52 to 64, and 23 dBm on channels 100 to 144, but all 5GHz channels get limited to 16 dBm.  This is why I was assuming the limit was coming from the ETSI "Regulatory 5GHz" table entries, which are set to "22" everywhere, so might be capping all txpowers to the same value, regardless of the regulatory db entries?

s-2, 2022-10-06:
The ETSI regulations specify the total EIRP limit of the device, which may use multiple streams (i.e. multiple antennas transmitting at the same time), although in this case both devices seem to use 3-stream radios on 5GHz, so it should not make any difference here even if the values specified there probably refer to a single transmit chain only. Could you compare the transmit power declared in the beacon frame (should be visible in the output of `iw scan`) with OEM firmware or the device running OpenWRT?

@CodeFetch Regarding EEPROM dumping, I've only used the Zyxel NWA1123 devices with 5GHz radio on a dedicated miniPCIE card, which was an issue when setting the mac address (patching mac address into caldata did not work, since no caldata is loaded from the host), but I hadn't further checked which means of eeprom dumping would be available.

s-2, 2022-10-06:
It seems the Information Element 32 (power constraint) is part of 802.11h, I'm actually not sure if this is supported in ath9k by default, will have to check that. Maybe only ath10k radios support it due to TPC requirements on the outdoor / weather radar bands. :frowning_face:

CodeFetch, 2022-10-06:
@tim-seoss The MKK (Japan) regulatory Tx power limit is 30 dBm while ETSI and FCC are set to much lower values. Can you please try to set the country to Japan?

tim-seoss, 2022-10-06:
On channel 100 (JP reg db  23 dBm), that results in txpower being set to 21 dBm.

On channel 100 (GB reg db  26 dBm), that results in txpower being set to 15 dBm.

CodeFetch, 2022-10-06:
> On channel 100 (JP reg db 23 dBm), that results in txpower being set to 21 dBm.
> 
> On channel 100 (GB reg db 26 dBm), that results in txpower being set to 15 dBm.

So it looks like the vendor has just limited the power for FCC and ETSI regions, but I don't know the actual calculations for checking if the limits are correct. Some time ago I've talked to a maintainer of the regdom stuff and he honestly said he didn't know for sure either.

tim-seoss, 2022-10-07:
Thanks.  Looking at the FCC limits from the ROM dump, they seem to correspond reasonably with the product's FCC declaration of conformity as described here: https://fcc.report/FCC-ID/PY318300421/4018303.pdf on page 3.

It looks like the FCC values also have some additional tx power limitations limitations applied at the edges of the band, presumably to keep interference on neighbouring bands below the regulatory limits (e.g. for HT40 FCC,  5755 MHz 19.5 dBm  but at the band edge: 5795 Mhz 16.0 dBm).

I've confirmed that when the region is set to GB (ETSI), and the channel is set to 36, the txpower for this device increases to 19 dBm, and selecting channel 48 gives 20 dBm.   It looks like the ETSI table entries (see dump output above) are being used in the max txpower calculation, since the device allows txpower levels which are 4 or 5 dB higher for channels 36 to 48 inclusive, than it does for any of the other channels (52 to 165)..

I suspect that the reason the ETSI table values apply a cap of 10 dBm for everything above 5260 MHz (channel 52+),  is to apply the lowest minimum of any ETSI jurisdiction for all ETSI countries (this would be inline with the recommendation made in 2013 (2 yrs before this product was released) in https://ec.europa.eu/docsroom/documents/9922/attachments/1/translations/en/renditions/native - "In a general way it should be ensured that changes of “Region of use” cannot alter the compliance against the essential requirements.").

Given the power reductions in these regional tables near the band edges (e.g. relatively lower ETSI values for ch 36, and FCC values for ch 36 and 165) it looks to me like these regional tables are being used to both enforce limits to ensure regulatory compliance for cross-band interference, and also to limit in-band tx power.

In the case of the WNDAP660, the ETSI txpower table values which are applied limit the in-band txpower significantly lower than the national regulations for some regions (e.g. 9 dB or so for GB on some 5GHz channels).

The perfect solution would be to put this device through testing, to determine correct co-channel and co-band interference limits, and then to use the determined maximums instead of the firmware table values to set interference limits only, (and rely on the reg db values for the in-channel limits).  This is probably not practical.

Maybe a less perfect solution exists which would also improve the current state of OpenWRT on these ath9k devices (given that a 9 dB power restriction is significant!)

e.g. My gut feeling is that the mid-band values (only) for the ETSI table could be raised, thereby forcing the national regulatory database values to be used instead (i.e. leaving the limits for the extreme ends of the band as the currently stand). I think this would carry a low risk of non-compliance, but to be certain, expensive test equipment would be required which I don't currently have access to.

CodeFetch, 2022-10-08:
> I think this would carry a low risk of non-compliance

Unfortunately OpenWrt policy in similar cases is setting it no higher than the stock firmware.

tim-seoss, 2022-10-08:
Yes, that makes sense, and is likely the best thing to do in the absence of additional certification.  So the other thing to do in this case would be to try and get tx power values from the last available manufacturer firmware, in case the manufacturer firmware is applying higher values which it stores outside of this ROM, or is otherwise behaving differently?