openwrt/openwrt #10842: v22.03.0 rc5+, iOS Devices no IP unless WMM Disabled

arrmo, 2022-09-29:
Hi,

Some more details [here](https://forum.openwrt.org/t/ios-dhcp-failing-v22-03-0-rc6/133931), but it comes down to - iOS devices (happening with my iPhone 13, iPad Pro ... both on the latest OS) => no IP assigned as of rc5 (it worked in rc4). I see other posts about this as well, noting that WMM has to be disable, but many / most of them note older devices. That's not the case here ... these are current generation iOS / iPadOS devices, but they are not assigned an IP address unless WMM is disabled.

Is there a fix for this?

Thanks!

chendajun000104, 2022-10-02:
I have the same problem🤣

ruildias, 2022-10-14:
Me as well!

chendajun000104, 2022-10-26:
Does this problem sloved in 22.03.2 ?
I have no time to test it now.

ruildias, 2022-10-27:
Yes, with my devices, I no longer have the issue! Thus I assume it’s solved.

chendajun000104, 2022-10-28:
> Yes, with my devices, I no longer have the issue! Thus I assume it’s solved.
> […](#)

ok, thank you, brother

arrmo, 2022-10-30:
Yep, confirmed here as well. Awesome!

closed 2022-10-30 by arrmo