openwrt/openwrt #10843: Time is not synced on (re)boot automatically after a sysupgrade to 22.03.0

birdie-github, 2022-09-29:
I don't want to file a bug report under packages because I've not been able to find the package for `ntpd`. I'm running `OpenWrt 22.03.0 r19685-512e76967f` . My device is `TP-Link TL-WDR3600 v1`. After a sysupgrade the date was set to September 3rd, 2022. Today it's the 29th of September.

Of course `opkg update` failed with `wget returned 5`. 

I couldn't find a command to sync the time in a SSH session, so I simply ran `/etc/init.d/sysntpd restart` which fixed the issue. I waited at the very least 10 minutes for the time to sync automatically. Didn't happen.

Please make sure after a sysupgrade time is indeed getting synced.

There are zero `ntp` related messages in the System Log.

Hurricos, 2022-11-22:
I've had the same issue for a few months now. This seems to have been a post-21 regression of *some* sort. It's definitely a dependency management issue, where some network access dependency is missing from `/etc/init.d/sysntpd`; I adjusted `/etc/init.d/sysntpd`:

```diff
 start_service() {                                   
         . /lib/functions/network.sh                 
+        sleeptime=20                                 
+        sleep $sleeptime                                   
+        logger "slept $sleeptime"                         
         validate_ntp_section ntp start_ntpd_instance      
 }
```

... and rebooted, getting:

```
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:44:27 UTC 2022
Tue Nov 22 12:46:35 PM EST 2022
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:44:27 2022 user.notice root: slept 20
Tue Nov 22 17:44:28 UTC 2022
Tue Nov 22 12:46:36 PM EST 2022
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:44:27 2022 user.notice root: slept 20
Tue Nov 22 17:44:28 UTC 2022
Tue Nov 22 12:46:36 PM EST 2022
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:44:27 2022 user.notice root: slept 20
Tue Nov 22 17:46:38 UTC 2022
Tue Nov 22 12:46:38 PM EST 2022
```

... but when I set `sleeptime=1`:

```
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:48:41 2022 user.notice root: slept 1
Tue Nov 22 17:48:48 2022 user.notice root: slept 1
Tue Nov 22 17:48:50 UTC 2022
Tue Nov 22 12:49:31 PM EST 2022
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:48:41 2022 user.notice root: slept 1
Tue Nov 22 17:48:48 2022 user.notice root: slept 1
Tue Nov 22 17:48:56 UTC 2022
Tue Nov 22 12:49:37 PM EST 2022
[mkennedy@imp ~]$ ssh root@ap3825i0.lan "logread | grep slept; date;"; date
Tue Nov 22 17:48:41 2022 user.notice root: slept 1
Tue Nov 22 17:48:48 2022 user.notice root: slept 1
Tue Nov 22 17:49:02 UTC 2022
Tue Nov 22 12:49:43 PM EST 2022
```

... `sysntpd` consistently does not successfully update.

Hurricos, 2022-11-22:
Two problems:

- `systntpd` does not wait for network access to attempt to get the current time
- `sysntpd`'s `procd` triggers are sufficient to invoke `start_service`, but this does not actually cause the service to re-attempt an NTP query (either by restarting the daemon or sending it a signal, etc.).

Hurricos, 2022-11-22:
There's a third problem here: `sysntpd`'s `procd` triggers are not always sufficient to invoke `start_service` in the case of a network state change if `use_dhcp` is off -- that is, we only have a network-state-trigger to reload if `use_dhcp` is on.

Unrelatedly, there's a hotplug race-condition in the network -> sysntpd `/etc/init.d/network` reload.

Hurricos, 2022-11-22:
A fourth problem might be that although busybox ntpd does not produce errors if it fails to act as an NTP client. Looking into what exactly fails in ntpd now.

Of the two sets (1 vs 4) of valid problems, only one need be solved.

The OpenWrt config with which ntpd doesn't succeed on boot involves an external DNS resolver, so I'm going to guess the issue is some sort of bug with refreshing DNS resolution after a failure that busybox hasn't yet addressed, which would be easiest solved by just waiting until DNS is back, but probably best solved by a patch to busybox.

Hurricos, 2022-11-22:
OK, it's a `ujail` porting issue combined. If you use an external resolver, you get:

```txt
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151348, tv_nsec=377491500}) = 0
open("/etc/hosts", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/resolv.conf", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
read(3, "", 248)                        = 0
close(3)                                = 0
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151348, tv_nsec=391753460}) = 0
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151348, tv_nsec=392073940}) = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC|SOCK_NONBLOCK, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151348, tv_nsec=403804320}) = 0
sendto(3, "\306M\1\0\0\1\0\0\0\0\0\0\0010\7openwrt\4pool\3ntp\3"..., 40, MSG_NOSIGNAL, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 40
sendto(3, "\2522\1\0\0\1\0\0\0\0\0\0\0010\7openwrt\4pool\3ntp\3"..., 40, MSG_NOSIGNAL, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 40
poll([{fd=3, events=POLLIN}], 1, 2500)  = 0 (Timeout)
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151350, tv_nsec=917659980}) = 0
sendto(3, "\306M\1\0\0\1\0\0\0\0\0\0\0010\7openwrt\4pool\3ntp\3"..., 40, MSG_NOSIGNAL, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 40
sendto(3, "\2522\1\0\0\1\0\0\0\0\0\0\0010\7openwrt\4pool\3ntp\3"..., 40, MSG_NOSIGNAL, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 40
poll([{fd=3, events=POLLIN}], 1, 2500)  = 0 (Timeout)
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151353, tv_nsec=422786220}) = 0
close(3)                                = 0
write(2, "ntpd: bad address '0.openwrt.poo"..., 43) = 43
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151353, tv_nsec=423631820}) = 0
poll([], 0, 143000n)                     = 0 (Timeout)
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151496, tv_nsec=524269380}) = 0
fork()                                  = 2
clock_gettime64(CLOCK_REALTIME, {tv_sec=1669151496, tv_nsec=526236600}) = 0
poll([], 0, 109000)                     = 0 (Timeout)
```

What I don't understand is this bit:

```txt
open("/etc/resolv.conf", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
read(3, "", 248)                        = 0
close(3)                                = 0
```

Why do I get a successful open, but nevertheless read 0 after attempting to read 248B?

Hurricos, 2022-11-22:
OK. The problem is that for systems which do not use dnsmasq for DNS caching, `sysntpd` starts (and its ujail is set up) before `/tmp/resolv.conf.d/resolv.conf.auto` (the ultimate target of `/etc/resolv.conf`) is finalized. 

On first boot:

```console
root@ap3825i0:/proc/2931# grep resolv mountinfo
43 37 0:16 /resolv.conf.d/resolv.conf.auto//deleted /etc/resolv.conf ro,noatime - tmpfs tmpfs rw
```

After a restart:
```console
root@ap3825i0:/proc/3540# grep resolv mountinfo
43 37 0:16 /resolv.conf.d/resolv.conf.auto /etc/resolv.conf ro,noatime - tmpfs tmpfs rw
```

This affects *all* OpenWrt versions post-ujail-transition, but it's not necessarily showing up on all of them, usually because `dnsmasq` is running and `ntpd.c` will default to querying 127.0.0.1:53 should it fail elsewhere.

Hurricos, 2022-11-22:
... and that is this: https://github.com/openwrt/openwrt/issues/10389

@birdie-github I'd appreciate if you can confirm that on first boot, if you find the process ID `$PID` of your ntpd with e.g. `ps www | grep ntpd | grep -v ujail`, that `grep resolv /proc/$PID/mountinfo` gives you roughly `43 37 0:16 /resolv.conf.d/resolv.conf.auto//deleted /etc/resolv.conf ro,noatime - tmpfs tmpfs rw` and not `43 37 0:16 /resolv.conf.d/resolv.conf.auto /etc/resolv.conf ro,noatime - tmpfs tmpfs rw`; that way, we can move on to merging this with #10389.

Hurricos, 2022-11-23:
Correction: two *restarts*. Reload only causes a restart when the resulting configuration is different.

odon-noda, 2023-01-06:
### Workaround
Start ntpd directly, without jail
```
# mv /etc/capabilities/ntpd.json /etc/capabilities/ntpd.json-disabled
# service sysntpd restart
```
Details:
[When will sysntpd sync time?](https://forum.openwrt.org/t/when-will-sysntpd-sync-time/109382/6?u=odon-noda)

C0rn3j, 2025-01-19:
[image]

My router is also incapable of using NTP and loses the time on every reboot.

No issues are reported in the Web UI at all despite the option clearly not working.

C0rn3j, 2025-02-13:
@lhotari I presume the issue is gone after you upgrade to 24.10?

@fblaese and I suppose this issue can be closed now that 24.10 is out?

lhotari, 2025-02-13:
> [@lhotari](https://github.com/lhotari) I presume the issue is gone after you upgrade to 24.10?

@C0rn3j I actually meant that I upgraded to 24.10.0 on TP-Link Deco M4R v1 and I hit the issue. There's a capture of a session where this problem can be seen: https://gist.github.com/lhotari/a196857a664e0f267e1dabc7018559fc . After upgrading the clock didn't sync without manually doing it.

dimitris-personal, 2025-02-13:
@lhotari the gist doesn't show any significant time change:

```
...
root@mydeco:~# date
Tue Feb  4 01:26:39 EET 2025
root@mydeco:~# /etc/init.d/sysntpd restart
root@mydeco:~# date
Tue Feb  4 01:26:45 EET 2025
root@mydeco:~# opkg update
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
...
```

I also went back to my own 24.10.0 device and couldn't reproduce this this (although, in that case it's been up for days so the log ring buffer has lost the initial entries so couldn't check for any evidence from its early uptime).

What time (roughly) was the gist text taken?  Is it possible to reproduce with the addition of a `date` command after the successful `opkg` commands?

lhotari, 2025-02-13:
> [@lhotari](https://github.com/lhotari) the gist doesn't show any significant time change:

@dimitris-personal That's true. I'm assuming that the time got updated just before running `opkg update`. I didn't record it, but the date was today's date after I checked it. 

> I also went back to my own 24.10.0 device and couldn't reproduce this this (although, in that case it's been up for days so the log ring buffer has lost the initial entries so couldn't check for any evidence from its early uptime).

I was able to reproduce with TP-Link Deco M4R v1. When I upgraded 24.10.0 for a Cudy WR3000, there wasn't such issue.

> What time (roughly) was the gist text taken? Is it possible to reproduce with the addition of a `date` command after the successful `opkg` commands?

The gist text is copy-pasted from a terminal a few minutes before creating the gist. As mentioned above, the `date` was correct after I checked. I didn't check it immediately since I first assumed that magically restarting sysntpd did something else like syncing a hardware clock etc. It seems that Deco M4R v1 doesn't have a hardware clock at all.
The only reasonable explanation seems to be that sysntpd made the update just before I ran `opkg update`.

I can try to reproduce again on Deco M4R v1. The same issue happened twice that even if the device was running for a long time, the time wasn't correct and `opkg update` would fail. It seems that after rebooting it, it also takes some time until the time gets updated. Perhaps it's not directly related to the upgrade but can sporadically happen after any reboot?

webknjaz, 2025-02-27:
Hi, just wanted to report that this is still happening on GL.iNet Flint 2 (GL-MT6000) with `OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.049.66344~2b8e93c`.

quinndiggity, 2025-04-11:
Experiencing this on `24.10` - in my case the wan is a bond connection, and there seems to be a race condition or insufficient event triggering for `sysntpd` to be started after the bond has slave connections. Working around via `rc.local`  with:
```
...
sleep 180
/etc/init.d/sysntpd restart || /bin/true
...
```

quinndiggity, 2025-04-11:
What is odd is that `sysntpd` never recovers after connectivity is established or lost + resumed - the service continues to run, while not performing its purpose

quyenvsp, 2025-06-28:
For anyone using syslog-ng and pbr you can do follow

/etc/syslog-ng.d/sysntpd.conf
```
filter f_sysntpd {
  match("(.*)on_start(.*): wan(.*)", value("MESSAGE"));
};

destination d_sysntpd {
  program("/etc/commands/sysntpd_restart.sh" template("<${PRI}>${DATE} ${HOST} ${MESSAGE}\n"));
};

log {
  source(src);
  filter(f_sysntpd);
  destination(d_sysntpd);
};
```

/etc/commands/sysntpd_restart.sh
```
#!/bin/sh

while read line ; do
  sleep 5
  logger -t sysntpd "Restarting sysntpd for sync time"
  /etc/init.d/sysntpd restart
done
```

webknjaz, 2026-10-02:
> Still happening on 25.12.5, GL.iNet GL-MT6000.

For people landing here from search engines, I've put the workaround disabling jail for ntp (until the underlying race condition is fixed) that @odon-noda posted above and [on the forum](https://forum.openwrt.org/t/when-will-sysntpd-sync-time/109382/6?u=webknjaz)

```console
$ mv /etc/capabilities/ntpd.json /etc/capabilities/ntpd.json-disabled
$ service sysntpd restart
```

---

Since I'm making my own ImageBuilder images, I've put the above into `/etc/uci-defaults/99-sysntpd-workaround` so that it runs on the first put upon flashing a new dumb AP.