openwrt/openwrt #10844: SNR (2.45GHz) very high with no associated STAs

Ohfalderal, 2022-09-29:
22.03.0  / TPLink C7 V5
R/T graph displays high SNR on the 2.45GHz band with no associated STAs. The LED also flashes rapidly. For the 5GHz the SNR is within the expected range.

brada4, 2022-09-30:
What are the high SNR values you see? -20dBm , i.e 1% noise 99% signal should be still ok. If you insist it is not wifi, it could be some other noise in the range like bluetooth, cheap walkie-talkie etc that do not show up in WiFi survey tools.

brada4, 2022-10-13:
set correct country code and auto-select the channel.

Ohfalderal, 2022-10-14:
That has been the case since (see image).

[image]

...and the content -

```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0'
	option htmode 'VHT80'
	option channel 'auto'
	option country 'NL'
	option distance '12'
	option txpower '20'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option wmm '0'
	option wpa_disable_eapol_key_retries '1'
	option key 'YourNameHere'
	option ssid '( (( ((( O ))) )) )'
	option ieee80211w '1'
	option encryption 'sae-mixed'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/ahb/18100000.wmac'
	option htmode 'HT40'
	option channel 'auto'
	option country 'NL'
	option txpower '17'
	option cell_density '0'
	option distance '5'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option wmm '0'
	option wpa_disable_eapol_key_retries '1'
	option key 'YourNameHere'
	option ssid '( (( ((( O ))) )) )'
	option ieee80211w '1'
	option encryption 'psk2'
```

brada4, 2022-10-14:
Purely GUI configuration, same tradename, but v5, I am in a saturated area inside a city, so HT20, but it should fall to HT20 automatically if finding other wifi access point (most if not all radios do not consider non-wifi noise). Try jumping channels, 1-5-9-13, if it is the neighbors' baby speaker it may help.
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/ahb/18100000.wmac'
        option band '2g'
        option htmode 'HT20'
        option channel 'auto'
        option country 'BE'
        option cell_density '0'
```

Ohfalderal, 2022-10-14:
@brada4 
Oke, maar zie ik niet hetzelfde gedrag op 21.04?

brada4, 2022-10-14:
@Ohfalderal Voor mij was hetzelfde met oude versie.
I cannot report regression or progression between 21.02.4 and 22.03.0. It was all good all the way through.