openwrt/openwrt #10848: Missing nftables string match support in OpenWRT 22.03

birdie-github, 2022-09-29:
Looks like you should have kept iptables a little longer.

nftables do not support string match which many sorely need.

https://serverfault.com/questions/1111903/translating-iptables-hex-string-match-algo-bm-rules-to-nftables

champtar, 2022-09-29:
Have you tried iptables-nft ?

birdie-github, 2022-09-29:
No idea how to use it and I bet iptables kernel modules need to be installed. :(

champtar, 2022-09-29:
You use it exactly as you use iptables, but it plays nicer with nft

brada4, 2022-09-30:
string module rules do not emit any nft rule whatsoever. 
It has been so for almost 10 years of iptables-nft and nobody ever complained at nftables bug tracker.

brada4, 2022-09-30:
You can detect such situation drilling your custom ruleset through iptables-translate , unsupported tules will be emitted as `nft # -A SOMETHING ....`
Particular case in serverfault posting is not very reliable in face of fragmentation, but if you have a specific case do not hesitate to share.

birdie-github, 2022-09-30:
The serverfault post contains the rules which I need to translate. Looks like I'll have to install iptables or maybe try to learn without these rules. At least one of them is no longer required (the hex DNS match): all my devices now use DNS over TLS to resolve IPs.

The other one is barely needed as well as I now use VPN pretty much all the time.

Maybe this bug report/feature request is worth closing since it's not actionable on your part.

At least you could update documentation and add something like:

"Some iptables features have yet to be implemented in nftables, for instance string matching, thus you can try using iptables rules in conjunction with nftables which is perfectly allowed and possible. You'll have to install the required packages because OpenWRT starting with version 22.03 by default only supports nftables".

brada4, 2022-09-30:
Timeproof option is to do blackholing in userspace. Something like `return 444` in nginx language.

birdie-github, 2022-09-30:
> Timeproof option

https://www.google.com/search?q=%22timeproof%22+site%3Aopenwrt.org

Zero results. Not sure what you're referring to.

birdie-github, 2022-09-30:
> Timeproof option is to do blackholing in userspace. Something like `return 444` in nginx language.

Those rules are to protect me **against** my ISP which filters traffic willy-nilly. I live in a totalitarian country.

brada4, 2022-09-30:
21.02 will serve you for some time. 
For DNS: https://www.icann.org/resources/pages/wildcard-prohibition-2014-01-29-en clicks away you find resources on how to tune recursor to discard poison responses