openwrt/openwrt #10849: Dropbear won't accept keys included in customized ROM images

M95D, 2022-09-29:
I build my own image with various configuration files included in the ROM image, to avoid (re)configuring it after the first boot.
One of these customized configuration is not working as expected: dropbear keys.
Due to wrong permissions set on /etc/dropbear/*, dropbear won't accept them and login fails. If I also disable root login with password, the only way to connect is a serial port.

Openwrt build system should set "600" permissions on /etc/dropbear/* after including the user key files.

Thanks

rockdrilla, 2022-11-11:
@M95D please describe how you're including your files in ROM.

M95D, 2022-11-11:
I have a symlink for "files" pointing to a directory outside the openwrt, on a NFS share from the real router (where I sync with the original files). Files are owned by a regular user, not root.
I'll try to set permissions to 600.
@rockdrilla What are the permissions of the original files you are including in the squashfs?

rockdrilla, 2022-11-12:
I don't include any files while building ROM yet but I'd plan to do this in future. I'll look for proper solution for file permissions.

rockdrilla, 2022-11-14:
@M95D please take a look at #11223.

commit 05100d86 (2024-02-09): dropbear: adjust file permissions

runtime:
- adjust ownership/permissions while starting dropbear
build time:
- correct file permissions for preseed files in $(TOPDIR)/files/etc/dropbear/ (if any)

closes #10849
closed 2023-12-11 by M95D