openwrt/openwrt #10856: LibreRouter-v1 support lacks watchdog driver

ilario, 2022-09-30:
LibreRouter-v1 has been supported since OpenWrt 19.07 with https://github.com/openwrt/openwrt/commit/24463d0d9479bf701e3aaa122f539a43cc97402b by @spiccinini.
It gets shipped with LibreRouterOS 1.5 based on LibreMesh 2020.1 based on OpenWrt 19.07.

During BattleMesh v14, @T-X realized that the hardware watchdog (more details here https://gitlab.com/librerouter/watchdog-fw and here https://gitlab.com/librerouter/boards) was not interacted with by the kernel (neither when using OpenWrt 22.03.0 nor with 19.07.10) as reported here https://github.com/freifunk-gluon/gluon/pull/2654, even if the instructions for that are included in the DTS file here: 
https://github.com/openwrt/openwrt/blob/24463d0d9479bf701e3aaa122f539a43cc97402b/target/linux/ath79/dts/qca9558_librerouter_librerouter-v1.dts#L58-L64

This means that during LibreRouterOS compilation some kernel modules/features are selected and these should be used also for compiling OpenWrt images for the LibreRouter-v1.
The interesting option should be one of these:
https://gitlab.com/librerouter/librerouteros/-/blob/7d02a29f8ca932b6edf747fb709d6f2d6ef0842e/configs/default_config

spiccinini, 2022-09-30:
Thanks Ilario for reporting it! 

I forgot to upstream this fix when I added 19.07 support to LibreRouterOs last year...

blocktrron, 2022-10-02:
Fixed with #10859

closed 2022-10-02 by blocktrron

ilario, 2022-10-02:
Amazing! Thanks!!