openwrt/openwrt #10858: iwlwifi: Microcode SW error detected

arrmo, 2022-09-30:
Hi,

First off, apologies if this is a duplicate of other issue(s) - it's not so clear if it's the same issue or not 😞. I have an `Intel(R) Dual Band Wireless AC 7260, REV=0x144` mini PCIe card running, in general relatively well ... as an AP, to make it clear. It works, except I do see issues like the following ... any suggestions?
```
[35370.871896] iwlwifi 0000:01:00.0: Queue 3 is active on fifo 5 and stuck for 10000 ms. SW [106, 10] HW [106, 10] FH TRB=0x080503071
[35370.885112] iwlwifi 0000:01:00.0: Microcode SW error detected.  Restarting 0x2000000.
[35370.893367] iwlwifi 0000:01:00.0: Start IWL Error Log Dump:
[35370.899114] iwlwifi 0000:01:00.0: Transport status: 0x0000004A, valid: 6
[35370.906040] iwlwifi 0000:01:00.0: Loaded firmware version: 17.3216344376.0 7260-17.ucode
[35370.914429] iwlwifi 0000:01:00.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN
[35370.921608] iwlwifi 0000:01:00.0: 0x008002F1 | trm_hw_status0
[35370.927575] iwlwifi 0000:01:00.0: 0x00000000 | trm_hw_status1
[35370.933541] iwlwifi 0000:01:00.0: 0x00000B30 | branchlink2
[35370.939230] iwlwifi 0000:01:00.0: 0x000164C0 | interruptlink1
[35370.945163] iwlwifi 0000:01:00.0: 0x00027F96 | interruptlink2
[35370.951123] iwlwifi 0000:01:00.0: 0x00000000 | data1
[35370.956248] iwlwifi 0000:01:00.0: 0x00000080 | data2
[35370.961400] iwlwifi 0000:01:00.0: 0x07030000 | data3
[35370.966525] iwlwifi 0000:01:00.0: 0x2740CA23 | beacon time
[35370.972224] iwlwifi 0000:01:00.0: 0x3B7485E7 | tsf low
[35370.977557] iwlwifi 0000:01:00.0: 0x00000008 | tsf hi
[35370.982772] iwlwifi 0000:01:00.0: 0x00000000 | time gp1
[35370.988195] iwlwifi 0000:01:00.0: 0x3B7485E8 | time gp2
[35370.993589] iwlwifi 0000:01:00.0: 0x00000000 | uCode revision type
[35370.999997] iwlwifi 0000:01:00.0: 0x00000011 | uCode version major
[35371.006377] iwlwifi 0000:01:00.0: 0xBFB58538 | uCode version minor
[35371.012793] iwlwifi 0000:01:00.0: 0x00000144 | hw version
[35371.018396] iwlwifi 0000:01:00.0: 0x40489204 | board version
[35371.024238] iwlwifi 0000:01:00.0: 0x0057011C | hcmd
[35371.029303] iwlwifi 0000:01:00.0: 0x00922060 | isr0
[35371.034338] iwlwifi 0000:01:00.0: 0x00000000 | isr1
[35371.039402] iwlwifi 0000:01:00.0: 0x00000002 | isr2
[35371.044438] iwlwifi 0000:01:00.0: 0x0041B0C0 | isr3
[35371.049503] iwlwifi 0000:01:00.0: 0x00000000 | isr4
[35371.054539] iwlwifi 0000:01:00.0: 0x00080110 | last cmd Id
[35371.060231] iwlwifi 0000:01:00.0: 0x00000000 | wait_event
[35371.065804] iwlwifi 0000:01:00.0: 0x0000E9FF | l2p_control
[35371.071498] iwlwifi 0000:01:00.0: 0x00000820 | l2p_duration
[35371.077250] iwlwifi 0000:01:00.0: 0x00000003 | l2p_mhvalid
[35371.082950] iwlwifi 0000:01:00.0: 0x00009000 | l2p_addr_match
[35371.088912] iwlwifi 0000:01:00.0: 0x00000007 | lmpm_pmg_sel
[35371.094664] iwlwifi 0000:01:00.0: 0x15041745 | timestamp
[35371.100178] iwlwifi 0000:01:00.0: 0x0034D8E0 | flow_handler
[35371.105970] iwlwifi 0000:01:00.0: Fseq Registers:
[35371.110861] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_ERROR_CODE
[35371.116965] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_TOP_INIT_VERSION
[35371.123533] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_CNVIO_INIT_VERSION
[35371.130268] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_OTP_VERSION
[35371.136420] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_TOP_CONTENT_VERSION
[35371.143252] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_ALIVE_TOKEN
[35371.149360] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_CNVI_ID
[35371.155152] iwlwifi 0000:01:00.0: 0x00000000 | FSEQ_CNVR_ID
[35371.160941] iwlwifi 0000:01:00.0: 0x00000000 | CNVI_AUX_MISC_CHIP
[35371.167273] iwlwifi 0000:01:00.0: 0x00000000 | CNVR_AUX_MISC_CHIP
[35371.173606] iwlwifi 0000:01:00.0: 0x00000000 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[35371.181686] iwlwifi 0000:01:00.0: 0x00000000 | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[35371.190167] iwlwifi 0000:01:00.0: Collecting data: trigger 2 fired.
[35371.196687] ieee80211 phy0: Hardware restart was requested
[35371.379101] iwlwifi 0000:01:00.0: Failing on timeout while stopping DMA channel 8 [0x07df0001]
```

Thanks!

arrmo, 2022-10-06:
NP, understand. But is it clear it's not the driver, vs. the firmware? Just to understand.

Thanks!

stintel, 2022-10-06:
It could be that the driver causes the firmware crash, but without knowledge of the firmware we can only guess why/where it happens.

arrmo, 2022-10-07:
Agreed! Will try to turn tracing on, see if I can help debug. Also though ... is it possible to pull in a later / more current driver (for testing)?

Thanks.