openwrt/openwrt #10866: firewall4: IPv6 DNAT requires masq6 on wan

ngehrsitz, 2022-10-01:
Firewall4 does not set up all required `nft` rules for IPv6 DNAT if the `wan` interface does not have the `masq6` option set. I have the following DNAT rule:
```
config redirect
    option name 'WAN IPv6 HTTP Redirect'
    option target 'DNAT'
    list proto 'tcp'
    option src 'wan'
    option dest 'lan'
    option family 'ipv6'
    option src_dport '80'
    option dest_ip '2001:16b8:28b3:c02::1' # 
    option dest_port '8080'
```
and the configuration of my `wan` interface:
```
config zone
    option name	'wan'
    list   network	'wan'
    list   network	'wan6'
    option input	DROP
    option output	ACCEPT
    option forward	DROP
    option masq		'1'
    option mtu_fix	'1'
```
When looking at the output of  `nft list ruleset` I can see 
```
        chain dstnat_wan {

                meta nfproto ipv6 tcp dport 80 counter packets 0 bytes 0 dnat ip6 to [2001:16b8:28b3:c02::1]:8080 comment "!fw4: WAN IPv6 HTTP Redirect"
        }
```
but my traffic does not get passed through. I figured out that the issue is that there is no snat in the other direction. The issue is the check `if (redir.src.zone[i ? "masq6" : "masq"] && length(rip[i])) {`[[Ll2849](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=29ae0534afbd50612d621db021c1d42602a7e940;hb=refs/heads/master#l2849)]. If I manually patch that to `if (length(rip[i])) {` then I see these additional snat rules and it works as expected:
```
        chain dstnat_lan {
                ip6 saddr { 2001:16b8:28c0:e202::/64, fd4f:49f9:19fe:2::/64 } ip6 daddr { 2001:16b8:2900:b207:c83c:4387:8055:e0b4, fe80::c83c:4387:8055:e0b4 } tcp dport 80 dnat ip6 to [2001:16b8:28b3:c02::1]:8080 comment "!fw4: WAN IPv6 HTTP Redirect (reflection)"
        }

        chain srcnat_lan {
                ip6 saddr { 2001:16b8:28c0:e202::/64, fd4f:49f9:19fe:2::/64 } ip6 daddr 2001:16b8:28b3:c02::1 tcp dport 8080 snat ip6 to 2001:16b8:28c0:e202::1 comment "!fw4: WAN IPv6 HTTP Redirect (reflection)"
        }
```
@jow- What purpose should the check in the current form serve?

ptpt52, 2022-10-01:
mark.

jow-, 2022-10-05:
This is intentional. If you do DNAT for IPv6 you need to take care of the reverse path too, either by doing an unconditional IPv6 masquerade or by setting up a corresponding SNAT rule. This is the very same behaviour as for IPv4, just that masquerading happens to be amost always enabled by default there so usually you do not have to deal with it.

The line you linked belongs to the reflection rule logic, it is required for hairpin and not intended to cover the normal reverse path requirement.

closed 2022-10-05 by jow-