openwrt/openwrt #10875: When using sysupgrade for swconfig <=> DSA images migration, --force is required even if -n is provided

ilario, 2022-10-02:
As indicated in the OpenWrt 21.02 release notes https://openwrt.org/releases/21.02/notes-21.02.0#upgrading_to_21020, there is not yet (some progress going on on #10796) a migration path from the swconfig configuration files to the DSA ones.

For this reason, a warning message is displayed "Image version mismatch. image 1.1 device 1.0 Please wipe config during upgrade (force required) or reinstall. Config cannot be migrated from swconfig to DSA Image check failed".

The problem is that this message gets displayed and `--force` is required even when using the `-n` option is provided.

To my understanding, when using the `-n` option for just discarding all the previous configuration, there should be no warning message. Its appearance is confusing.

pkgadd, 2022-10-03:
> The problem is that this message gets displayed and `--force` is required even when using the `-n` option is provided.

That is correct and the intended behaviour, not a bug. While not ideal, it's simply not possible to change the past and teach old sysupgrade versions about the image version without either not prompting the user or refusing to apply the new image (which is the intended and sole purpose of the compat version, to prevent blindly applying breaking changes).

ilario, 2022-10-03:
Let me go trough your comment to make sure I understood it:

> That is correct and the intended behaviour, not a bug. While not ideal, it's simply not possible to change the past and teach old sysupgrade versions about the image version

Mh, does it mean that the old versions of sysupgrade have no way to distinguish between these two cases:

* the upgrade is going to break stuff
* the upgrade is going to break stuff unless you discard the configuration (like in the swconfig <=> DSA case)

is this correct?

If this is the case, I agree that there is no alternative to the current behavior.

pkgadd, 2022-10-03:
Old sysupgrade (and that's what needs to be considered for at least 2-3 more major releases, if not indefinitely) doesn't know anything about the compat_version, it's a trick to make the image name incompatible and prevent problems. There are no finer nuances (for the old sysupgrade), other than the image not being considered compatible. 

New sysupgrade versions can ignore the compat_version mismatch selectively, but that only helps years down the lane for potential future compat_version changes.

ilario, 2022-10-04:
Ok! Understood! Thanks for the patience :)

closed 2022-10-04 by ilario