openwrt/openwrt #10879: ipq806x: ZyXEL NBG6817: Low TX speed on wan port

csharper2005, 2022-10-03:
### Description
Low (20-50 Mbps) TX speed on WAN port when WAN connected at 100Mbps (Fast Ethernet).

### How to reproduce
1. Connect WAN port at 100Mbps (using 4-wire ethernet cable);
2. Run SpeedTest;
3. Make sure that download speed is OK and upload speed is poor (20-50 Mbps).

### Affected OpenWrt releases
All:
1. OpenWrt 22.03.0
2. OpenWrt 21.02.x
3. Snapshot

### Affected devices
It is possible that all ipq806x (needs testing / confirmation).

### Known workarounds
1. Use dedicated USB NIC for WAN conection;
2. Install additional GE switch between router and WAN cable.

pesa1234, 2022-10-10:
Please try to disable with ethtool the flowcontrol on that lan interface on router...

csharper2005, 2022-10-10:
> Please try to disable with ethtool the flowcontrol on that lan interface on router...

@pesa1234, it seems FC is already disabled... Is this the correct way to check?

```
# ethtool eth1.1
Settings for eth1.1:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Link detected: yes

# ethtool eth0.2
Settings for eth0.2:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Link detected: yes

# ethtool --show-pause eth0
Pause parameters for eth0:
Autonegotiate:  on
RX:             off
TX:             off
RX negotiated:  off
TX negotiated:  off

# ethtool --show-pause eth1
Pause parameters for eth1:
Autonegotiate:  on
RX:             off
TX:             off
RX negotiated:  off
TX negotiated:  off

# ethtool -A eth0 rx off tx off
rx unmodified, ignoring
tx unmodified, ignoring
no pause parameters changed, aborting

# ethtool -A eth1 rx off tx off
rx unmodified, ignoring
tx unmodified, ignoring
no pause parameters changed, aborting
```

pesa1234, 2022-10-10:
Can you please show /etc/config/network?

csharper2005, 2022-10-10:
> Can you please show /etc/config/network?

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fde2:fe9e:xxxx::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1.1'

config device
	option name 'eth1.1'
	option macaddr '8c:xx:xx:xx:xx:xx'

config interface 'LAN'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.128'
	option ipaddr 'xxx.xx.xxx.xxx'
	list dns '127.0.0.1'
	option ip6assign '64'
	option ip6hint 'xxxx'
	option ip6ifaceid '::xxxx:xxxx'

config device
	option name 'eth0.2'
	option macaddr '8c:xx:xx:xx:xx:xx'

config interface 'ETH'
	option proto 'dhcp'
	option metric '40'
	option hostname '*'
	option device 'eth0.2'
	option peerdns '0'
	list dns '127.0.0.1'

config interface 'ETH6'
	option proto 'dhcpv6'
	option peerdns '0'
	option reqprefix 'auto'
	option device '@ETH'
	option metric '42'
	option reqaddress 'try'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option vid '1'
	option ports '6t 1 2 3 4'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option vid '2'
	option ports '0t 5'

[... virtual interfaces ...]
```

pesa1234, 2022-10-10:
Sorry where is connected your Fast Ethernet device?

csharper2005, 2022-10-10:
> Sorry where is connected your Fast Ethernet device?

wan port:
```
interface 'ETH'
option device 'eth0.2'
```

lan - 1 port connected (Gbit Ethernet switch)

pesa1234, 2022-10-10:
> > Sorry where is connected your Fast Ethernet device?
> 
> wan port:
> 
> ```
> interface 'ETH'
> option device 'eth0.2'
> ```
> 
> lan - 1 port connected (Gbit Ethernet switch)

Sorry, I don't understand you... I don't understand where is connected your fast ethernet devices, from ethtool output seems that is connected to a Gbit lan port

csharper2005, 2022-10-10:
> Sorry, I don't understand you... I don't understand where is connected your fast ethernet devices, from ethtool output seems that is connected to a Gbit lan port.

I'm sorry too. I have no ideas why it happens.

[image]
But:
```
# ethtool eth0.2
Settings for eth0.2:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Link detected: yes
```

pesa1234, 2022-10-10:
Ok, now on wan show output of ethtool

csharper2005, 2022-10-10:
> Ok, now on wan show output of ethtool

Is it possible that ethtool shows CPU port?
```
# ethtool eth0.2
Settings for eth0.2:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Link detected: yes
```

pesa1234, 2022-10-10:
I think that your vLAN config is not good.
I think that WAN network should be attached to eth0, not eth0.2

quarkysg, 2022-10-13:
I don't the switch port's flow-control can be turned off using ethtool for ipq806x devices.  There's no support built-in to do it for the ipq806x switch ports.

I believe the issue is caused by the wan port stopping traffic via flow-control due to buffer full from GB LAN client sending to FE WAN link.

The community provided NSS enabled ipq806x firmware should be able to turn off flow-control of switch ports if the QCA SSDK and QCA SSDK Shell is compiled in.  The QCA SSDK Shell is finicky to use tho.

csharper2005, 2022-10-13:
@quarkysg 
Thanks for your idea. And I tried to install the NSS firmware. Unfortunately it does not contain the modules I need for LTE modems and no one compiles it with the ALL_KMODS option. Maybe someday I'll try to build it myself with all necessary kmods.

racingcirc, 2024-01-27:
Looks like this issue is the one behind https://github.com/openwrt/packages/issues/23112

Is there any fix in the works at all? Having the TX speed limited to ~ 4Mbps per connection is quite a show-stopper for Transmission.

brada4, 2024-03-28:
Actually it is a total killer for speedtest-netperf.sh too as exposed in other issue. Here I see that it is swconfig platform. Cannot advise more, but you need to work out pause with swconfig, its been long time, follow help texts in command line.
Any pause in stats is bad. Few checksum errors like one in a million or few at reconnect are within expected.