openwrt/openwrt #10888: Linksys E2000 crash on boot after upgrade to 22.03

janecekp, 2022-10-04:
Hi,
   After upgrade to 22.03.0, Linksys E2000 gets stuck in a reboot loop:
...
[   16.134489] random: jshn: uninitialized urandom read (4 bytes read)
[   17.133563] urngd: v1.0.2 started.
[   17.422505] kmodloader: loading kernel modules from /etc/modules.d/*
[   18.068664] random: crng init done
[   18.072179] random: 19 urandom warning(s) missed due to ratelimiting
[   18.118256] Loading modules backported from Linux version v5.15.58-0-g7d8048d4e064
[   18.126149] Backport generated by backports.git v5.15.58-1-0-g42a95ce7
[   18.550976] PPP generic driver version 2.4.2
[   18.566140] NET: Registered protocol family 24
[   18.600897] b43-phy0: Broadcom 4716 WLAN found (core revision 17)
[   18.607668] b43-phy0: Found PHY: Analog 8, Type 4 (N), Revision 5
[   18.613966] b43-phy0: Found Radio: Manuf 0x17F, ID 0x2056, Revision 7, Version 0
[   18.622559] Broadcom 43xx driver loaded [ Features: NL ]
[   18.870464] kmodloader: done loading kernel modules from /etc/modules.d/*
[   36.568204] Reserved instruction in kernel code[#1]:
[   36.573326] CPU: 0 PID: 1085 Comm: ip Not tainted 5.10.138 #0
[   36.579178] $ 0   : 00000000 00000001 72665f68 81873c68
[   36.584563] $ 4   : 00000001 80020b4c 805f0040 756c665f
[   36.589932] $ 8   : 00000000 00000002 00000000 ffffff7f
[   36.595301] $12   : 808066b0 00000281 00000400 00000024
[   36.600672] $16   : 81873c78 00000000 00000000 00002000
[   36.606041] $20   : 00000000 80142824 00000000 00000010
[   36.611409] $24   : 00000044 00000002                  
[   36.616779] $28   : 81872000 81873c78 00000cc0 77dbbfff
[   36.622149] Hi    : 00013f50
[   36.625094] Lo    : 344546d0
[   36.628049] epc   : 77dbc001 0x77dbc001
[   36.631967] ra    : 77dbbfff 0x77dbbfff
[   36.635876] Status: 11008403 KERNEL EXL IE 
[   36.640181] Cause : 00800028 (ExcCode 0a)
[   36.644273] PrId  : 00019740 (MIPS 74Kc)
[   36.648270] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flo
w_table_inet b43 pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt cordic compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[   36.709801] Process ip (pid: 1085, threadinfo=f12bb65d, task=19ebbaae, tls=77e7edf0)
[   36.717687] Stack : 77910000 8072bcb8 806e9c98 800f76f4 8157a14c 00000000 81873c90 81873c90
[   36.726270]         00000000 00000001 80a78000 8059c71c 80a784a0 00000000 00000000 80a78000
[   36.734859]         80a62a0c 8034d548 81873d08 800441d8 00000cc0 00000000 80020b0c 805f0000
[   36.743446]         81554780 80a78000 8059c71c 00000000 00000000 8039d820 00000cc0 81554784
[   36.752033]         81873d04 8011c8d8 77dbc000 8157a540 80a78000 00000000 80a78000 00001003
[   36.760619]         ...
[   36.763134] Call Trace:
[   36.763159] [<800f76f4>] 0x800f76f4
[   36.769258] [<8034d548>] 0x8034d548
[   36.772826] [<800441d8>] 0x800441d8
[   36.776393] [<80020b0c>] 0x80020b0c
[   36.779961] [<8039d820>] 0x8039d820
[   36.783535] [<8011c8d8>] 0x8011c8d8
[   36.787104] [<8039dce8>] 0x8039dce8
[   36.790666] [<80020b0c>] 0x80020b0c
[   36.794233] [<80121974>] 0x80121974
[   36.797806] [<8039dd84>] 0x8039dd84
[   36.801376] [<80391e7c>] 0x80391e7c
[   36.804949] [<804739a4>] 0x804739a4
[   36.808533] [<80477df4>] 0x80477df4
[   36.812100] [<80142824>] 0x80142824
[   36.815669] [<80392124>] 0x80392124
[   36.819234] [<803c97d8>] 0x803c97d8
[   36.822801] [<8036e65c>] 0x8036e65c
[   36.826394] [<8016ef94>] 0x8016ef94
[   36.829974] [<8036c7bc>] 0x8036c7bc
[   36.833553] [<8036e890>] 0x8036e890
[   36.837125] [<8000e42c>] 0x8000e42c
[   36.840710] 
[   36.842236] Code: 0000  0000  0000 <457f> 464c  0101  0001  0000  0000 
[   36.849085] 
[   36.850821] ---[ end trace f58a791156b32824 ]---
[   36.855632] Kernel panic - not syncing: Fatal exception
[   36.860983] Rebooting in 3 seconds..
[   39.857198] bcm47xx: Please stand by while rebooting the system...

ynezz, 2022-10-04:
@janecekp it seems, that you've serial console available, could you please try to reproduce it with a [snapshot build](https://downloads.openwrt.org/snapshots/targets/bcm47xx/mips74k/) and post the crash here as well? It should help with resolving following magic numbers in stack trace:

```
[ 36.763134] Call Trace:
[ 36.763159] [<800f76f4>] 0x800f76f4
[ 36.769258] [<8034d548>] 0x8034d548
```
Once we get this resolved, it should be more clear where and why it crashes.

janecekp, 2022-10-04:
The snapshot gets stuck with the `Starting program at 0x80001000` message (nothing happened for 20 minutes).  Now, what should I try to tftp in?  21.03 doesn't work, as in the second log.
```
# sysupgrade -v /tmp/openwrt-bcm47xx-mips74k-linksys_e2000-v1-squashfs.bin 
Mon Oct  3 19:56:33 CEST 2022 upgrade: Image metadata not present
Found CyberTAN image with device pattern: 32XN
Found a valid TRX version 1
Mon Oct  3 19:56:37 CEST 2022 upgrade: Saving config files...
etc/config/dhcp
etc/config/dhcp-opkg
etc/config/dropbear
etc/config/firewall
etc/config/luci
etc/config/network
etc/config/openvpn
etc/config/rpcd
etc/config/system
etc/config/ucitrack
etc/config/uhttpd
etc/config/wireless
etc/dnsmasq.conf
etc/dropbear/authorized_keys
etc/dropbear/dropbear_dss_host_key
etc/dropbear/dropbear_ed25519_host_key
etc/dropbear/dropbear_rsa_host_key
etc/group
etc/hosts
etc/inittab
etc/iproute2/rt_tables
etc/luci-uploads/.placeholder
etc/opkg/customfeeds.conf
etc/opkg/keys/0b26f36ae0f4106d
etc/opkg/keys/1035ac73cc4e59e3
etc/opkg/keys/2f8b0b98e08306bf
etc/opkg/keys/5151f69420c3f508
etc/opkg/keys/53bad1233d4c98c5
etc/opkg/keys/72a57f2191b211e0
etc/opkg/keys/792d9d9b39f180dc
etc/opkg/keys/9ef4694208102c43
etc/opkg/keys/b26f36ae0f4106d
etc/opkg/keys/b2d571e0880ff617
etc/opkg/keys/b5043e70f9a75cde
etc/opkg/keys/c10b9afab19ee428
etc/opkg/keys/dace9d4df16896bf
etc/opkg/keys/dd6de0d06bbd3d85
etc/opkg/keys/de98a2dd1d0f8a07
etc/opkg/keys/f94b9dd6febac963
etc/passwd
etc/profile
etc/protocols
etc/rc.local
etc/shadow
etc/shells
etc/shinit
etc/sysctl.conf
etc/uhttpd.crt
etc/uhttpd.key
etc/uhttpd.key
etc/uhttpd.crt
Mon Oct  3 19:56:37 CEST 2022 upgrade: Commencing upgrade. Closing all shell sessions.
Watchdog handover: fd=3
- watchdog -
killall: telnetd: no process killed
Mon Oct  3 19:56:40 CEST 2022 upgrade: Sending TERM to remaining processes ... hostapd wpa_supplicant netifd odhcpd uhttpd udhcpc odhcp6c ntpd dnsmasq ubusd urngd logd rpcd
[  167.366096] device wlan0 left promiscuous mode
[  167.371160] br-lan: port 2(wlan0) entered disabled state
Mon Oct  3 19:56:46 CEST 2022 upgrade: Sending KILL to remaining processes ... ntpd
[  171.749100] sh (2430): drop_caches: 3
Mon Oct  3 19:56:47 CEST 2022 upgrade: Switching to ramdisk...
Mon Oct  3 17:56:52 UTC 2022 upgrade: Performing system upgrade...
[  177.400823] sh (2430): drop_caches: 3
[... 134 lines cut ...]
  Offset1: 0x91C	(2332)	(0xBC04091C)
  Offset2: 0x22E400	(2286592)	(0xBC26E400)
  Header CRC:    0x4518E74F
  Calculate CRC: 0x4518E74F
Image 1 is OK
Try to load image 1.
Waiting for 5 seconds to upgrade ...
CMD: [load -raw -addr=0x807a1360 -max=0x185eca0 :]
Loader:raw Filesys:tftp Dev:eth0 File:: Options:(null)
Loading: _tftpd_open(): retries=0/5
_tftpd_open(): retries=1/5
_tftpd_open(): retries=2/5
.. 5443616 bytes read

### Start=414516802 E=414579103 Delta=62301 ###
Entry at 0x807a1360
CMD: [boot -raw -z -addr=0x80001000 -max=0x185eca0 -fs=memory :0x807a1360]
Loader:raw Filesys:memory Dev:eth0 File::0x807a1360 Options:(null)
Loading: . 0 bytes read

### Start=422556237 E=424920760 Delta=2364523 ###
Failed.
Could not load :0x807a1360: Error
CFE> 
```

ynezz, 2022-10-04:
@rmilecki any idea?

rmilecki, 2022-10-04:
As for release image: not a single guess without a proper backtrace. That
```
[ 36.763134] Call Trace:
[ 36.763159] [<800f76f4>] 0x800f76f4
[ 36.769258] [<8034d548>] 0x8034d548
```
doesn't help :(

---

As for snapshot image: maybe including symbols makes image too big to load. @hauke ?

janecekp, 2022-10-04:
Before I can try anything, can you please help me unbrick the router?  As wrote above, I'm stuck in CFE, as tftp recovery doesn't accept standard fw image, not even the original Linksys.

rmilecki, 2022-10-05:
> Waiting for 5 seconds to upgrade ...
> CMD: [load -raw -addr=0x807a1360 -max=0x185eca0 :]
> Loader:raw Filesys:tftp Dev:eth0 File:: Options:(null)
> Loading: _tftpd_open(): retries=0/5
> _tftpd_open(): [...]

I don't know if CFE's auto-loading from TFTP server can work / be used.

For E2500 V3 model I used `flash` command **manually** to get raw TRX image from tftp server and flash it. It worked. See:
https://forum.openwrt.org/t/e2500-v3-problems-with-tftp-recovery/114529/10

ynezz, 2022-10-05:
> As for release image: not a single guess without a proper backtrace. That
> 
> ```
> [ 36.763134] Call Trace:
> [ 36.763159] [<800f76f4>] 0x800f76f4
> [ 36.769258] [<8034d548>] 0x8034d548
> ```
> 
> doesn't help :(

It can be done manually, with https://downloads.openwrt.org/releases/22.03.0/targets/bcm47xx/mips74k/kernel-debug.tar.zst and gdb:

```
$ gdb vmlinux

>>> l *0x8039d820
0x8039d820 is in __dev_open (net/core/dev.c:1611).

>>> l *0x80020b0c
0x80020b0c is in vm_area_dup (kernel/fork.c:358).

>>> l *0x800441d8
0x800441d8 is in raw_notifier_call_chain (kernel/notifier.c:88).

>>> l *0x8034d548
0x8034d548 is in bgmac_open (drivers/net/ethernet/broadcom/bgmac.c:1184).

>>> l *0x800f76f4
0x800f76f4 is in __pagevec_lru_add_fn (./include/linux/mm_inline.h:52).
```

IMO this doesn't make much sense to me, so probably corrupted stack, compiler issue?

Djfe, 2023-05-20:
Your device should've been deprecated already (32M RAM), sorry that you're stuck with a brick right now.
https://openwrt.org/supported_devices/864_warning

openwrt 23 won't build images for your device anymore (by default, you can still build images yourself)
https://github.com/openwrt/openwrt/pull/12672

We can't support devices with such a low amount of ressources any longer and have to suggest moving on to a newer device :)
If you manage to unbrick your Linksys, then I suggest staying with OpenWrt 21.

closed 2023-06-12 by ynezz

rmilecki, 2025-04-20:
FWIW this was solved by the
```
commit 4970dd027bced78f8a72aa2baed22c962d3b587b
Author: Rafał Miłecki <rafal@milecki.pl>
Date:   Tue Feb 7 18:09:40 2023 +0100

    bcm47xx: revert bgmac back to the old limited max frame size
```