openwrt/openwrt #10893: ustream-ssl wants to build but fails if it's not selected

yogo1212, 2022-10-04:
Somewhere between fbf6992f2b8960cbca36cd652bcdc71d69931076 and bd6783f4fb8f6171927e9067c0005a6d69fc13fe ustream-ssl decided that it was to be built even though it's not selected.

If I change the conditional `select` for libmbedtls to a `select`

<details><summary>like so</summary>

```
diff --git a/package/libs/ustream-ssl/Makefile b/package/libs/ustream-ssl/Makefile
index 3181f66c7e..9416fbe39f 100644
--- a/package/libs/ustream-ssl/Makefile
+++ b/package/libs/ustream-ssl/Makefile
@@ -44,7 +44,7 @@ endef
 define Package/libustream-mbedtls
   $(Package/libustream/default)
   TITLE += (mbedtls)
-  DEPENDS += +PACKAGE_libustream-mbedtls:libmbedtls
+  DEPENDS += libmbedtls
   CONFLICTS := libustream-openssl libustream-wolfssl
   VARIANT:=mbedtls
   DEFAULT_VARIANT:=1
```
</details>

It isn't built and everything works as expected.

<details><summary>Error message from `make world` + config</summary>

```
    ERROR: package/libs/ustream-ssl failed to build (build variant: mbedtls).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/media/openwrt/include/toplevel.mk:231: world] Error 1
$ grep -E '(ustream|mbedtls)' .config
CONFIG_DEFAULT_libustream-wolfssl=y
# CONFIG_PACKAGE_libmbedtls is not set
# CONFIG_PACKAGE_libustream-mbedtls is not set
# CONFIG_PACKAGE_libustream-openssl is not set
# CONFIG_PACKAGE_libustream-wolfssl is not set
# CONFIG_PACKAGE_px5g-mbedtls is not set
# CONFIG_PACKAGE_mbedtls-util is not set
```
</details>

<details><summary>build log ustream-ssl</summary>

```
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
make[2]: Entering directory '/media/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/media/openwrt/scripts/config'
make[1]: Entering directory '/media/openwrt'
make[2]: Entering directory '/media/openwrt/package/libs/ustream-ssl'
rm -rf /media/openwrt/build_dir/target-mipsel_24kc_musl/ustream-ssl-mbedtls/ustream-ssl-2022-01-16-868fd881
rm -f /media/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.ustream-ssl.mbedtls_installed
rm -f /media/openwrt/staging_dir/target-mipsel_24kc_musl/packages/ustream-ssl.mbedtls.list
make[2]: Leaving directory '/media/openwrt/package/libs/ustream-ssl'
time: package/libs/ustream-ssl/mbedtls/clean#0.13#0.04#0.25
make[1]: Leaving directory '/media/openwrt'
make[2]: Entering directory '/media/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/media/openwrt/scripts/config'
make[1]: Entering directory '/media/openwrt'
make[2]: Entering directory '/media/openwrt/package/libs/toolchain'
echo "libc" >> /media/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
[... 54 lines cut ...]

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_FIND_USE_PACKAGE_REGISTRY
    CMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY
    CMAKE_MODULE_LINKER_FLAGS
    DL_LIBRARY

CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [Makefile:75: /media/openwrt/build_dir/target-mipsel_24kc_musl/ustream-ssl-mbedtls/ustream-ssl-2022-01-16-868fd881/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/media/openwrt/package/libs/ustream-ssl'
time: package/libs/ustream-ssl/mbedtls/compile#0.26#0.13#0.36
    ERROR: package/libs/ustream-ssl failed to build (build variant: mbedtls).
make[1]: *** [package/Makefile:116: package/libs/ustream-ssl/compile] Error 1
make[1]: Leaving directory '/media/openwrt'
make: *** [/media/openwrt/include/toplevel.mk:231: package/ustream-ssl/compile] Error 2
```

</details>

ynezz, 2022-10-05:
> Somewhere between https://github.com/openwrt/openwrt/commit/fbf6992f2b8960cbca36cd652bcdc71d69931076 and https://github.com/openwrt/openwrt/commit/bd6783f4fb8f6171927e9067c0005a6d69fc13fe ustream-ssl decided that it was to be built even though it's not selected.

Probably d94c94d79542558d91c5a5de8d5f594dbafcf3c8 by @borkra ?

> $ grep -E '(ustream|mbedtls)' .config

Provide output of `./scripts/diffconfig.sh`

borkra, 2022-10-05:
Actually this was fixing similar issue with mbedtls.  It was building mbedtks even though it's not selected. To solve your problem we need to look at which package depends on ustream-ssl. 

Openwrt has an issue, dependencies for sub packages are built even when not selected, when main/any package for makefile  is selected. The only fix to this is to explicitly specify config precondition on each dependency.

borkra, 2022-10-05:
ustream-ssl is the dependency of the following packages:

- uhttpd 
- uclient 
- cshark
- cache-domains
- luci-ssl

if any of those packages selected ustream-ssl will be built

yogo1212, 2022-10-05:
thanks for your replies!

@ynezz here's the output:

<details><summary>diffconfig</summary>

```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_cudy_m1800=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_cudy_wr2100=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
# CONFIG_BUSYBOX_CONFIG_IP is not set
CONFIG_BUSYBOX_CONFIG_MKFS_VFAT=y
CONFIG_BUSYBOX_CONFIG_NC_110_COMPAT=y
CONFIG_BUSYBOX_CONFIG_NC_SERVER=y
# CONFIG_BUSYBOX_CONFIG_SWAPOFF is not set
# CONFIG_BUSYBOX_CONFIG_SWAPON is not set
CONFIG_BUSYBOX_CONFIG_TIMEOUT=y
CONFIG_BUSYBOX_CONFIG_USLEEP=y
CONFIG_FASTD_ENABLE_CIPHER_NULL=y
CONFIG_FASTD_ENABLE_CIPHER_SALSA2012=y
CONFIG_FASTD_ENABLE_MAC_UHASH=y
CONFIG_FASTD_ENABLE_METHOD_COMPOSED_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_GENERIC_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_NULL=y
CONFIG_FASTD_ENABLE_METHOD_NULL_L2TP=y
CONFIG_FASTD_WITH_OFFLOAD_L2TP=y
CONFIG_FASTD_WITH_STATUS_SOCKET=y
CONFIG_IMAGEOPT=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_batctl-default=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridge_mac_table=y
CONFIG_PACKAGE_fastd=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_ip-full=y
# CONFIG_PACKAGE_jansson is not set
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-l2tp=y
CONFIG_PACKAGE_kmod-l2tp-eth=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-mii=y
# CONFIG_PACKAGE_kmod-nft-fib is not set
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-hid=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-mbim=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb-serial-cp210x=y
CONFIG_PACKAGE_kmod-usb-serial-ftdi=y
CONFIG_PACKAGE_kmod-usb-serial-option=y
CONFIG_PACKAGE_kmod-usb-serial-qualcomm=y
CONFIG_PACKAGE_kmod-usb-serial-wwan=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-veth=y
CONFIG_PACKAGE_leggin=y
CONFIG_PACKAGE_libbase64=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libconfigd=y
CONFIG_PACKAGE_libdnscache=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libethmac=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libevent2-core=y
CONFIG_PACKAGE_libevent2-extra=y
CONFIG_PACKAGE_libevent2-openssl=y
CONFIG_PACKAGE_libevmqtt=y
CONFIG_PACKAGE_libevtsock=y
CONFIG_PACKAGE_libevtssl=y
CONFIG_PACKAGE_libevubus=y
CONFIG_PACKAGE_libjson-c-parser=y
CONFIG_PACKAGE_liblocalauth=y
CONFIG_PACKAGE_libmulticast_if=y
CONFIG_PACKAGE_libnames=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnlifinfo=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libringbuffer=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libsaferjson=y
CONFIG_PACKAGE_libsnot=y
# CONFIG_PACKAGE_libucode is not set
CONFIG_PACKAGE_libuecc=y
CONFIG_PACKAGE_libugly=y
# CONFIG_PACKAGE_libustream-wolfssl is not set
CONFIG_PACKAGE_libwifi_util=y
# CONFIG_PACKAGE_libwolfssl is not set
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_nftables-nojson=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_ucode is not set
CONFIG_PACKAGE_umbim=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_utftp=y
CONFIG_PACKAGE_uthash=y
CONFIG_PACKAGE_vpn-routes=y
CONFIG_PACKAGE_wget-ssl=y
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_zlib=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_PREINIT_DISABLE_FAILSAFE=y
# CONFIG_WPA_WOLFSSL is not set
```

</details>

@borkra i got to be honest - don't think i would've found that :-D
indeed, uclient-fetch is active:

```
CONFIG_DEFAULT_uclient-fetch=y
CONFIG_PACKAGE_libuclient=y
CONFIG_PACKAGE_uclient-fetch=y
```

ok, i'll disable opkg and uclient-fetch for the moment and see where it goes from there.
can't find the mentioned issue on github atm. but it should be possible to find this issue now that the words 'dependency', 'selected', and 'build' have been mentioned a few times.
i'll throw in a 'builds', 'regardless', and an 'anyway' just for good measure.

rdslw, 2024-04-28:
got the same problem, while building custom firmware w/o luci etc.
solved by patch as proposed by @yogo1212 to package/libs/ustream-ssl/Makefile