openwrt/openwrt #10894: pppd cannot reconnect - 22.03.00 - ipq806x

NilsRo, 2022-10-04:
With Zyxel NBG6817 (ipq806x) I have issues with pppd daemon. After some days it cannot handle the disconnect after 24h and I have to reboot the router. This is the log from it. Will go back to 21.02.03 which works without any problems. It a simple fiber connection with PPPoE.

Oct  4 15:55:07 OpenWrt netifd: Network device 'pppoe-wan' link is down
Oct  4 15:55:07 OpenWrt netifd: Network alias 'pppoe-wan' link is down
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan6' has link connectivity loss
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan_6' has link connectivity loss
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' has lost the connection
Oct  4 15:55:07 OpenWrt netifd: wan_6 (27979): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" $
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' is now down
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' is disabled
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' is enabled
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' is setting up now
Oct  4 15:55:07 OpenWrt netifd: VLAN 'eth0.2' link is down
Oct  4 15:55:07 OpenWrt netifd: Interface 'wan' has link connectivity loss
Oct  4 15:55:07 OpenWrt netifd: Network device 'eth0' link is down
Oct  4 15:55:07 OpenWrt netifd: wan (14185): Command failed: ubus call network.interface notify_proto { "action": 1, "command": [ "\/usr\/sbin\/pppd", "nodetach", "ipparam",$
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan' is now down
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan' is disabled
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan' is enabled
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan_6' is now down
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan6' is now down
Oct  4 15:55:08 OpenWrt netifd: Interface 'wan6' is disabled
Oct  4 15:57:09 OpenWrt netifd: Network device 'wlan0-2' link is down
Oct  4 15:57:09 OpenWrt netifd: Network device 'wlan0-1' link is down

Neustradamus, 2024-11-03:
@NilsRo: Have you always the problem?

NilsRo, 2024-11-04:
@Neustradamus 👍 

Now with version "OpenWrt 23.05.2 r23630-842932a63d"

Nov  3 10:55:56 OpenWrt pppd[32456]: LCP terminated by peer
Nov  3 10:55:56 OpenWrt pppd[32456]: Connect time 1440.0 minutes.
Nov  3 10:55:56 OpenWrt pppd[32456]: Sent 301632865 bytes, received 970575123 bytes.
Nov  3 10:55:56 OpenWrt odhcp6c[307]: Failed to send RS (Network unreachable)
Nov  3 10:55:56 OpenWrt odhcp6c[307]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Nov  3 10:55:56 OpenWrt netifd: Interface 'wan6' is disabled
Nov  3 10:55:56 OpenWrt netifd: Network device 'pppoe-wan' link is down
Nov  3 10:55:56 OpenWrt netifd: Network alias 'pppoe-wan' link is down
Nov  3 10:55:56 OpenWrt netifd: Interface 'wan6' has link connectivity loss
Nov  3 10:55:56 OpenWrt netifd: wan6 (307): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface"
: "wan6" } (Permission denied)
Nov  3 10:55:56 OpenWrt netifd: Interface 'wan' has lost the connection
Nov  3 10:55:56 OpenWrt dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Nov  3 10:55:56 OpenWrt pppd[32456]: Modem hangup
Nov  3 10:55:56 OpenWrt pppd[32456]: Connection terminated.
Nov  3 10:55:56 OpenWrt pppd[32456]: Connect time 1440.0 minutes.
Nov  3 10:55:56 OpenWrt pppd[32456]: Sent 301632865 bytes, received 970575123 bytes.
Nov  3 10:55:56 OpenWrt pppd[32456]: Exit.
Nov  3 10:55:56 OpenWrt netifd: Interface 'wan' is now down
Nov  3 10:55:56 OpenWrt netifd: Interface 'wan' is setting up now
Nov  3 10:55:56 OpenWrt pppd[2645]: Plugin pppoe.so loaded.
Nov  3 10:55:56 OpenWrt pppd[2645]: PPPoE plugin from pppd 2.4.9
Nov  3 10:55:57 OpenWrt odhcp6c[307]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Nov  3 10:55:57 OpenWrt netifd: Interface 'wan6' is now down