openwrt/openwrt #10895: base-files: functions.sh: config_foreach unexpected behaivour

protectivedad, 2022-10-04:
The '[Configuration in scripts](https://openwrt.org/docs/guide-developer/config-scripting)' documentation says that `return 1` will exit the foreach loop. Which would be a very usefull feature, unfortunately the doicumentation is wrong.  Both `return 0` and `return 1` are threated in the same manner.

Running test.sh:
```
#!/bin/sh

. /lib/functions.sh

parse_interface() {
   echo $1

   return 1
}

config_load network
config_foreach parse_interface interface
```
Gives and output:
```
loopback
lan
bat0
mgmt
not
iot
batmesh
batmesh2
```
The expected output is:
```
loopback
```
 A simple change:
``eval "$___function \"\$section\" \"\$@\""``
to
``eval "$___function \"\$section\" \"\$@\"" || break``
will fix the code to perform as the documentation says. It also has the benefit of speeding up batch files that use the feature.

protectivedad, 2022-10-04:
There is a bug in `/lib/wifi/mac80211.sh`. It uses a `return 1` in a function that should not but does pass the `return 1` on to the `config_foreach` function.

januszdziedzic, 2023-10-03:
patch looks good and follow openwrt doc - allow us to break config_foreach - looks useful

Any reason why not merged?

protectivedad, 2023-10-03:
I'm guessing because it is a small change that effects a lot of code. The problem is that it has been broke for so long that everyone has either coded to avoid it or they just got lucky. There is a lot of "undefined" behaviour in the scripts that use it and when I looked into them I couldn't figure out if they meant to continue after an "error" or they really wanted to ignore the error return.

januszdziedzic, 2023-10-03:
so, I see two options here:
 - use your patch and fix all callbacks (we are using today) to return 0 while this is today behaviour
 - change documentation and answer there is no option to break config_foreach from callback

protectivedad, 2023-10-03:
I agree. It would be ideal to fix things, because I am willing to bet there is are bugs somewhere that nobody knows about due to this unintended behaviour. I don't see the will to do this. It could be done slowly by creating a list of scripts that call the function and as they are updated they are fixed. When they are done my change is released.

guidosarducci, 2023-10-05:
A quick check shows ~500 instances of using `config_foreach` in the main OpenWRT repos. Given that fixing the "short-circuit" feature is an optimization at best, but risks breaking many scripts where authors misunderstand or ignore return management, I tend to lean with "fixing" the documentation to drop the feature.

And since the feature doesn't exist, I doubt any working scripts rely on its presence anyway. But it could still be added in the future...

Do you have access to edit the wiki? Let me know if you'd like me to change it.

protectivedad, 2023-10-05:
> A quick check shows ~500 instances of using `config_foreach` in the main OpenWRT repos. Given that fixing the "short-circuit" feature is an optimization at best, but risks breaking many scripts [...]

Sure, go ahead. With a note saying the loop will process all entries regardless of return. Plus there is no way to differentiate multiple entries that fail/pass.

Well I know some scripts try to use the return value. Some error because of the last command before the script exits errors (wifi did that) which causes problems because it assumes it will continue on. Lots of trouble. It could still be used to root out problem scripts later, I guess.

guidosarducci, 2023-11-21:
Forgive the delay; I've updated the wiki now with the following comment:

> Note that `config_foreach` will iterate through all sections without regard to the callback function's return value.

Thanks for spotting this. Too many dismiss POSIX shell as a programming language and overlook many subtleties, hence Lots of Trouble. :-)