openwrt/openwrt #10898: sysupgrade: ENOSPC

stintel, 2022-10-05:
Running an x86/64 image with CONFIG_NO_STRIP=y, image size 243997724 in a Qemu VM with 512MB RAM, sysupgrade fails silently. Attaching serial console floods ENOSPC. Made some modifications to the sysupgrade code to figure out what is going on.

```diff
diff --git a/package/base-files/files/lib/upgrade/common.sh b/package/base-files/files/lib/upgrade/common.sh
index 5af061f6a4..85040bf13d 100644
--- a/package/base-files/files/lib/upgrade/common.sh
+++ b/package/base-files/files/lib/upgrade/common.sh
@@ -22,6 +22,7 @@ install_file() { # <file> [ <file> ... ]
                [ -f "$file" -a ! -f "$dest" ] && {
                        dir="$(dirname "$dest")"
                        mkdir -p "$dir"
+                       v "Copying $file to $dest"
                        cp "$file" "$dest"
                }
        done
diff --git a/package/base-files/files/lib/upgrade/stage2 b/package/base-files/files/lib/upgrade/stage2
index 5e0d73631b..86b8f121a4 100755
--- a/package/base-files/files/lib/upgrade/stage2
+++ b/package/base-files/files/lib/upgrade/stage2
@@ -36,6 +36,8 @@ switch_to_ramfs() {
        RAMFS_COPY_LOSETUP="$(command -v /usr/sbin/losetup)"
        RAMFS_COPY_LVM="$(command -v lvm)"

+       v "switch_to_ramfs - df: $(df)"
+
        for binary in \
                /bin/busybox /bin/ash /bin/sh /bin/mount /bin/umount    \
                pivot_root mount_root reboot sync kill sleep   
```

Logs from serial console:

```
Wed Oct  5 08:24:24 UTC 2022 upgrade: Sending TERM to remaining processes ...
Wed Oct  5 08:24:24 UTC 2022 upgrade: Sending signal TERM to lldpd (3033)
Wed Oct  5 08:24:24 UTC 2022 upgrade: Sending signal TERM to lldpd (3135)
Wed Oct  5 08:24:28 UTC 2022 upgrade: Sending KILL to remaining processes ...
[  114.642629] stage2 (4440): drop_caches: 3
Wed Oct  5 08:24:35 UTC 2022 upgrade: Switching to ramdisk...
Wed Oct  5 08:24:35 UTC 2022 upgrade: switch_to_ramfs - df: Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root               230400    230400         0 100% /rom
devtmpfs                242784         0    242784   0% /rom/dev
tmpfs                   244780    244100       680 100% /tmp
/dev/loop0              553984    118336    435648  21% /overlay
overlayfs:/overlay      553984    118336    435648  21% /
/dev/sda1               253864      6880    241824   3% /boot
/dev/sda1               253864      6880    241824   3% /boot
tmpfs                      512         0       512   0% /dev
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /bin/busybox to /tmp/root//bin/busybox
cp: write error: No space left on device
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /sbin/mount_root to /tmp/root//sbin/mount_root
cp: write error: No space left on device
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /lib/libfstools.so to /tmp/root//lib/libfstools.so
cp: write error: No space left on device
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /usr/libexec/tar-gnu to /tmp/root//usr/libexec/tar-gnu
cp: write error: No space left on device
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /usr/lib/libacl.so.1.1.2301 to /tmp/root//usr/lib/libacl.so.1.1.2301
cp: write error: No space left on device
Wed Oct  5 08:24:35 UTC 2022 upgrade: Copying /usr/lib/libattr.so.1.1.2501 to /tmp/root//usr/lib/libattr.so.1.1.2501
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/sbin/partx to /tmp/root//usr/sbin/partx
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/lib/libblkid.so.1.1.0 to /tmp/root//usr/lib/libblkid.so.1.1.0
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/lib/libsmartcols.so.1.1.0 to /tmp/root//usr/lib/libsmartcols.so.1.1.0
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/bin/fwtool to /tmp/root//usr/bin/fwtool
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/sbin/grub-bios-setup to /tmp/root//usr/sbin/grub-bios-setup
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /tmp/resolv.conf.d/resolv.conf.auto to /tmp/root//tmp/resolv.conf.d/resolv.conf.auto
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions.sh to /tmp/root//lib/functions.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/caldata.sh to /tmp/root//lib/functions/caldata.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/leds.sh to /tmp/root//lib/functions/leds.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/migrations.sh to /tmp/root//lib/functions/migrations.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/network.sh to /tmp/root//lib/functions/network.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/preinit.sh to /tmp/root//lib/functions/preinit.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/procd.sh to /tmp/root//lib/functions/procd.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/service.sh to /tmp/root//lib/functions/service.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/system.sh to /tmp/root//lib/functions/system.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/functions/uci-defaults.sh to /tmp/root//lib/functions/uci-defaults.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/upgrade/common.sh to /tmp/root//lib/upgrade/common.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/upgrade/fwtool.sh to /tmp/root//lib/upgrade/fwtool.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/upgrade/luci-add-conffiles.sh to /tmp/root//lib/upgrade/luci-add-conffiles.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/upgrade/platform.sh to /tmp/root//lib/upgrade/platform.sh
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /lib/upgrade/do_stage2 to /tmp/root//lib/upgrade/do_stage2
cp: write error: No space left on device
Wed Oct  5 08:24:36 UTC 2022 upgrade: Copying /usr/share/libubox/jshn.sh to /tmp/root//usr/share/libubox/jshn.sh
cp: write error: No space left on device
```

For devices where it is difficult to attach a serial console, this is extremely difficult and frustrating to debug. We should do a minimal space check early, fail sysupgrade and show the user a message so they don't have to guess what is going on.

DragonBluep, 2022-10-05:
The same problem exists in the ramips platform. I guess it's not caused by a lack of space. I have 168 MB free RAM and 90 MB free ROM space, but my firmware size is only 8 MB. It works well one week ago.

```
root@OpenWrt:/tmp# ls -al |grep .bin
-rw-r--r--    1 root     root       7517233 Oct  5 15:31 openwrt-ramips-mt7621-sim_simax1800t-squashfs-sysupgrade.bin
root@OpenWrt:/tmp# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 4608      4608         0 100% /rom
tmpfs                   124944      7596    117348   6% /tmp
/dev/ubi0_1              92436        40     87640   0% /overlay
overlayfs:/overlay       92436        40     87640   0% /
tmpfs                      512         0       512   0% /dev
root@OpenWrt:/tmp# free
              total        used        free      shared  buff/cache   available
Mem:         249888       46992      181192        7596       21704      163672
Swap:             0           0           0
root@OpenWrt:/tmp# sysupgrade openwrt-ramips-mt7621-sim_simax1800t-squashfs-sysupgrade.bin
Wed Oct  5 15:27:37 CST 2022 upgrade: Saving config files...
Wed Oct  5 15:27:37 CST 2022 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
--------------
And then the system crashed.
```

stintel, 2022-10-05:
> The same problem exists in the ramips platform. I guess it's not caused by a lack of space.

If it's not caused by a lack of space, it's a different problem. Please attach a serial console and open a new issue with its output.

DragonBluep, 2022-10-07:
> If it's not caused by a lack of space, it's a different problem. Please attach a serial console and open a new issue with its output.

Sorry I reported the wrong information. After I do a clean build, the error has disappeared. Please ignore my noise.