openwrt/openwrt #10901: Non working binary release for ax820

dohnuts, 2022-10-05:
Hello,

The official release available in the openwrt.org :

https://downloads.openwrt.org/releases/22.03.0/targets/ramips/mt7621/openwrt-22.03.0-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin

( size 5725964 , MD5 (openwrt-22.03.0-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin) = f8c47c48251b59a8c2a7ab5f0edbba15 )

Is not working on the AP:

`
U-Boot 2018.09-svn33 (Mar 12 2021 - 16:47:34 +0800)

CPU:   MediaTek MT7621AT ver 1, eco 3
Clocks: CPU: 880MHz, DDR: 1200MHz, Bus: 220MHz, XTAL: 40MHz
Model: MediaTek MT7621 reference board
DRAM:  256 MiB
Loading Environment from SPI Flash... SF: Detected w25q128bv with page size 256 Bytes, erase size 64 KiB, total 16 MiB
*** Warning - bad CRC, using default environment

In:    uartlite0@1e000c00
Out:   uartlite0@1e000c00
Err:   uartlite0@1e000c00
Net:
Warning: eth@1e100000 (eth0) using random MAC address - 5e:69:c8:f8:cf:5b
eth0: eth@1e100000
Hit any key to stop autoboot:  0
=> bootp
BOOTP broadcast 1
Using eth@1e100000 device
TFTP from server 192.168.2.1; our IP address is 192.168.2.151
Filename 'pxeboot'.
Load address: 0x80010000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #
         2.6 MiB/s
done
Bytes transferred = 5725964 (575f0c hex)
=> bootm
## Booting kernel from Legacy Image at 80010000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.138
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5725900 Bytes = 5.5 MiB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... lzma compressed: uncompress error 1
Must RESET board to recover

Web failsafe UI started
`

dohnuts, 2022-10-06:
Loading at a different avoid erasing information.

```
=> bootp 0x81001000
=> bootm 0x81001000
```

Will boot the https://downloads.openwrt.org/releases/22.03.0/targets/ramips/mt7621/openwrt-22.03.0-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin

robimarko, 2022-10-06:
Do you know what is the RAM base address?
It could be the case that there is not enough RAM and trying to unpack will overwrite part of the loaded image