openwrt/openwrt #10904: ath79: gl-ar300m16: usb regulator not enabled after 22.03

leojrfs, 2022-10-05:
After upgrading to 22.03, there is no power in the USB, and the regulator seems to be off:
```
root@OpenWrt:/# cat /sys/devices/platform/reg_usb_vbus/regulator/regulator.1/state 
disabled
```

[This PR ](https://github.com/openwrt/openwrt/pull/9720)seems to be the culprit, commit: https://github.com/openwrt/openwrt/commit/6e9c814022e41a7cfa4db3b1a41e257fd1515bea

reproduced on GL-AR300M16

john-tho, 2022-10-16:
Only notes, no solution yet, and looking through OpenWrt master, not 22.03:
- I saw that the nand subtarget in ath79 is the only ath79 kernel config that does not have `CONFIG_REGULATOR_FIXED_VOLTAGE=y`, which I think we need that to drive the regulator: https://elixir.bootlin.com/linux/v5.15.74/source/drivers/regulator/fixed.c#L319

- I am not sure if `generic-ehci` driver supports regulators:
  https://github.com/openwrt/openwrt/blob/32c683ddceba8c078faae3706d69eef103736773/target/linux/ath79/dts/qca953x.dtsi#L189
  https://elixir.bootlin.com/linux/v5.15.74/source/drivers/usb/host/ehci-platform.c#L489 does not use `#include <linux/regulator/consumer.h>`
  On an mt7621 device (different USB driver) with a fixed regulator for USB, there are symlinks both ends between the `/sys/class/regulator/regulator.3` _supplier_ device, and the _consumer_ USB device at `/sys/devices/platform/1e1c0000.xhci`, and the regulator gets disabled only when I unbind that USB driver.

- May be able to use `regulator-boot-on` and / or `regulator-always-on` (see: https://elixir.bootlin.com/linux/v5.15.74/source/Documentation/devicetree/bindings/regulator/regulator.yaml#L36)

castelan-jesus, 2022-11-07:
@leojrfs Yup. Really annoying and I don't want to downgrade my GL-AR300M16.

xabolcs, 2022-11-10:
@john-tho are you saying that commit 6e9c814022e41a7cfa4db3b1a41e257fd1515bea just reveals some hidden config problems or it is originally wrong and should be reverted?

john-tho, 2022-11-13:
Finally got back to this… My early ar300m output garbage on uart, and my builds were not loading. Eventually worked out happy loads without `CONFIG_MTD_PARTITIONED_MASTER=y`

We need the regulator-fixed kernel config: https://github.com/john-tho/openwrt/pull/3/commits/be8cf4538cad0f815fe3533a03952f893937a204, and a consumer, userspace control, or `regulator-always-on`.

It looks to me like only [userspace-consumer](https://elixir.bootlin.com/linux/latest/source/drivers/regulator/userspace-consumer.c) can modify the userspace sysfs state, but it does not have devicetree bits [until 6.2][1]

Would like @hauke's input on the best way to proceed:
- use usb-conn-gpio (with a fake id-gpios pin as a driver workaround to force USB HOST mode)
  See demo here: https://github.com/john-tho/openwrt/pull/3, with more details in this commit: https://github.com/john-tho/openwrt/commit/66de4df716b09f2fc2bc1f32dcdc3ab219ebf57f Can open PR against OpenWrt if that looks okay
- forget regulator control, and add the DTS property to reg_usb_vbus: `regulator-always-on`
- [patch ehci-platform.c to be a regulator consumer][0]
- backport the [6.2 regulator-output series][1]

[0]: https://lore.kernel.org/lkml/1519897899-8577-1-git-send-email-amelie.delaunay@st.com/t/#u
[1]: https://lore.kernel.org/all/20221031233704.22575-1-zev@bewilderbeest.net/t/#u

Cheers

castelan-jesus, 2022-11-14:
I have applied this [patch](https://forum.gl-inet.com/t/gl-ar300m16-openwrt-22-03-0-rc5-usb-port-power-off-by-default/23199/2) against openwrt-22.03 branch and it worked for me. I don't know much about it but worked.

john-tho, 2022-11-14:
> I have applied this [patch](https://forum.gl-inet.com/t/gl-ar300m16-openwrt-22-03-0-rc5-usb-port-power-off-by-default/23199/2) against openwrt-22.03 branch and it worked for me. I don't know much about it but worked.

From link:
```diff
diff --git a/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi b/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi
index d64ffa7115..5aaa149940 100644
--- a/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi
+++ b/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi
@@ -43,8 +43,7 @@
                regulator-name = "usb_vbus";
                regulator-min-microvolt = <5000000>;
                regulator-max-microvolt = <5000000>;
-               gpio = <&gpio 2 GPIO_ACTIVE_HIGH>;
-               enable-active-high;
+               gpio = <&gpio 2 GPIO_ACTIVE_LOW>;
        };
 
        leds {
```

That makes gpio think that regulator is enabled by taking the GPIO low.
So when regulator is disabled because it has no consumer, GPIO goes high, and (hardware regulator) turns on…
I think you would be better off using `regulator-always-on` instead, at least DTS would be more accurate.

castelan-jesus, 2022-11-14:
Thank your for the explanation! :D

> From link:
>
> ```diff
> diff --git a/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi b/target/linux/ath79/dts/qca9531_glinet_gl-ar300m.dtsi
> index d64ffa7115..5aaa149940 100644
> --- [...]

EKNYESE, 2022-12-03:
Is it possible for anybody to test, if "regulator-always-on" is the better solution?
Or do we want to propose "gpio = <&gpio 2 GPIO_ACTIVE_LOW>;" to fix this bug asap?

luochongjun, 2022-12-15:
> Only notes, no solution yet, and looking through OpenWrt master, not 22.03:
>
> * I saw that the nand subtarget in ath79 is the only ath79 kernel config that does not have [...]

I submitted a patch on the https://github.com/openwrt/openwrt/pull/11521#issue-1498145190 and explains the reason, Your guess is correct, generic-ehci didn't realize the regulator related code, So it can't be used

xabolcs, 2023-04-03:
Thanks to **hauke**, the fix was backported to `openwrt-22.03`: 70d2e8b4db0c5c9f1a2d40827928bef8083b19d6. :pray: 
Itt will be included in the next OpenWrt 22.03 release: 22.03.4

sifive-eblot, 2023-04-07:
Any target date for releasing 22.03.4? I'd prefer not to use a nightly build to run a router but I really need USB support. 
Thanks.

pexcn, 2023-10-12:
GL-INET 6416 has the same problem. (openwrt 23.05-rc4)