openwrt/openwrt #10907: 464xlat: causes log spam

patrakov, 2022-10-06:
If the 464xlat package is installed, but some uplink does not need it, then logs are spammed with this:

```
Thu Oct  6 14:30:15 2022 daemon.notice netifd: Interface 'lte_6_4' is now down
Thu Oct  6 14:30:15 2022 daemon.notice netifd: Interface 'lte_6_4' is setting up now
Thu Oct  6 14:30:17 2022 daemon.notice netifd: Interface 'wan_6_4' is now down
Thu Oct  6 14:30:17 2022 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Thu Oct  6 14:30:18 2022 daemon.notice netifd: Interface 'lte_6_4' is now down
Thu Oct  6 14:30:18 2022 daemon.notice netifd: Interface 'lte_6_4' is setting up now
Thu Oct  6 14:30:20 2022 daemon.notice netifd: Interface 'wan_6_4' is now down
Thu Oct  6 14:30:20 2022 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Thu Oct  6 14:30:21 2022 daemon.notice netifd: Interface 'lte_6_4' is now down
Thu Oct  6 14:30:21 2022 daemon.notice netifd: Interface 'lte_6_4' is setting up now
Thu Oct  6 14:30:23 2022 daemon.notice netifd: Interface 'wan_6_4' is now down
```

This log spam means that any troubleshooting based on historical logs becomes impossible, because they are quickly rotated out.

CodeFetch, 2022-10-06:
@patrakov Why do you think it has anything to do with 464xlat? The 464xlat does not do anything on its own. It's just a kernel module that can create two interfaces where the lower interface receives IPv4 and transmits the packets encapsulated as IPv6 with the specific CLAT-Prefix on the upper interface and vice versa decapsulates packets to IPv4.
The netifd notice you are seeing here likely results from a bad configuration of something like multiwan.

Leo-PL, 2022-10-14:
@CodeFetch I've seen this way too many times, actually - the default configuration or autodetection seems broken. When I manually set the prefix to typical and standardized 64:ff9b::/96 in networks which support it, it starts to work.
@patrakov as a workaround, you may try adding 'option iface_464xlat '0'` to its parent interface, not sure if it'll work with sub-subinterface like on QMI. If that's QMI, then see https://github.com/openwrt/openwrt/pull/10912 - it might help you if you have dual-stack connection.

Cineray, 2023-05-06:
> option iface_464xlat '0'

remove ds-lite and luci-proto-ipv6 packages can resolve this.

Leo-PL, 2023-05-06:
> > option iface_464xlat '0'
> 
> remove ds-lite and luci-proto-ipv6 packages can resolve this.

No, it won't. 464xlat is separate protocol from DS-lite type and will still cause log spam. Removing luci-proto-ipv6 will only hide the problem from GUI.

Cineray, 2023-05-06:
@Leo-PL 

> Removing luci-proto-ipv6 will only hide the problem from GUI.

Is that true? add option iface_464xlat '0' and option dhcpv6 '0' did not help, the error comes up every 3 sec.
the system log would not show the error after remove these two packeages, where can i find log spam if it not resolved?

Leo-PL, 2023-05-06:
Removing `464xlat` package might have actually helped, because it will remove its protocol handler. Are we really discussing 464xlat here, or your configuration is totally different from OP's? Please show your configuration, otherwise this discussion makes no sense.

DS-lite is an entirely different thing. Also, if using QMI or any other protocol that uses DHCPv6 for sub-interfaces like NCM - you haven't stated any details about your configuration, so this is wild guess -  it isn't possible to pass `iface_464xlat '0'` down to a sub-interface. At least for QMI, there is a patch that could fix that: https://github.com/openwrt/openwrt/pull/10913/commits/53d8fb1958c9ca56e7eb95f0fa2d1f9abc8fb6a9

patrakov, 2023-05-06:
OP here.

While passing down `iface_464xlat '0'` down would help, the feature request here is that it should not be necessary.

1. netifd should not attempt to activate an implicit 464xlat if there is already an IPv4 address on the same physical interface.
2. netifd should not retry endlessly, unless something changes in the network, if activating 464xlat fails repeatedly.

Leo-PL, 2023-05-06:
> OP here.
> 
> While passing down `iface_464xlat '0'` down would help, the feature request here is that it should not be necessary.
> 
>     1. netifd should not attempt to activate an implicit 464xlat if there is already an IPv4 address on the same physical interface.
Agreed. the patch I mentioned in previous post does that, but only for QMI. NCM and MBIM should have the same behaviour.

> 
>     2. netifd should not retry endlessly, unless something changes in the network, if activating 464xlat fails repeatedly.

Agreed. With this one, the autodetection of v6 prefix, or handling of the default :64:ff9b::/96 prefix could be improved. Strange thing is, that 464xlat never actually worked for me unless I explicitly specified this very prefix, which should be the default if one provided by DHCPv6 options doesn't state otherwise.