openwrt/openwrt #10929: Request to add default firewall rule of WAN2WAN forwarding for (strongswan) policy based vpn

sunshinejnjn, 2022-10-08:
Due to the default block on forwarding from the WAN zone and default installing of Virtual IP on "outbound" interface (the WAN) by strongswan IPSec VPN using swanctl, which is known as a policy-based VPN, client traffic via IPSec tunnel to the wan side would arrive from the WAN zone on the wan interface and then got blocked on forwarding by wan_forwarding chain of "inet fw4" nft table, and won't be SNATted/MASQUERADed correctly by srcnat rules at postrouting.

A default allow-wan2wan forwarding rule would solve this, something like the following in /etc/config/firewall

config rule
        option name 'AllowWAN2WAN'
        list proto 'all'
        option src 'wan'
        option dest 'wan'
        option target 'ACCEPT'

stintel, 2022-12-14:
I'm tempted to NAK this. If needed, the user can add this themselves.

Spudz76, 2022-12-14:
seems like it should make a virtual interface like OpenVPN does, and then that would be its own zone such as "VPN", instead of gangbanging IP addresses on the wan device and causing the problem in the first place.

stintel, 2022-12-14:
Closing this. Like I said before, if it's needed, the user can configure it themselves. I'll give a hint for an alternative solution: use an xfrm interface and link that to a firewall zone.

closed 2022-12-14 by stintel