openwrt/openwrt #10944: wolfssl: wolfssl 5.5.1 compilation fails on WOLFSSL_HAS_DEVCRYPTO_*

zokl, 2022-10-12:
Hello @cotequeiroz @ynezz.

Compilation of the wolfssl 5.5.1 fails if devcrypto support is enabled (WOLFSSL_HAS_DEVCRYPTO_AES etc.).

I have tried it on OpenWRT 21.02.

Compilation log follows ...

```
-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-i
nitializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result
 -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/src/port/devcrypto/src_libwolfssl_la-wc_devcrypto.lo -MD -MP -MF wolfcrypt/src/port/devcrypto/.deps/src_libwolfssl_la-wc_devcrypto.Tpo -c wolfcrypt/src/port/devcrypt
o/wc_devcrypto.c  -fPIC -DPIC -o wolfcrypt/src/port/devcrypto/.libs/src_libwolfssl_la-wc_devcrypto.o
In file included from ./wolfssl/wolfcrypt/wc_port.h:572,
                 from ./wolfssl/wolfcrypt/types.h:35,
                 from ./wolfssl/wolfcrypt/error-crypt.h:34,
                 from wolfcrypt/src/port/devcrypto/wc_devcrypto.c:33:
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h: In function 'snprintf':
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:101:2: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
  return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
  ^~~~~~
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h: In function 'sprintf':
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:110:3: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
   __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
   ^~~
/mnt/data/ftester/ftester_production/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify/stdio.h:114:3: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
   __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
   ^~~
wolfcrypt/src/port/devcrypto/wc_devcrypto.c: In function 'wc_DevCryptoCreate':
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:36: error: 'u_int8_t' undeclared (first use in this function); did you mean 'uint8_t'?
             ctx->sess.mackey    = (u_int8_t*)key;
                                    ^~~~~~~~
                                    uint8_t
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:36: note: each undeclared identifier is reported only once for each function it appears in
wolfcrypt/src/port/devcrypto/wc_devcrypto.c:125:45: error: expected expression before ')' token
             ctx->sess.mackey    = (u_int8_t*)key;
                                             ^
make[5]: *** [Makefile:6300: wolfcrypt/src/port/devcrypto/src_libwolfssl_la-wc_devcrypto.lo] Error 1
make[5]: Leaving directory '/mnt/data/ftester/ftester_production/build_dir/target-x86_64_musl/wolfssl-5.5.1-stable'
make[4]: *** [Makefile:6884: all-recursive] Error 1
make[4]: Leaving directory '/mnt/data/ftester/ftester_production/build_dir/target-x86_64_musl/wolfssl-5.5.1-stable'
make[3]: *** [Makefile:3984: all] Error 2
make[3]: Leaving directory '/mnt/data/ftester/ftester_production/build_dir/target-x86_64_musl/wolfssl-5.5.1-stable'
make[2]: *** [Makefile:130: /mnt/data/ftester/ftester_production/build_dir/target-x86_64_musl/wolfssl-5.5.1-stable/.built] Error 2
make[2]: Leaving directory '/mnt/data/ftester/ftester_production/package/libs/wolfssl'
time: package/libs/wolfssl/compile#54.35#13.13#66.63
    ERROR: package/libs/wolfssl failed to build.

```

ynezz, 2022-10-12:
It looks like [upstream bug](https://github.com/wolfSSL/wolfssl/commit/35d40f263ec3e0b4db7718d2987a49bcd9d0827e#diff-869644daadc85244298e10c42587a7f5685ff291ef4deebe8e415f0531e32f45R125) and should be reported there as well. We likely need following fix/patch:

```
diff --git a/wolfcrypt/src/port/devcrypto/wc_devcrypto.c b/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
index 2a21d676c899..5c8d99217910 100644
--- a/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
+++ b/wolfcrypt/src/port/devcrypto/wc_devcrypto.c
@@ -122,7 +122,7 @@ int wc_DevCryptoCreate(WC_CRYPTODEV* ctx, int type, byte* key, word32 keySz)
         case CRYPTO_SHA2_512_HMAC:
             ctx->sess.cipher = 0;
             ctx->sess.mac    = type;
-            ctx->sess.mackey    = (u_int8_t*)key;
+            ctx->sess.mackey    = (uint8_t*)key;
             ctx->sess.mackeylen = keySz;
             break;
```

zokl, 2022-10-12:
Hi, the patch works. I report his bug to the upstream of wolfssl.

embhorn, 2022-10-12:
Fix in wolfSSL PR https://github.com/wolfSSL/wolfssl/pull/5695

commit 17169150 (2022-12-11): wolfssl: fix build with /dev/crypto

Backport upstream patch to fix build error when
/dev/crypto enabled.

https://github.com/wolfSSL/wolfssl/commit/dc9f46a3be00b5e82684a158605189d1278e324c

Fixes: #10944
PR #11455 (2022-12-11): wolfssl: fix build with /dev/crypto
Bakcport upstream patch to fix build error when
/dev/crypto enabled.

https://github.com/wolfSSL/wolfssl/commit/dc9f46a3be00b5e82684a158605189d1278e324c

Fixes: #10944
commit 39dcfcd9 (2022-12-21): wolfssl: fix build with /dev/crypto

Backport upstream patch to fix build error when
/dev/crypto enabled.

https://github.com/wolfSSL/wolfssl/commit/dc9f46a3be00b5e82684a158605189d1278e324c

Fixes: #10944

(cherry picked from commit 171691500eca0737c59d4fff50578b74a90583be)
closed 2022-12-11 by openwrt-bot