openwrt/openwrt #10947: No packet seen with tcpdump on DSA port with Mt7621 / Mt7530 switch

DRTEK, 2022-10-12:
Hello ,
i am wondering if this has anything to do with mt7530 dma or some sort of hardware offloading in place.
i tried with openwrt 21.02 as well as 22.03 got the same outcome 
because cpu wont see any packages ebtables wont work either.
the other note that i need to mention is that with mt7615e dispatching feature same outcome happened with wifi interfaces as well
are there any way to disable this dispatching feature ?
thank you in advance

quarkysg, 2022-10-13:
From what I understand, with hardware offloading enabled, received packets will not get forwarded to the CPU.  It is re-routed by the PPE to the target destination port and the CPU will see zero load for such packets.

DRTEK, 2022-10-13:
You explained it better 👍

quarkysg, 2022-10-13:
@DRTEK are you using fw4 for the 22.03 firmware?  From what I understand, nftables should incorporate ebtables functionalities, i.e. making ebtables redundant.  Hardware flow-offload works in conjunction with nft rules, as opposed to previously for fw3 where it relies on iptables rules only.  So if the nft rules (i.e. fw4) allows connections establishment, only then will the PPE take over the connection.

If you really need ebtables to work and still need to use iptables (i.e. fw3) you will have to disable all forms of offloading.

DRTEK, 2022-10-13:
Hello,
I beleive nftables does not have brouting feature for that reason I am using fw3 iptables  / ebtables. how do we disable offloading ?
Thank you

champtar, 2022-10-13:
> I beleive nftables does not have brouting feature for that reason I am using fw3 iptables / ebtables.

nftables can definitly do stateful L2 filtering, you just need to install kmod-nft-bridge package and use the bridge table.
I switch a package of mine some months ago from iptables / ebtables to nftables, maybe it can help you looking at the diff:
https://github.com/openwrt/packages/commit/66b7c19992688b924d2ecbbbc20781b32a82452f
https://github.com/nccgroup/phantap/commit/b066ce2c2bb21038958a117b3b67413e9a0ea0a3

DRTEK, 2022-10-13:
@champtar  thank you so much for sharing the info,
i will give it a try 
@quarkysg   it is essential for me to disable offloading  because  my application relies heavily on libpcap.
i tried ethtool but no luck so far 
can you please show me how to do offloading ? do i need to change the mt7530 code files ?
thank you all

quarkysg, 2022-10-14:
@DRTEK offloading should be turn-off by default if you install OpenWrt with default config.  S/W and H/W offloading for the mt7621 routers are configured via the Firewall Luci page if you have Luci installed.  Otherwise it has to be specifically turned on in the /etc/config/firewall configuration file.

DRTEK, 2022-10-16:
unfortunately it did not work,
however mlilja01 was having the same problem and he made a  patch for it. 
[https://forum.openwrt.org/t/bridge-wan-and-lan-and-still-see-packets-in-the-cpu-until-flowoffload/113424/8?u=drtek](url)

mlilja01, 2022-10-18:
The issue is not flow offload it is 'switch offload'. The 'dsa' implementation introduced in 21.03 automatically enable L2 Mac learning and also offload port-to-port switching without the user being able to prevent it. In my case I wanted to control, via flow offloading what and when things should be offloaded hence my patch is disabling MAC learning in the switch. The patch is not pretty and should likely be controllable from userspace if MAC learning offload should be on/off?