openwrt/openwrt #10948: IPv4 address not assigned when connected with AP

atyouspace, 2022-10-13:
My openwrt board has connected with ap router, however, no IPv4 address is assigned.

wlan0     Link encap:Ethernet  HWaddr 34:82:4A:0B:B7:D8
          inet6 addr: fe80::3682:4aff:fe0b:b7d8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:487 errors:0 dropped:0 overruns:0 frame:0
          TX packets:37 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:121090 (118.2 KiB)  TX bytes:4587 (4.4 KiB)

In addition, I connect my PC with my openwrt board directly using a network cable, however, no IPv4 address is assigned to my PC.

eth1      Link encap:Ethernet  HWaddr 00:A0:C9:00:00:02
          inet6 addr: fe80::2a0:c9ff:fe00:2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:154 errors:0 dropped:0 overruns:0 frame:0
          TX packets:576 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:49588 (48.4 KiB)  TX bytes:195424 (190.8 KiB)
          Memory:1100000-117ffff

Wahjong1974, 2022-10-13:
I have the exact same issue with a wired connection.

Hardware: R7800 (netgear)
Software: OpenWrt 22.03.1

Spudz76, 2023-02-27:
wlan and eth should be within br-lan bridge otherwise broadcast stuff from the rest of the network won't make it across.