openwrt/openwrt #10950: OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions

Matthaiks, 2022-10-13:
After upgrading my Totolink X5000R from 22.03.0 to 22.03.1, I'm getting this kernel error on bootup:
```
Thu Oct 13 16:44:37 2022 kern.notice kernel: [    0.657173] 4 fixed-partitions partitions found on MTD device spi0.0
Thu Oct 13 16:44:37 2022 kern.err kernel: [    0.663573] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Thu Oct 13 16:44:37 2022 kern.err kernel: [    0.670873] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Thu Oct 13 16:44:37 2022 kern.err kernel: [    0.679062] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Thu Oct 13 16:44:37 2022 kern.err kernel: [    0.686407] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Thu Oct 13 16:44:37 2022 kern.notice kernel: [    0.693986] Creating 4 MTD partitions on "spi0.0":
```

I guess this might be due to the kernel changes in 22.03.1:

>     Backport mtd dynamic partition patch
>     Fix possible mtd NULL pointer dereference

The router seems working fine. Is this error something to worry about?

roqueeee, 2022-10-13:
Chiming in to report the same kernel errors for TP-Link EAP245 v3:
```
...
Thu Oct 13 16:47:40 2022 kern.notice kernel: [    0.383354] 10 fixed-partitions partitions found on MTD device spi0.0
Thu Oct 13 16:47:40 2022 kern.err kernel: [    0.390069] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Oct 13 16:47:40 2022 kern.err kernel: [    0.396918] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Oct 13 16:47:40 2022 kern.err kernel: [    0.404409] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Oct 13 16:47:40 2022 kern.err kernel: [    0.411330] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Oct 13 16:47:40 2022 kern.notice kernel: [    0.418407] Creating 10 MTD partitions on "spi0.0":
...
```

It appears this issue is already being discussed in #10232.

My device seems to be working fine as well. Some clarification would be nice though.

neheb, 2022-10-16:
ping @Ansuel

john-tho, 2022-10-16:
`OF: Bad cell count for`…`/partitions` Should be harmless.
See this mail on the device tree list for an explanation: https://lore.kernel.org/all/2fff1d95-c257-2756-ff63-5efe7b3c528c@gmail.com/

rmilecki, 2023-07-29:
It's a harmless error as explained in the `of_platform_populate() for address-less nodes (OF: Bad cell count for ...)` e-mail thread discussion.

closed 2023-07-29 by rmilecki

namiltd, 2024-05-14:
Or maybe it was worth adding:
```
#address-cells = <1>;
#size-cells = <1>;
```
for subsequent dts(i) files containing spi nor flash (flash@)?
For some routers this is already set and works, it turns off this unnecessary message.

raenye, 2024-05-17:
See also e74219c795a03e8b9adccbb5e6a20d4566872f8a.

neheb, 2024-05-17:
```
this is not always necessary for some nodes
in devicetree that are undergoing address translation.
```

so there's a different way to silence this?

heis2201, 2025-02-03:
I am experiencing the same issue on a TP EAP245 v1.

`OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions`