openwrt/openwrt #10966: Linux-ramips_mt7621 kernel errors

BarnOwlDown, 2022-10-14:
Hi!
I've some problems (glitchy connection and, sometimes, low speed) with WiFi 5 on my Cudy WR1300 and I found these errors in kernel log. I hope they will be useful for someone.

```
[126660.802158] ------------[ cut here ]------------
[126660.806930] WARNING: CPU: 0 PID: 1636 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-08-26-5ec78e1e/mt7603/mac.c:208 0x82794864 [mt7603e@d600c99d+0x96c0]
[126660.821607] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact
[126660.822526]  ledtrig_usbport ifb sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[126660.932591] CPU: 0 PID: 1636 Comm: hostapd Not tainted 5.10.138 #0
[126660.938839] Stack : 00000000 807200d8 809a0000 80000000 00000000 00000000 00000000 00000000
[126660.947272]         00000000 00000000 00000000 00000000 00000000 00000001 82a67928 d59d9fd3
[126660.955702]         82a679c0 00000000 00000000 82a677d0 00000038 80399864 ffffffea 00000000
[126660.964132]         82a677dc 00000170 807c2868 ffffffff 82a67908 806ef0cc 80000000 806f0000
[126660.972561]         00000009 00000000 00003a98 80000000 00000000 80403190 00000000 80980000
[126660.980997]         ...
[126660.983521] Call Trace:
[126660.983539] [<80399864>] 0x80399864
[126660.989608] [<80403190>] 0x80403190
[126660.993166] [<80007b08>] 0x80007b08
[126660.996723] [<80007b10>] 0x80007b10
[126661.000278] [<8037e984>] 0x8037e984
[126661.003862] [<82794864>] 0x82794864 [mt7603e@d600c99d+0x96c0]
[126661.009666] [<8007fa1c>] 0x8007fa1c
[126661.013231] [<82794864>] 0x82794864 [mt7603e@d600c99d+0x96c0]
[126661.019033] [<8002ff38>] 0x8002ff38
[126661.022602] [<82794864>] 0x82794864 [mt7603e@d600c99d+0x96c0]
[126661.028407] [<80030000>] 0x80030000
[126661.032008] [<82794864>] 0x82794864 [mt7603e@d600c99d+0x96c0]
[126661.037835] [<82791ac8>] 0x82791ac8 [mt7603e@d600c99d+0x96c0]
[126661.043658] [<825f3018>] 0x825f3018 [mt76@10352f93+0xafc0]
[126661.049235] [<825f3188>] 0x825f3188 [mt76@10352f93+0xafc0]
[126661.054788] [<82707dac>] 0x82707dac [mac80211@6ad028f5+0x81220]
[126661.060780] [<82709770>] 0x82709770 [mac80211@6ad028f5+0x81220]
[126661.066758] [<80099b64>] 0x80099b64
[126661.070327] [<82709b10>] 0x82709b10 [mac80211@6ad028f5+0x81220]
[126661.076314] [<82693cac>] 0x82693cac [cfg80211@a74cabdb+0x45f40]
[126661.082293] [<80573108>] 0x80573108
[126661.085850] [<80571bb0>] 0x80571bb0
[126661.089407] [<804dcd94>] 0x804dcd94
[126661.092964] [<80571220>] 0x80571220
[126661.096534] [<82693bec>] 0x82693bec [cfg80211@a74cabdb+0x45f40]
[126661.102523] [<80572fb4>] 0x80572fb4
[126661.106080] [<80571ef4>] 0x80571ef4
[126661.109640] [<8057285c>] 0x8057285c
[126661.113198] [<805714c0>] 0x805714c0
[126661.116754] [<801a11d8>] 0x801a11d8
[126661.120308] [<80203294>] 0x80203294
[126661.123867] [<80571958>] 0x80571958
[126661.127424] [<80075630>] 0x80075630
[126661.130978] [<804df91c>] 0x804df91c
[126661.134539] [<805715f4>] 0x805715f4
[126661.138095] [<804d3158>] 0x804d3158
[126661.141649] [<80322f50>] 0x80322f50
[126661.145211] [<804d4fe8>] 0x804d4fe8
[126661.148778] [<8062ea00>] 0x8062ea00
[126661.152343] [<804d449c>] 0x804d449c
[126661.155904] [<804d5570>] 0x804d5570
[126661.159463] [<804d51f8>] 0x804d51f8
[126661.163021] [<804d1e58>] 0x804d1e58
[126661.166576] [<8020226c>] 0x8020226c
[126661.170135] [<80014578>] 0x80014578
[126661.173700] 
[126661.175598] ---[ end trace 1e2cde223e6847d2 ]---
[132233.661500] ------------[ cut here ]------------
[132233.666330] WARNING: CPU: 2 PID: 892 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-08-26-5ec78e1e/mt76x02_mac.c:850 0x827d3298 [mt76x02_lib@3d5989c1+0x9e60]
[132233.681566] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact
[132233.682121]  ledtrig_usbport ifb sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[132233.792085] CPU: 2 PID: 892 Comm: napi/phy1-9 Tainted: G        W         5.10.138 #0
[132233.799978] Stack : 00000000 807200d8 809a0000 00000028 00000000 00000000 00000000 00000000
[132233.808418]         00000000 00000000 00000000 00000000 00000000 00000001 82637c98 d59d9fd3
[132233.816858]         82637d30 00000000 00000000 82637b40 00000038 80399864 ffffffea 00000000
[132233.825317]         82637b4c 000001ab 807c2868 ffffffff 82637c78 806ef0cc 00000000 806f0000
[132233.833754]         00000009 00000000 00000000 00000028 00000000 80403190 00000008 80980008
[132233.842185]         ...
[132233.844713] Call Trace:
[132233.844789] [<80399864>] 0x80399864
[132233.850870] [<80403190>] 0x80403190
[132233.854449] [<80007b08>] 0x80007b08
[132233.858009] [<80007b10>] 0x80007b10
[132233.861568] [<8037e984>] 0x8037e984
[132233.865155] [<827d3298>] 0x827d3298 [mt76x02_lib@3d5989c1+0x9e60]
[132233.871309] [<8007fa1c>] 0x8007fa1c
[132233.874877] [<827d3298>] 0x827d3298 [mt76x02_lib@3d5989c1+0x9e60]
[132233.881028] [<8002ff38>] 0x8002ff38
[132233.884605] [<827d3298>] 0x827d3298 [mt76x02_lib@3d5989c1+0x9e60]
[132233.890759] [<80030000>] 0x80030000
[132233.894350] [<827d3298>] 0x827d3298 [mt76x02_lib@3d5989c1+0x9e60]
[132233.900526] [<827d6030>] 0x827d6030 [mt76x02_lib@3d5989c1+0x9e60]
[132233.906691] [<804e15d8>] 0x804e15d8
[132233.910291] [<825f1ab0>] 0x825f1ab0 [mt76@10352f93+0xafc0]
[132233.915841] [<80060714>] 0x80060714
[132233.919430] [<80504194>] 0x80504194
[132233.923008] [<806d2290>] 0x806d2290
[132233.926581] [<80504358>] 0x80504358
[132233.930140] [<80504374>] 0x80504374
[132233.933711] [<800509a8>] 0x800509a8
[132233.937289] [<805042b4>] 0x805042b4
[132233.940843] [<80050c10>] 0x80050c10
[132233.944405] [<80050ad4>] 0x80050ad4
[132233.947975] [<80050ad4>] 0x80050ad4
[132233.951535] [<80050ad4>] 0x80050ad4
[132233.955094] [<80002f38>] 0x80002f38
[132233.958660] 
[132233.960504] ---[ end trace 1e2cde223e6847d3 ]---
```

PS: I'm on OpenWRT 22.03.0 with 5.10.138 kernel

neheb, 2022-10-15:
ping @nbd168

neheb, 2022-10-15:
btw, version .1 is out.

BarnOwlDown, 2022-10-16:
Upgraded to 22.03.2. I'll test it in these days!

lynxis, 2022-11-16:
I can also confirm this bug on the tplink,archer-c6-v3 (mt7621) with the 2.4ghz interface.
```
[444807.705185] ------------[ cut here ]------------
[444807.709968] WARNING: CPU: 2 PID: 1366 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-09-06-d7054646/mt7603/mac.c:208 0x82694864 [mt7603e@5c412bdb+0x96e0]
[444807.724648] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[444807.789724] CPU: 2 PID: 1366 Comm: hostapd Not tainted 5.10.146 #0
[444807.796204] Stack : 00000000 8072637c 809b0000 80000000 00000000 00000000 00000000 00000000
[444807.804649]         00000000 00000000 00000000 00000000 00000000 00000001 82a41928 63afd353
[444807.813090]         82a419c0 00000000 00000000 82a417d0 00000038 80399bc4 ffffffea 00000000
[444807.821533]         82a417dc 000001fd 807ca868 ffffffff 82a41908 806f5114 80000000 806f0000
[444807.829973]         00000009 00000002 00003a98 80000000 00000000 80403500 00000008 80990008
[444807.838414]         ...
[444807.840941] Call Trace:
[444807.840990] [<80399bc4>] 0x80399bc4
[444807.847102] [<80403500>] 0x80403500
[444807.850671] [<80007b08>] 0x80007b08
[444807.854226] [<80007b10>] 0x80007b10
[444807.857786] [<8037ecd4>] 0x8037ecd4
[444807.861371] [<82694864>] 0x82694864 [mt7603e@5c412bdb+0x96e0]
[444807.867187] [<8007fb60>] 0x8007fb60
[444807.870769] [<82694864>] 0x82694864 [mt7603e@5c412bdb+0x96e0]
[444807.876587] [<8003007c>] 0x8003007c
[444807.880185] [<82694864>] 0x82694864 [mt7603e@5c412bdb+0x96e0]
[444807.886014] [<80030144>] 0x80030144
[444807.889575] [<806db4a8>] 0x806db4a8
[444807.893149] [<82694864>] 0x82694864 [mt7603e@5c412bdb+0x96e0]
[444807.898961] [<806db544>] 0x806db544
[444807.902539] [<82691ac8>] 0x82691ac8 [mt7603e@5c412bdb+0x96e0]
[444807.908374] [<826139bc>] 0x826139bc [mac80211@c60ebcda+0x81240]
[444807.914376] [<825632e8>] 0x825632e8 [mt76@ba2876f0+0xafe0]
[444807.919942] [<80099ca8>] 0x80099ca8
[444807.923500] [<80091cac>] 0x80091cac
[444807.927090] [<82563458>] 0x82563458 [mt76@ba2876f0+0xafe0]
[444807.932669] [<82609780>] 0x82609780 [mac80211@c60ebcda+0x81240]
[444807.938650] [<80099ca8>] 0x80099ca8
[444807.942218] [<82609b20>] 0x82609b20 [mac80211@c60ebcda+0x81240]
[444807.948219] [<82513d2c>] 0x82513d2c [cfg80211@45c5f770+0x45fe0]
[444807.954204] [<80579304>] 0x80579304
[444807.957768] [<80577dac>] 0x80577dac
[444807.961335] [<804e2e58>] 0x804e2e58
[444807.964890] [<8057741c>] 0x8057741c
[444807.968479] [<82513c6c>] 0x82513c6c [cfg80211@45c5f770+0x45fe0]
[444807.974484] [<805791b0>] 0x805791b0
[444807.978047] [<805780f0>] 0x805780f0
[444807.981616] [<80578a58>] 0x80578a58
[444807.985183] [<805776bc>] 0x805776bc
[444807.988748] [<801a1300>] 0x801a1300
[444807.992334] [<80577b54>] 0x80577b54
[444807.995906] [<80182b28>] 0x80182b28
[444807.999465] [<80021694>] 0x80021694
[444808.003032] [<805777f0>] 0x805777f0
[444808.006590] [<804d9214>] 0x804d9214
[444808.010161] [<803232a0>] 0x803232a0
[444808.013734] [<804db0a4>] 0x804db0a4
[444808.017314] [<802eeef0>] 0x802eeef0
[444808.020875] [<80182f98>] 0x80182f98
[444808.024464] [<804db2b4>] 0x804db2b4
[444808.028026] [<804d7f14>] 0x804d7f14
[444808.031611] [<80014578>] 0x80014578
[444808.035179] 
[444808.037187] ---[ end trace 9714b98c58fa71ea ]---
root@archerc6:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```