openwrt/openwrt #10968: ramips Belkin F9K1109 fails to boot due to large kernel

nicefile, 2022-10-14:
wiki page https://openwrt.org/toh/belkin/f9k1109_v1 contains information about boot limitation  .I can confirm that  with 21.02.x and 22.03.x  . It boot 19.07.10 !
As an workaround. I've build OpenWrt SNAPSHOT r20705-1722e23ffc that boot my device
recipe :
```
rt3883.mk
+ $(Device/uimage-lzma-loader)
+ KERNEL := kernel-bin | append-dtb | lzma -d10 | uImage lzma
```
and strip kernel exports

xabolcs, 2022-10-14:
Does it work without the `KERNEL := ...` line? The `$(Device/uimage-lzma-loader)` line isn't enough?

nicefile, 2022-10-17:
@xabolcs Of course it's not . My modification look like below now . Even that without kernel strip booting device is impossible
```
define Device/belkin_f9k1109v1
$(Device/uimage-lzma-loader)
  SOC := rt3883
  BLOCKSIZE := 64k
  DEVICE_VENDOR := Belkin
  DEVICE_MODEL := F9K1109
  DEVICE_VARIANT := Version 1.0
  DEVICE_PACKAGES := kmod-usb-ohci kmod-usb2 kmod-usb-ledtrig-usbport
  IMAGE_SIZE := 7808k
  KERNEL := kernel-bin | append-dtb | lzma -d10 | uImage lzma
  # Stock firmware checks for this uImage image name during upload.
  UIMAGE_NAME := N750F9K1103VB
endef
TARGET_DEVICES += belkin_f9k1109v1
```

xabolcs, 2022-10-17:
Do you have serial access to your device?
With that, you could easily verify that the `$(Device/uimage-lzma-loader)` addition is works or not.

If it not used, then the log is like below:
```
    ## Transferring control to Linux (at address 80000000) ...
    ## Giving linux memsize in MB, 64

    Starting kernel ...

    [    0.000000] Linux version 5.4.188 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16554-1d4dea6d4f)) #0 Sat Apr 16 12:59:34 2022
    [    0.000000] SoC Type: Ralink RT3883 ver:1 eco:5
    [    0.000000] printk: bootconsolde [early0] enabled
    [    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
    [    0.000000] MIPS: machine is D-Link DIR-645
    [    0.000000] Initrd not found or empty - disabling initrd
```

If it's used, then the serial log is like below. Note the `OpenWrt kernel loader for MIPS based SoC` string:
```
    ## Transferring control to Linux (at address 80000000) ...
    ## Giving linux memsize in MB, 64

    Starting kernel ...

    OpenWrt kernel loader for MIPS based SoC
    Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
    Decompressing kernel... done!
    Starting kernel at 80000000...

    [    0.000000] Linux version 5.10.144 (xabolcs@ut2004) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20774+2-b71affaf8b) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 Tue Sep 27 23:02:30 2022
    [    0.000000] SoC Type: Ralink RT3883 ver:1 eco:5
    [    0.000000] printk: bootconsole [early0] enabled
    [    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
    [    0.000000] MIPS: machine is D-Link DIR-645
```

nicefile, 2022-10-17:
Yes It works. That's whole point .lzma loader work but can We optimize this entry for size (other way than I've did)
`KERNEL := kernel-bin | append-dtb | lzma -d16 | uImage lzma`
?

xabolcs, 2022-10-17:
No dictionary magic is needed for that `uimage-lzma-loader`! See commit 1e5d014ba237cf47092c41cf3657ec64d3b99b41.

Could you post serial log about boot? I think ` KERNEL := kernel-bin | append-dtb | lzma -d16 | uImage lzma` is disabling `uimage-lzma-loader`.
Take issue #10634 and it's [support topic](https://forum.openwrt.org/t/v22-03-0-release-does-not-work-on-d-link-dir-645/136435) as example!

nicefile, 2022-10-17:
@xabolcs  `uimage-lzma-loader` did not work with my setup as you predicted :

Now my old mod bootlog:
```
ARC Uboot:1.7.4 (Sep 14 2012 - 15:09:21)

Board: Ralink APSoC DRAM:  64 MB
relocate_code Pointer at: 83fa0000
spi_wait_nsec: 30
spi device id: c2 20 17 c2 20 (2017c220)
find flash: MX25L6405D
..============================================
Ralink UBoot Version: 3.5.2.0
--------------------------------------------
ASIC 3883_MP (MAC to VITESSE Mode)
DRAM_CONF_FROM: Boot-Strapping
DRAM_TYPE: DDR2
DRAM_SIZE: 512 Mbits
DRAM_WIDTH: 16 bits
DRAM_TOTAL_WIDTH: 16 bits
TOTAL_MEMORY_SIZE: 64 MBytes
Flash component: SPI Flash
Date:Sep 14 2012  Time:15:09:21
============================================
icache: sets:512, ways:4, linesz:32 ,total:65536
dcache: sets:256, ways:4, linesz:32 ,total:32768

 ##### The CPU freq = 500 MHZ ####
 estimate memory size =64 Mbytes

Please choose the operation:
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Entr boot command line interface.
   7: Load Boot Loader code then write to Flash via Serial.
   9: Load Boot Loader code then write to Flash via TFTP.                     4                                                                              initializing CHIP_RTL8367R_VB 1010                                            0
initializing CHIP_RTL8367R_VB 1010

3: System Boot system code via Flash.
## Booting image at bc050000 ...
.   Image Name:   N750F9K1103VB
   Created:      2022-09-21   5:54:21 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1890833 Bytes =  1.8 MB
   Load Address: 80000000
   Entry Point:  80000000
.............................   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 64

Starting kernel ...

[    0.000000] Linux version 5.10.143 (user@chuwi14) (mipsel-openwrt-linux-musl-                                                                             gcc (OpenWrt GCC 11.3.0 r20705-1722e23ffc) 11.3.0, GNU ld (GNU Binutils) 2.37) #                                                                             0 Wed Sep 21 05:54:21 2022
[    0.000000] SoC Type: Ralink RT3883 ver:1 eco:5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] MIPS: machine is Belkin F9K1109 Version 1.0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32                                                                              bytes
[    0.000000] Zone ranges:

```

And now with the loader working and below config . Look at empty KERNEL line
```
rt3883.mk
define Device/belkin_f9k1109v1
  $(Device/uimage-lzma-loader)
  SOC := rt3883
  BLOCKSIZE := 64k
  DEVICE_VENDOR := Belkin
  DEVICE_MODEL := F9K1109
  DEVICE_VARIANT := Version 1.0
  DEVICE_PACKAGES := kmod-usb-ohci kmod-usb2 kmod-usb-ledtrig-usbport
  IMAGE_SIZE := 7808k
  # Stock firmware checks for this uImage image name during upload.
  UIMAGE_NAME := N750F9K1103VB
endef
TARGET_DEVICES += belkin_f9k1109v1
```
boot log

```
initializing CHIP_RTL8367R_VB 1010

3: System Boot system code via Flash.
## Booting image at bc050000 ...
.   Image Name:   N750F9K1103VB
   Created:      2022-09-21   5:54:21 UTC
   Image Type:   MIPS Linux Kernel Image (uncompressed)
   Data Size:    1603483 Bytes =  1.5 MB
   Load Address: 80000000
   Entry Point:  80000000
.........................   Verifying Checksum ... OK
OK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 64

Starting kernel ...

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80000000...

[    0.000000] Linux version 5.10.143 (user@chuwi14) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20705-1722e23ffc) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 Wed Sep 21 05:54:21 2022
[    0.000000] SoC Type: Ralink RT3883 ver:1 eco:5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] MIPS: machine is Belkin F9K1109 Version 1.0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
```

now it's PR time

xabolcs, 2022-10-19:
This is also reported on forums: [Rt3883 based Belkin F9K1109 gets LZMA error 1 -.> v22.03 and v21.02 release](https://forum.openwrt.org/t/rt3883-based-belkin-f9k1109-gets-lzma-error-1-v22-03-and-v21-02-release/140155)

Looks like the fix should be backported to 21.02 too.

commit ac296f62 (2022-10-22): ramips: rt3883: enable lzma-loader for Belkin F9K1109v1

Fixes boot loader LZMA decompression issues

Fixes: #10968
commit ef5b1ffb (2022-10-23): ramips: rt3883: enable lzma-loader for Belkin F9K1109v1

Fixes boot loader LZMA decompression issues

Fixes: #10968

(cherry picked from commit ac296f621058119501ccd54e7cb2a243af5dc5a0)
commit 09a649ff (2022-10-23): ramips: rt3883: enable lzma-loader for Belkin F9K1109v1

Fixes boot loader LZMA decompression issues

Fixes: #10968

(cherry picked from commit ac296f621058119501ccd54e7cb2a243af5dc5a0)