openwrt/openwrt #10983: ipq806x Linksys EA8500 snapshot: Ethernet switch is dead

paulcarroty, 2022-10-15:
No connection, zero packets rx/tx, on lan and wan.

jaharon, 2022-10-24:
I can confirm that this only happens with kernel 5.15.
@Ansuel this happened even before the recent move to 5.15 as the default kernel.
I'm available to debug over wifi (still works) or serial.

Ansuel, 2022-10-24:
@jaharon thanks for the report... can you provide a boot log and a 
`cat  /sys/kernel/debug/devices_deferred`

jaharon, 2022-10-24:
Sure,

`cat /sys/kernel/debug/devices_deferred` is empty

Bootlog: [ea8500_5.15_boot.log](https://github.com/openwrt/openwrt/files/9855481/ea8500_5.15_boot.log)

Ansuel, 2022-10-24:
```
[    3.308228] mdio_bus 37000000.mdio-mii: MDIO device at address 0 is missing.
[    3.309036] mdio_bus 37000000.mdio-mii: MDIO device at address 4 is missing.
```

internal mdio is not working o.o or probably some pin are fked up?

Ansuel, 2022-10-24:
@jaharon can you also provide a bootlog from 5.10?

janaz, 2022-10-24:
I had exactly the same issue with kernel 5.15 and found this bug report. Downgrading to kernel v5.10 fixed the problem for me. Here's my 5.10 bootlog 
[ea8500-bootlog-5.10.txt](https://github.com/openwrt/openwrt/files/9855666/ea8500-bootlog-5.10.txt)

jaharon, 2022-10-24:
Here's mine:
[ea8500_5.10_boot.log](https://github.com/openwrt/openwrt/files/9855802/ea8500_5.10_boot.log)

Ansuel, 2022-10-24:
can you guys test a custom image? or that would require times?

jaharon, 2022-10-24:
Can do, but If you have a public branch with patches, I can also test that.

jaharon, 2022-10-24:
@Ansuel Didn't work
[ea8500_5.15_patch1_boot.log](https://github.com/openwrt/openwrt/files/9855970/ea8500_5.15_patch1_boot.log)

```diff --git a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
index 1c6a4bdacd..da779d7b3e 100644
--- a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
+++ b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
@@ -10,8 +10,6 @@
        };

        aliases {
-               mdio-gpio0 = &mdio0;
-
                led-boot = &led_power;
                led-failsafe = &led_power;
                led-running = &led_power;

Ansuel, 2022-10-24:
@jaharon can you replace gmac1 and gmac2 node with the following values? (keep the alias removed)
```
&gmac1 {
	status = "okay";

	phy-mode = "rgmii";
	qcom,id = <1>;
	qcom,phy_mdio_addr = <4>;
	qcom,poll_required = <1>;
	qcom,rgmii_delay = <0>;
	qcom,phy_mii_type = <0>;
	qcom,emulation = <0>;
	qcom,irq = <255>;
	mdiobus = <&mdio0>;

	fixed-link {
		speed = <1000>;
		full-duplex;
	};
};

&gmac2 {
	status = "okay";

	phy-mode = "sgmii";
	qcom,id = <2>;
	qcom,phy_mdio_addr = <0>;	/* none */
	qcom,poll_required = <0>;	/* no polling */
	qcom,rgmii_delay = <0>;
	qcom,phy_mii_type = <1>;
	qcom,emulation = <0>;
	qcom,irq = <258>;
	mdiobus = <&mdio0>;

	fixed-link {
		speed = <1000>;
		full-duplex;
	};
};
```

The configuration in ea8500 is very different from standard... wonder if they did something stupid and didn't connect the internal mdio controller to the switch in some way... if this doesn't work then we have to revert to the slower bitbang driver... (so the solution is always there... it's just strange...)

jaharon, 2022-10-25:
@Ansuel  Same.
[ea8500_5.15_patch2_boot.log](https://github.com/openwrt/openwrt/files/9856087/ea8500_5.15_patch2_boot.log)

I appreciate the time and effort into supporting this non standard router.
If you have other things to check before (or after) going back to bitbang let me know, even if it is just exploratory.

```diff --git a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
index 1c6a4bdacd..c4b9ffb663 100644
--- a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
+++ b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
@@ -10,8 +10,6 @@
        };

        aliases {
-               mdio-gpio0 = &mdio0;
-
                led-boot = &led_power;
                led-failsafe = &led_power;
                led-running = &led_power;
@@ -113,16 +111,39 @@
 };

 &gmac1 {
+       status = "okay";
+
+       phy-mode = "rgmii";
+       qcom,id = <1>;
        qcom,phy_mdio_addr = <4>;
        qcom,poll_required = <1>;
        qcom,rgmii_delay = <0>;
+       qcom,phy_mii_type = <0>;
        qcom,emulation = <0>;
+       qcom,irq = <255>;
+       mdiobus = <&mdio0>;
+
+       fixed-link {
+               speed = <1000>;
+               full-duplex;
+       };
 };

-/* LAN */
 &gmac2 {
+       status = "okay";
+
+       phy-mode = "sgmii";
+       qcom,id = <2>;
        qcom,phy_mdio_addr = <0>;       /* none */
        qcom,poll_required = <0>;       /* no polling */
        qcom,rgmii_delay = <0>;
+       qcom,phy_mii_type = <1>;
        qcom,emulation = <0>;
+       qcom,irq = <258>;
+       mdiobus = <&mdio0>;
+
+       fixed-link {
+               speed = <1000>;
+               full-duplex;
+       };
 };

Ansuel, 2022-10-25:
@jaharon there is one thing to test but it's really strange... imho it's just not connected but ok i'm preparing a patch for swconfig to test that...

Ansuel, 2022-10-25:
@jaharon actually now that i notice it... on 5.10 we already had the mdio internal thing and it worked correctly... interesting.... also the error is from mdio not from swconfig so wonder if there is something wrong there?

Ansuel, 2022-10-25:
@jaharon can you give me output of `ls /sys/class/mdio_bus/37000000.mdio-mii`

jaharon, 2022-10-25:
@Ansuel 
5.10
```
ls /sys/class/mdio_bus/37000000.mdio-mii
37000000.mdio-mii:00  37000000.mdio-mii:04  device                of_node               statistics            subsystem             uevent
```
5.15 - with patch2
```
ls /sys/class/mdio_bus/37000000.mdio-mii
device      of_node     statistics  subsystem   uevent
```
Let me know if you want me to check it with no patches.

Ansuel, 2022-10-25:
@jaharon i need to investigate this with a clear mind and here is 3 am... hope it's ok for you if we can continue this tomorrow.

thanks for helpming me bisecting this

Ansuel, 2022-10-25:
@jaharon `cat /sys/class/mdio_bus/37000000.mdio-mii/statistics/*`

Ansuel, 2022-10-25:
anyway i would first test if gpio bitbang works just to understand if there is a regression in the mdio driver or in the system itself.

you need to add this to mdio node

```
		compatible = "virtual,mdio-gpio";
		gpios = <&qcom_pinmux 1 GPIO_ACTIVE_HIGH>,
			<&qcom_pinmux 0 GPIO_ACTIVE_HIGH>;
```

jaharon, 2022-10-25:
@Ansuel 

[mdio_stats_5_15.txt](https://github.com/openwrt/openwrt/files/9863188/mdio_stats_5_15.txt)
[mdio_stats_5_10.txt](https://github.com/openwrt/openwrt/files/9863189/mdio_stats_5_10.txt)

The switch still does not work with the following patch and `/sys/class/mdio_bus/37000000.mdio-mii/statistics/*` does not exist:
```diff --git a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
index 1c6a4bdacd..191ff4d3b4 100644
--- a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
+++ b/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ea8500.dts
@@ -107,6 +107,10 @@
 };

 &mdio0 {
+       compatible = "virtual,mdio-gpio";
+       gpios = <&qcom_pinmux 1 GPIO_ACTIVE_HIGH>,
+               <&qcom_pinmux 0 GPIO_ACTIVE_HIGH>;
+
        phy4: ethernet-phy@4 {
                reg = <4>;
        };
```

Ansuel, 2022-10-25:
@jaharon with the virtual mdio can you give me also the bootlog?

jaharon, 2022-10-25:
@Ansuel Sure
[ea8500_5.15_virtual_mdio_boot.log](https://github.com/openwrt/openwrt/files/9863357/ea8500_5.15_virtual_mdio_boot.log)

robimarko, 2022-10-26:
It would be good to compare GPIO levels on 5.10 and 5.15, maybe this thing has reset GPIO for the switch

Ansuel, 2022-10-26:
@robimarko yep that is also my idea... we are silently changing a gpio and the switch is stuck in reset... but any idea how to check that from a running system ?

robimarko, 2022-10-26:
You can just `cat /sys/kernel/debug/gpio` that will give you all of the GPIO-s and their level

jaharon, 2022-10-26:
[ea8500_gpios_5.10.txt](https://github.com/openwrt/openwrt/files/9870208/ea8500_gpios_5.10.txt)
[ea8500_gpios_5.15.txt](https://github.com/openwrt/openwrt/files/9870209/ea8500_gpios_5.15.txt)

```diff

diff ea8500_gpios_5.10.txt ea8500_gpios_5.15.txt
6c6
<  gpio2   : in  high func0 2mA pull down
---
>  gpio2   : in  low  func0 2mA pull down
56c56
<  gpio52  : in  low  func2 8mA no pull
---
>  gpio52  : in  high func2 8mA no pull
63,64c63,64
<  gpio59  : in  low  func2 8mA no pull
<  gpio60  : in  low  func2 8mA no pull
---
>  gpio59  : in  high func2 8mA no pull
>  gpio60  : in  high func2 8mA no pull
66,67c66,67
<  gpio62  : in  low  func2 8mA no pull
<  gpio63  : out low  func0 12mA no pull
---
>  gpio62  : in  high func2 8mA no pull
>  gpio63  : out high func0 12mA no pull
```

Ansuel, 2022-10-26:
we don't have gpio2 in ipq8064 dtsi but aside from gpio63 every other is related to rgmii2... thing we got him?

robimarko, 2022-10-26:
Well, toggle it during runtime and check with mdio tools if there is something on MDIO bus after that

robimarko, 2022-10-26:
@jaharon Any chance you can build a image with `kmod-mdio-netlink` and `mdio-tools`?

That would allow toggling the GPIO2 by exporting it in `/sys/class/gpio` and then toggling

jaharon, 2022-10-26:
@robimarko 
```
ls /sys/class/gpio/
export       gpiochip443  unexport
```
```
echo "445" > /sys/class/gpio/export

echo "out" > /sys/class/gpio/gpio445/direction

echo "1" > /sys/class/gpio/gpio445/value

cat /sys/kernel/debug/gpio
...
 gpio2   : out high func0 2mA pull down
...
```
Still not seeing anything in
```
ls /sys/class/mdio_bus/37000000.mdio-mii
device      of_node     statistics  subsystem   uevent
```

robimarko, 2022-10-26:
Kernel will not reprobe the MDIO devices, that is why you need mdio-tools to be able to see whats on the MDIO bus when you want

jaharon, 2022-10-26:
I ran this before and after the gpio2 change, and the output is the same:
```
mdio

37000000.mdio-mii
fixed-0
```
```
mdio 37000000.mdio-mii

 DEV      PHY-ID  LINK
 ```
 ```
mdio fixed-0

 DEV      PHY-ID  LINK 
 ```

Ansuel, 2022-10-26:
@jaharon you should really restore each gpio to 5.10 value and not just gpio2

jaharon, 2022-10-27:
I could not change gpio63:
```
echo "506" > /sys/class/gpio/export
ash: write error: Resource busy
```
All the rest were easy to change
```diff
diff ea8500_gpios_5.10.txt ea8500_gpios_5.15_manual_gpio_change.txt
6c6
<  gpio2   : in  high func0 2mA pull down
---
>  gpio2   : out high func0 2mA pull down
56c56
<  gpio52  : in  low  func2 8mA no pull
---
>  gpio52  : out low  func0 8mA no pull
63,64c63,64
<  gpio59  : in  low  func2 8mA no pull
<  gpio60  : in  low  func2 8mA no pull
---
>  gpio59  : out low  func0 8mA no pull
>  gpio60  : out low  func0 8mA no pull
66,67c66,67
<  gpio62  : in  low  func2 8mA no pull
<  gpio63  : out low  func0 12mA no pull
---
>  gpio62  : out low  func0 8mA no pull
>  gpio63  : out high func0 12mA no pull
```
[ea8500_gpios_5.15_manual_gpio_change.txt](https://github.com/openwrt/openwrt/files/9875868/ea8500_gpios_5.15_manual_gpio_change.txt)

No change using the mdio command.

robimarko, 2022-10-27:
Can you enable `CONFIG_KERNEL_DEVMEM`, `CONFIG_BUSYBOX_CUSTOM` and `CONFIG_BUSYBOX_CONFIG_DEVMEM`.

Then run the following script: https://raw.githubusercontent.com/marxfang/ipq807x-spf100-cs/master/chipcode/NHSS.QSDK.10.0/apss_proc/out/proprietary/QSDK-Base/gpio-debug/qca/feeds/qca-platform-utils/gpio-debug/files/gpio-debug-ipq806x.sh

Make sure to pass the all argument, so: `./gpio-debug-ipq806x.sh all`

That will tell us if the desired pinmuxing is actually being set

jaharon, 2022-10-27:
@robimarko NP - [ea8500_gpio-debug_5_15.txt](https://github.com/openwrt/openwrt/files/9881183/ea8500_gpio-debug_5_15.txt)

robimarko, 2022-10-27:
@jaharon Any chance you can do the same with 5.10?

Honestly, MDIO pins that set no pull-up on are weird

jaharon, 2022-10-27:
@robimarko [ea8500_gpio-debug_5_10.txt](https://github.com/openwrt/openwrt/files/9881512/ea8500_gpio-debug_5_10.txt)
no diff between ea8500_gpio-debug_5_10.txt and ea8500_gpio-debug_5_15.txt

robimarko, 2022-10-27:
@jaharon Thanks for the patience, but nothing looks out of the ordinary.

I am running out of ideas

jaharon, 2022-10-27:
@robimarko I understand. Thanks heaps!

Ansuel, 2022-10-27:
@jaharon I'm pretty busy in these days so I would ask you if you can use a stable 22.03 image while i think of possible cause... think for now i will have to flag the target as broken but this definitely have to be fixed. Hope it's not a problem for you.

jaharon, 2022-10-28:
@Ansuel All good. Ping me if you need me.

robimarko, 2022-10-28:
@jaharon Can you send me the 5.10 and 5.15 images so I dont have to build them myself?

I want to compare the generated DTB-s

jaharon, 2022-10-29:
@robimarko Sure, emailed them to you.

jaharon, 2022-10-30:
@robimarko This might be beneficial:

I used ```device-tree-compiler```  
to convert 
```build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/image-qcom-ipq8064-ea8500.dtb```
 to
  ```image-qcom-ipq8064-ea8500.dts```

using 
```i="image-qcom-ipq8064-ea8500.dtb";  dtc -I dtb -O dts "$i" -o "`echo $i | sed -e s'/\.dtb/\.dts/'`"```

[image-qcom-ipq8064-ea8500_5_10.dts.txt](https://github.com/openwrt/openwrt/files/9895269/image-qcom-ipq8064-ea8500_5_10.dts.txt)

[image-qcom-ipq8064-ea8500_5_15.dts.txt](https://github.com/openwrt/openwrt/files/9895270/image-qcom-ipq8064-ea8500_5_15.dts.txt)

The diff is quite large 
[ea8500_dts_diff_5_10_vs_5_15.txt](https://github.com/openwrt/openwrt/files/9895289/ea8500_dts_diff_5_10_vs_5_15.txt)

paulcarroty, 2022-10-30:
Thanks all for digging in, found something strange in latest dts diff:

```
< 			mdio0_pins {
< 				phandle = <0x28>;
---
> 			mdio0-pins {
> 				phandle = <0x2d>;
```

```
< 	smem {
< 		compatible = "qcom,smem";
< 		memory-region = <0x2d>;
< 		hwlocks = <0x2e 0x03>;
< 	};
```

robimarko, 2022-10-30:
You cant really diff directly as a lot of nodes got reordered, I looked at the 5.15 vs 5.10 and only difference I could spot is SPC got disabled, but that doesn't really have anything to do with MDIO.

Ansuel, 2022-11-09:
@paulcarroty @jaharon I wonder if you guys can do something crazy... I can build a fw for each kernel version from 5.10 to 5.15 but i need to ask you guys if you can test them...

Is it a problem or it's too much to ask? Ideally they should be tested with initramfs so serial can be required (but we can sort a way to make it optional)

I would really like to fix this as currently also EA7500 is broken (as I suspected and I was right with also disabling that on master)

jaharon, 2022-11-09:
@Ansuel Not a problem. Send away.

Ansuel, 2022-11-09:
@jaharon sooo you can test initramfs with serial? Just confirming that if i can send initramfs or factory image

jaharon, 2022-11-09:
@Ansuel I have a serial setup. It might take me some time to setup a tftp server and find the right boot command (been a while), but I'm sure that I'll get it done over the weekend.

Ansuel, 2022-11-09:
@jaharon ok i will prepare the image... for the command and the tool it's easy i can pass you the command and the tool is just tftpd64

Would be fun if things magically works with a tftp image and problem is just a custom thing in the linksys bootloader than now conflicts with 5.15

paulcarroty, 2022-11-10:
I can get the serial connect too, but it's complicated a bit 'cause no headers. By the way, this is not the first EA8500 switch issue: https://github.com/openwrt/openwrt/issues/7379, https://github.com/openwrt/openwrt/pull/2047

Ansuel, 2022-11-11:
@jaharon @paulcarroty can you test the 2 commit present here?
https://git.openwrt.org/?p=openwrt/staging/ansuel.git;a=shortlog;h=refs/heads/fix-eax500

jaharon, 2022-11-11:
@ansuel The switch is not working.

New boot log
[ea8500_5.15_boot_with_new_patches.log](https://github.com/openwrt/openwrt/files/9993275/ea8500_5.15_boot_with_new_patches.log)

```
ls /sys/class/mdio_bus/37000000.mdio-mii
device      of_node     statistics  subsystem   uevent
```

The gpios diff doesn't read to me as I expected with this patch:
```diff
diff a8500_gpios_5.15.txt ea8500_gpios_5.15_with_new_patches.txt
67c67
<  gpio63  : out high func0 12mA no pull
---
>  gpio63  : out high func0 10mA pull down
```

The new dts
[image-qcom-ipq8064-ea8500_5_15_with_new_patches.dts.txt](https://github.com/openwrt/openwrt/files/9993292/image-qcom-ipq8064-ea8500_5_15_with_new_patches.dts.txt)

dts diff 
[dts_diff_5_15_VS_5_15_with_new_patches.txt](https://github.com/openwrt/openwrt/files/9993298/dts_diff_5_15_VS_5_15_with_new_patches.txt)

The patches I've used in my build box:
generic: ar8327: add optional gpio reset support
ipq806x: fix non working switch for eax500 devices

Ansuel, 2022-11-11:
Nice! we are on something!

```
[   21.371873] ipq806x-gmac-dwmac 37200000.ethernet: Failed to reset the dma
[   21.371913] ipq806x-gmac-dwmac 37200000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   21.377722] ipq806x-gmac-dwmac 37200000.ethernet eth0: __stmmac_open: Hw setup failed
```

Ansuel, 2022-11-11:
@jaharon how you installed this? sysupgrade and softreboot? can you try to power cycle the device?

jaharon, 2022-11-11:
@Ansuel

I sysupgraded from the web interface, after the initial reboot I grabbed all the data. I then did a power cycle and checked the switch.

I now did a soft reboot to the second partition with 5.15 (plus patches), no switch with 
```ipq806x-gmac-dwmac 37200000.ethernet: Failed to reset the dma``` in the log.
I then power cycled the router (5 seconds power off and then on), no switch and same message in the logs.

Ansuel, 2022-11-12:
btw i'm analyzing the linksys GPL source and yes i found the gpio63 thing and also ea7500 is just ea8500 with less thing... the source of 7500 is based on the source of ea8500 so I was right with the idea that they are based on the same hw design from linksys...

Ansuel, 2022-11-13:
@jaharon can you check one thing? what are the leds state while the fw is booting? at some time ethernet leds should all power on and then off... (that means we are correctly resetting the switch)

[21 more replies]

closed 2023-02-17 by openwrt-bot