openwrt/openwrt #10986: mt7622: I/O error saving mtdblock2 from LuCI

kevinoid, 2022-10-16:
On a Belkin RT3200 running OpenWrt 22.03.1 (UBI), attempting to save mtdblock contents for the "factory" mtdblock using the LuCI web interface results in a download which is 131,072 bytes instead of the expected 1,048,576 bytes (as indicated by `/sys/class/block/mtdblock2/size` and in the `Content-Type` header of the truncated HTTP response).

Running `dd if=/dev/mtdblock2 of=/dev/null` produces `dd: /dev/mtdblock2: I/O error`.  After encountering the error, `dmesg` contains:

    blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x84700 phys_seg 32 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 264 op 0x0:(READ) flags 0x84700 phys_seg 31 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 272 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 280 op 0x0:(READ) flags 0x84700 phys_seg 29 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 288 op 0x0:(READ) flags 0x84700 phys_seg 28 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 296 op 0x0:(READ) flags 0x84700 phys_seg 27 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 304 op 0x0:(READ) flags 0x84700 phys_seg 26 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 312 op 0x0:(READ) flags 0x84700 phys_seg 25 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 320 op 0x0:(READ) flags 0x84700 phys_seg 24 prio class 0
    blk_update_request: I/O error, dev mtdblock2, sector 328 op 0x0:(READ) flags 0x84700 phys_seg 23 prio class 0
    Buffer I/O error on dev mtdblock2, logical block 32, async page read
    Buffer I/O error on dev mtdblock2, logical block 32, async page read

Full [dmesg](https://github.com/openwrt/openwrt/files/9793449/openwrt-22.03.1-mt7622-dmesg.txt) and [logread](https://github.com/openwrt/openwrt/files/9793450/openwrt-22.03.1-mt7622-logread.txt).

I'd appreciate any suggestions for how to debug the issue further.  Let me know if there's any additional information I can provide.

Thanks,
Kevin

DragonBluep, 2022-10-16:
Same issue on ath79/nand/netgear,r6100/master branch.
Maybe it is related to #10232.
If you have time you can try to revert 8835da4d9047cb11d6267b59cf414bd4de3984bc to see if it works properly.

kevinoid, 2022-10-16:
Good idea, thanks @DragonBluep!  I'm building an image from the master branch with https://github.com/openwrt/openwrt/commit/8835da4d9047cb11d6267b59cf414bd4de3984bc reverted.  The build will take a while, so let me know if that's not what you had in mind or anything about the build commands looks off.  Otherwise, I'll plan to test the resulting image tomorrow.

<details>
<summary>Build commands used for testing</summary>

```sh
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git revert --no-edit 8835da4d9047cb11d6267b59cf414bd4de3984bc
./scripts/feeds update -a
./scripts/feeds install -a
wget -O .config https://downloads.openwrt.org/releases/22.03.1/targets/mediatek/mt7622/config.buildinfo
make -j"$(nproc)" defconfig download clean world
``` 
</details>

DragonBluep, 2022-10-16:
If you are trying to build master branch, you need to revert 79af0593a30bfe382b350e9540ecfafa133b6ac0 and 41e1e838fbbb0308ad7353a3714e04b8c4585e7e. And it is recommended to execute `make target/linux/refresh` before `make -j"$(nproc)"`.

kevinoid, 2022-10-16:
Thanks again @DragonBluep!  Attempting to revert 79af0593a30bfe382b350e9540ecfafa133b6ac0 or 41e1e838fbbb0308ad7353a3714e04b8c4585e7e causes merge conflicts on master (cb24be47ffda12616a0670e9d29a96f9530a19b8).  Do you think it would be sufficient to test with v22.03.2 (ea845f76ea2406a4c70d96141d6386b81bba5431)?

DragonBluep, 2022-10-16:
> Do you think it would be sufficient to test with v22.03.2 ([ea845f7](https://github.com/openwrt/openwrt/commit/ea845f76ea2406a4c70d96141d6386b81bba5431))?

Yep, the two branches are not very different at present. But before that, could you please test official build v22.03.0? This patch was introduced just a little late than v22.03.0 release.
[image]

kevinoid, 2022-10-16:
> But before that, could you please test official build v22.03.0?

Sure thing.  I am also able to reproduce the issue with v22.03.0: [dmesg](https://github.com/openwrt/openwrt/files/9793576/openwrt-22.03.0-mt7622-dmesg.txt), [logread](https://github.com/openwrt/openwrt/files/9793577/openwrt-22.03.0-mt7622-logread.txt).

DragonBluep, 2022-10-16:
> Sure thing. I am also able to reproduce the issue with v22.03.0: [dmesg](https://github.com/openwrt/openwrt/files/9793576/openwrt-22.03.0-mt7622-dmesg.txt), [...]

So my guess is wrong, not caused by this patch. 😂 
When I have time, I will use git bisect to find that bad commit. Thanks for your reply.
At that time, I found this problem when I tried to back up the u-boot partition through LuCI. Finally, I achieved my goal through dd command (although there is an error log, same as yours), so I didn't propose this problem.

kevinoid, 2022-10-16:
No worries. Thanks for investigating!  Let me know if there's anything else I can do to assist or test.

DragonBluep, 2022-10-16:
It would be appreciated if you could test 22.03.0-rc1/4/5/6.

kevinoid, 2022-10-16:
Sure thing.  I'm also able to reproduce the issue with 22.03.0-rc1: [dmesg](https://github.com/openwrt/openwrt/files/9793637/openwrt-22.03.0-rc1-mt7622-dmesg.txt) [logread](https://github.com/openwrt/openwrt/files/9793638/openwrt-22.03.0-rc1-mt7622-logread.txt)
Would you like me to confirm rc4/5/6 as well?

DragonBluep, 2022-10-16:
No, it's unnecessary. It seems that this problem existed long ago. It's just that we haven't found it before. 😂

kevinoid, 2022-10-16:
Always fun to find a new (old) bug! :joy:
Let me know if there's anything else I can do to assist.

DragonBluep, 2022-10-16:
This will waste a lot of your time. First, you need to compile and test the very original commit aa94e34c1d7ce40684ec01b8ddc056548ecd34f0. 
```
git reset --hard aa94e34c1d7ce40684ec01b8ddc056548ecd34f0
```
If it works properly, then you can use the `git bisect` command to locate which commit broke it.
If it does not work, we have to backport RT3200 support to a more ancient branch.
The worst thing is that the problem may only exist in a few devices. I mean, not all RT3200. So, it's really hard.

DragonBluep, 2022-10-16:
@graysky2 Hey Audia, it seems that you have an RT3200. Does your router have this problem? Look forward to your reply， thanks.

kevinoid, 2022-10-16:
> First, you need to compile and test the very original commit https://github.com/openwrt/openwrt/commit/aa94e34c1d7ce40684ec01b8ddc056548ecd34f0.

Sure thing.  Will do.  Is there a trick to build with old feeds/packages?  I'm seeing failures due to missing include/meson.mk, include/openssl-engine.mk, etc. and cherry-picking everything necessary to build current packages against aa94e34c1d7ce40684ec01b8ddc056548ecd34f0 will be painful.

kevinoid, 2022-10-16:
I've made some progress.  The current build script I'm using is:

```sh
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git switch -d aa94e34c1d7ce40684ec01b8ddc056548ecd34f0
# Support for gcc 11+
git cherry-pick -n \
	ac7034cd12c2018338d58eb1e1911755912f33b0 \
	cae69d558135456976b8fc6cb08530d1358cf6d5 \
	81d0ddc806e1f088328ca29eb56db9285b2a4f7b
# include/meson.mk
git cherry-pick -n \
	c4dd2441e7875c9550820f8380b3e41ca619ef27 \
	97258f53634d7237a2962aec3387f011047ce83b \
	4b898afdd2fd9dbbc99dbedca0d50785e3ec063d \
	c4dfdde2eaa399025de1032bed8ee8a5f8babb24 \
	c2222f74c8eff592032473ab2d29d5e8febace63 \
	7c0fb874436c0613a042c8b07aa00621d08f4eaa \
	3bd31cc4d2ff02c2d3c3db80ade7badab08f3e93 \
	75093d1b1de7af82acd3b8bcc61f308c37493400 \
	4c42e2d9dda739baa6a144cdb009aef1a33824a0 \
	fb6cf22866b3bb6dfcbee0659f785982d071c863 \
	628970a1959c0a0fbc0c3deb3d92945af63c2ddc \
	19f3fcc884cab348cfba823f1175baf7aa4de15f
# include/openssl-engine.mk
git cherry-pick -n \
	0bd0de7d43b3846ad0d7006294e1daaadfa7b532 \
	7119fd32d397567931e63dbbf72014e95624018f \
	ac8673ff85fe334913382b5cabb6cf3079b91e24 \
	def9565be632b316c82ffc5a7b28c789e9df75b4 \
	17a6ca12d3dfbb8808ca6d8a5300ff2a9121ba36 \
	30b0351039850d01c382b745a1f40b81b4be2a93 \
	0134f845dab95e0e7f885f4e227d9a41aedf909d \
	e17c6ee62770005e398364ee5d955c9a8ab6f016 \
	1bf94b6797a1e04c0d3accd08b3efb666fbb280e
./scripts/feeds update -a
./scripts/feeds install -a
wget -O .config https://downloads.openwrt.org/releases/22.03.1/targets/mediatek/mt7622/config.buildinfo
make target/linux/refresh
make -j"$(nproc)" defconfig download clean world
```
Unfortunately, the build produces several missing package dependency warnings, such as:

    WARNING: Makefile 'package/feeds/routing/alfred/Makefile' has a dependency on 'libgps', which does not exist
    WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'libxslt', which does not exist
    ...

along with a recursive dependency error:

    tmp/.config-package.in:56462:error: recursive dependency detected!
    tmp/.config-package.in:56462:	symbol PACKAGE_iptables is selected by PACKAGE_ip6tables
    tmp/.config-package.in:56396:	symbol PACKAGE_ip6tables is selected by PACKAGE_olsrd
    tmp/.config-package.in:58165:	symbol PACKAGE_olsrd is selected by PACKAGE_luci-mod-freifunk-community
    tmp/.config-package.in:21746:	symbol PACKAGE_luci-mod-freifunk-community depends on PACKAGE_iptables

[Full build output](https://github.com/openwrt/openwrt/files/9793819/openwrt-aa94e34c1d7ce40684ec01b8ddc056548ecd34f0-buildlog.txt)

Suggestions appreciated.

DragonBluep, 2022-10-16:
Sorry for late. Usually the feeds warnings can be ignored, and we don't check-pick anything.
You need to update your .config file instead of getting it from OpenWrt site, just run
```
make menuconfig
# then choose target: mediatek subtarget: mt7622 device: RT3200, maybe E8450, and luci package is essential.
```
Finally, you can execute and wait for a long time:
`make -j"$(nproc)"`

xabolcs, 2022-10-16:
> Is there a trick to build with old feeds/packages?

There is, in the Wiki!

[Build system usage / Selecting hash](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#selecting_hash)

justinkb, 2022-10-16:
I think this is a really old known issue? https://github.com/dangowrt/owrt-ubi-installer/issues/56

hnyman, 2022-10-16:
Likely so, and actually caused by Belkin/Linksys not using ECC when originally writing those blocks into flash. We encountered the problem in August 2021 when mt7622 was moved to use a new NAND driver that is more picky about missing ECC data no flash.

The same problem is also related to the 5 GHz radio calibration data, and dangowrt's installer reads and rewrites that flash section to enable radio to have the needed calibration data. But to avoid risk of flash bad block problems, the other parts of the mtd2/factory are not rewritten. 

Lots of discussion in forum, starting around here (messages 362-417):
* https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/362

* https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/403

Discussion also in https://github.com/openwrt/openwrt/pull/4179#issuecomment-907631223

justinkb, 2022-10-16:
About this 5GHz calibration data, did the installer always fix that? How can i verify if i used an installer that had this fix, since I've been on these builds for a pretty long time. After fresh reboots my 5GHz takes a while to get online and I get spam of "Survey is missing noise floor" in logs. Or is this completely unrelated?

hnyman, 2022-10-17:
> How can i verify if i used an installer that had this fix. 

The fix has been applied if the 5 GHz radio works at all.

> After fresh reboots my 5GHz takes a while to get online

If you use DFS channels, it first needs to monitor for weather radar before enabling the radio interfaces. It will always take 1-2 min.