openwrt/openwrt #10997: [22.03.x] mvebu: Eth switch forwarding all incoming traffic to all ports disregarding VLAN configuration on WRT3200ACM

pmendla, 2022-10-17:
After upgrade from 21.02 eth switch start to forward all incoming traffic to all ports.
I can see ALL traffic using tcpdump on computer connected to one port and all eth traffic leds are blinking always simultaneously.
Same configuration on 21.02 works ok - VLANs and individual traffic are isolated.

ynezz, 2022-10-18:
We can't tell what is wrong unless you share your `/etc/config/network` file.

pmendla, 2022-10-18:
Configuration is quire simple, 3 VLANs, 
Computer connected to lan2 port can see any incoming traffic.

My configuration (loopback and wlan sections removed for clarity)
```
config device
	option type 'bridge'
	option name 'br0'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'
	option ipv6 '0'
	option bridge_empty '1'

config bridge-vlan
	option device 'br0'
	option vlan '1'
	list ports 'lan1:t'
	list ports 'lan2:u*'
	list ports 'lan3:u*'
	list ports 'lan4:u*'
	list ports 'wan:t'

config bridge-vlan
	option device 'br0'
	option vlan '2'
	list ports 'wan:t'

config bridge-vlan
	option device 'br0'
	option vlan '3'
	list ports 'lan1:t'
	list ports 'wan:t'
	
config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.1.1'
	option device 'br0.1'

config interface 'wan'
	option proto 'dhcp'
	option device 'br0.2'

config interface 'guest'
	option proto 'static'
	option ipaddr '192.168.3.1'
	option netmask '255.255.255.0'
	option device 'br0.3'
```

drujd, 2022-10-24:
Since my duplicate just got closed: On Turris Omnia, this happens even on completely stock/default 22.03.2. It is basically this bug: https://gitlab.nic.cz/turris/os/build/-/issues/336

Turris team fixed this for their 5.15 kernel...

Until this gets fixed, no device with the same switch (mv88e6085, maybe other similar Marvell switches?) should be considered as supported for 22.03.2, quite a blocker I'd say...

borkra, 2022-11-08:
Try disabling flow offload

drujd, 2022-11-08:
> Try disabling flow offload

That does not help at all. The switch does not support hardware flow offloading, and software flow offloading happens in the CPU (so packets are sent to the CPU port at the same time they are being sent to all switch ports). Also flow offloading is off by default...

champtar, 2022-12-22:
> Same issue on wrt1900acs, with kernel 5.15 is ok

Turris Omnia / OpenWrt SNAPSHOT r21563-9ed1830bdc (2022/12/20) / kernel 5.15.84 I still see issues.
Wired devices can access the internet, same for wireless devices, but packets going from wireless to wired (and maybe the other way around) keep their tags and leak on port they should not.

vlan 2 is my private vlan, vlan 3 my iot vlan, I have wireless devices connected to both, using my laptop connected to lan 4 which is supposed to be vlan 2 untagged only, I see packets tagged with both vlan 2 and vlan 3.
Software offloading is not enabled.

I haven't done much testing to properly qualify exactly what's broken but I can if it helps. Also happy to test patches.

<details>
  <summary>/etc/config/network (simplified)</summary>

```
config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.2.1'
	option device 'br-lan.2'

config interface 'lan_iot'
	option proto 'static'
	option netmask '255.255.255.0'
	option ipaddr '192.168.3.1'
	option device 'br-lan.3'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan0'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config bridge-vlan
	option device 'br-lan'
	option vlan '2'
	list ports 'lan4:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '3'
	list ports 'lan3:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '4'
	list ports 'lan2:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '5'
	list ports 'lan1:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '6'
	list ports 'lan0:u*'
```
</details>

<details>
  <summary>/etc/config/wireless (simplified)</summary>

```
config wifi-iface 'default_radio0'
	option network 'lan'

[... 125 lines cut ...]
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_ss=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_uhttpd_lua=y
```
</details>

champtar, 2022-12-23:
Just went back to 22.03-SNAPSHOT r20013-275c51bf40 / kernel 5.10.160, everything works ok to me
(wireless <-> wired communication is ok, no packets seems to leak, ...)

@drujd what does your config look like ? can you try latest 22.03 snapshot ?

champtar, 2022-12-23:
Don't know if Turris Omnia got different switch model/rev, here is mine
```
[    1.302061] mv88e6085 f1072004.mdio-mii:10: switch 0x1760 detected: Marvell 88E6176, revision 1
```

champtar, 2022-12-23:
Looking a bit at the dts & openwrt code, we have `310-linksys-use-eth0-as-cpu-port.patch` (4149d2b91c43784ada37efe928c5f6b610c74259) so WRT3200ACM / wrt1900acs will use port 6 / eth0 instead of port 5 / eth2, Omnia uses port 5 / eth1, don't know if it makes a difference

drujd, 2022-12-23:
> Just went back to 22.03-SNAPSHOT r20013-275c51bf40 / kernel 5.10.160, everything works ok to me
> (wireless <-> wired communication is ok, no packets seems to leak, ...)
> 
> @drujd what does your config look like ? can you try latest 22.03 snapshot ?

The issue manifests with the default OpenWrt configuration without any changes. No vlans, wireless does not need to be configured. This has always been a problem with 5.10, AFAIK, Turris project worked around it by slapping 5.15 on top of OpenWrt 22.03.

I won't get to testing yet another 22.03 snapshot anytime soon, but I am almost certain it is still not fixed. Again, nothing to do with VLANS or wifi, just the switch acting like a hub, even with the default config. You are probably experiencing a different issue.

champtar, 2022-12-23:
I can only do limited testing as this is the house router and I have family around, but here my understanding of everyone reports:
- @drujd: Turris Omnia switch without vlan act as a hub with 5.10, works fine with 5.15
- @pmendla: WRT3200ACM switch act as hub with 22.03 (5.10), vlan configured with some port tagged
- @pesa1234: "same issue" with wrt1900acs, ok with 5.15
- @wyj8bc5odzfl: WRT3200ACM switch act as hub with 22.03 (5.10) with stock config
- @champtar: Turris Omnia with vlan but no tagged port, everything works with 22.03 / 5.10, broken with master / 5.15

@drujd it seems 5.15 is fixing one use case and breaking another one :(
@pmendla can you test latest 22.03 and/or latest master ?
@pesa1234 are you using bridge vlan filtering ?

drujd, 2022-12-24:
> everything works with 22.03 / 5.10, broken with master / 5.15

With all due respect, if you have an Omnia, then with 5.10 you are probably experiencing the "switch is a hub" issue as well, just not noticing, I doubt "everything works".

As for your wifi bridge with vlans issue, have you compared master 5.10 vs master 5.15 to make sure the issue is not caused by something else than kernel?

champtar, 2022-12-24:
> > everything works with 22.03 / 5.10, broken with master / 5.15
> 
> With all due respect, if you have an Omnia, then with 5.10 you are probably experiencing the "switch is a hub" issue as well, just not noticing, I doubt "everything works".

To remove any of your doubts, all my tests are done with a laptop connected to lan4, running `sudo tcpdump -i enp0s31f6 -nne -Qin`. With 5.10, I only see untagged packets from device in vlan2 (arp/ndp/multicast), with 5.15 I see tagged packets with both vlan 2 and vlan 3, which should just not happen.
I do have a device connected in lan3 port, and leds are not blinking at the same time between lan3 and lan4.

> As for your wifi bridge with vlans issue, have you compared master 5.10 vs master 5.15 to make sure the issue is not caused by something else than kernel?

I did this test this afternoon but didn't comment, master 5.10 (b2bfea481e8030561346a49f92818e07518c08c2 + revert of 5f973c1eefa91050f2659c29a276284a7793985b) works fine with my network config, same as 22.03.

I also tried to configure a port as "trunk" (right now all my ports are "access" ports) but 5.10 was still ok.

I just found an old email that I sent to the ML, https://lists.openwrt.org/pipermail/openwrt-devel/2021-May/035037.html,
starting 21.02 I had to use bridge vlan filtering to make my Omnia work

champtar, 2022-12-25:
Interesting patches merged in 5.18: https://lore.kernel.org/all/20220215170218.2032432-1-vladimir.oltean@nxp.com/

TDFKAOlli, 2023-01-15:
O.k. so correct test to verify would be?
- ping in local lan from device 1 to device 2 both being connected on the lan ports of the router
- tcpdump on openWrt device on WAN interface (tcpdump -nne -i wan net 192.168.1.0/24) and check if ping traffic can be seen
Would that be the correct way to verify if a device is affected?

Pinging the internet from device 1 shows incoming packets on the device 2 on my device, so the problem on the LAN switch acting as a hub is seen. Looking at the above way of verifying LAN to WAN I cannot see internal packets on the WAN interface. (Linksys WRT1900ACS). But not sure if the way of testing it is correct.

pmendla, 2023-01-21:
There is no latest  22.03.3 for WRT3200ACM as this model is now not supported.
I've checked latest SNAPSHOT (5.15) and switch seems to work properly.
So probably we have to wait until next stable release 23.x

eehmke, 2023-04-18:
WRT3200ACM is still not supported in OpenWrt 22.03.4.

nmschulte, 2023-06-09:
Does https://firmware-selector.openwrt.org/?version=23.05.0-rc1&target=mvebu%2Fcortexa9&id=linksys_wrt3200acm fix this?

eehmke, 2023-06-10:
@nmschulte I guess so because the firmware selector offers the device. However, I had my WRT3200ACM running on the snapshot release some time now, and the release candidate is not offered via the attended sysupgrade. Must I do a full upgrade via file download?

PalebloodSky, 2023-06-14:
> Does https://firmware-selector.openwrt.org/?version=23.05.0-rc1&target=mvebu%2Fcortexa9&id=linksys_wrt3200acm fix this?

Mostly. WRT32X is working great for me, been on kernel 5.15 builds for many months now. It's still not perfect, since we have the VLAN tagging bug that was just partially fixed on master with a commit today (removing the hack that broke it), but based on what they wrote it won't be full fixed until kernel 6.1, unless it's backported.

Anyway in short, yes 23.05 is working great, the switch is working mostly as it should again.

Scottie35, 2023-11-02:
I just installed 23.0.5.0 on my WRT1900ACSv2, and I also can confirm it's working beautifully again.
That means load balancing rules in MWAN3 are also working again like they should.
Finally, quick note that even though it's not officially supported, I successfully upgraded from v21 directly to 23.05.0, reinstalled packages, then restored backed up v21 config - and it works perfectly.

mmortal03, 2023-11-03:
@pmendla , have you given it a shot?

pmendla, 2023-11-07:
Yes, I'm using 23.05 since rc1 and switch is working properly with configured VLANs.

mmortal03, 2023-11-08:
> Yes, I'm using 23.05 since rc1 and switch is working properly with configured VLANs.

@pmendla , based on that, since you opened this bug report, should you close it now?

pmendla, 2023-11-08:
Fixed in 23.05

closed 2023-11-08 by pmendla