openwrt/openwrt #10998: Client mode breaks wireless scan [22.3.2]

cristianvdr, 2022-10-17:
Hello there is an issue with all OpenWrt 22.3.x versions since rc4 or rc5, till the latest one 22.3.2. Wireless scanning stops working when the device connects as a client.
Steps to reproduce:
Fresh install OpenWrt 22.03.2 r19803-9a599fee93
Device: TP-Link CPE220 v3 ath79/generic QCA9533
After instalation wireless scan works as expected, selecting a network to connect to it, then after it connects to it you can't scan again to choose another network because nothing happens, it keeps scanning without results.
Temporary solution is to disable client mode, then the scanning works again. 
There are no errors in the log except for this:
`daemon.notice wpa_supplicant[1266]: wlan0: Unknown event 37`

tracnunya, 2022-10-20:
I also am having this issue with the latest released on a Netgear ex6150v2 extender. It will scan to find a network and let you connect to it, then it will scan no more and just hangs and timeouts, it is locked up until it completes. If I delete the association to get back to needing a new scan (LUCI - WIRELESS - click ADD on radio), then I can scan again, and connect to a local wifi, then it no longer scans again. It also breaks the screen in LUCI - STATUS - CHANNEL ANALYSIS
I only use 5g so that is what breaks, I have a 2.4 radio as well but it is not used, so the channel analysis still works for it.
In 5g the analysis has a spinner on the bottom STARTING WIRELESS SCAN that just goes nowhere
The ethernet still pings but LUCI is dead and non responsive even in another tab with a new session.
This is never ending in the analysis page unless i click in luci to go somewhere else. Then luci is responsive again and a new tab can start a new luci session
In LUCI WIRELESS SCAN a popup join network: wireless scan is presented, it also just sits and spins forever.

The following is what was added to my rebooted log AFTER I click SCAN
Wed Oct 19 01:39:43 2022 daemon.warn dnsmasq-dhcp[1]: DHCP packet received on wlan1 which has no address
Wed Oct 19 01:40:02 2022 daemon.warn dnsmasq-dhcp[1]: DHCP packet received on wlan1 which has no address
Wed Oct 19 01:40:18 2022 daemon.notice wpa_supplicant[1036]: wlan1: SME: Trying to authenticate with d8:ec:5e:3e:85:e0 (SSID='camelot' freq=5745 MHz)
Wed Oct 19 01:40:18 2022 daemon.notice netifd: Network device 'wlan1' link is down
Wed Oct 19 01:40:18 2022 daemon.notice netifd: Interface 'camelot' has link connectivity loss
Wed Oct 19 01:40:18 2022 kern.info kernel: [  118.949320] wlan1: disconnect from AP 50:6a:03:e6:9a:9c for new auth to d8:ec:5e:3e:85:e0
Wed Oct 19 01:40:18 2022 daemon.notice netifd: camelot (1968): udhcpc: received SIGTERM
Wed Oct 19 01:40:18 2022 daemon.notice netifd: camelot (1968): udhcpc: entering released state
Wed Oct 19 01:40:18 2022 kern.warn kernel: [  119.078545] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 1
Wed Oct 19 01:40:18 2022 kern.info kernel: [  119.088828] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
Wed Oct 19 01:40:18 2022 kern.info kernel: [  119.089508] wlan1: authenticate with d8:ec:5e:3e:85:e0
Wed Oct 19 01:40:18 2022 daemon.notice wpa_supplicant[1036]: wlan1: CTRL-EVENT-AUTH-REJECT d8:ec:5e:3e:85:e0 auth_type=0 auth_transaction=2 status_code=1
Wed Oct 19 01:40:18 2022 daemon.notice wpa_supplicant[1036]: BSSID d8:ec:5e:3e:85:e0 ignore list count incremented to 3, ignoring for 60 seconds
Wed Oct 19 01:40:18 2022 kern.info kernel: [  119.098266] wlan1: send auth to d8:ec:5e:3e:85:e0 (try 1/3)
Wed Oct 19 01:40:18 2022 kern.info kernel: [  119.100598] wlan1: d8:ec:5e:3e:85:e0 denied authentication (status 1)
Wed Oct 19 01:40:18 2022 daemon.notice netifd: camelot (1968): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "camelot" } (Permission denied)
Wed Oct 19 01:40:18 2022 daemon.notice netifd: Interface 'camelot' is now down
Wed Oct 19 01:40:18 2022 daemon.notice netifd: Interface 'camelot' is disabled
Wed Oct 19 01:40:18 2022 daemon.notice netifd: Interface 'camelot' is enabled
Wed Oct 19 01:40:18 2022 kern.warn kernel: [  119.137325] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 1
Wed Oct 19 01:40:18 2022 kern.info kernel: [  119.138008] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
Wed Oct 19 01:40:21 2022 daemon.notice wpa_supplicant[1036]: wlan1: SME: Trying to authenticate with 50:6a:03:e6:9a:9c (SSID='camelot' freq=5745 MHz)
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.555656] wlan1: authenticate with 50:6a:03:e6:9a:9c
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.563658] wlan1: send auth to 50:6a:03:e6:9a:9c (try 1/3)
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.565619] wlan1: authenticated
Wed Oct 19 01:40:21 2022 daemon.notice wpa_supplicant[1036]: wlan1: Trying to associate with 50:6a:03:e6:9a:9c (SSID='camelot' freq=5745 MHz)
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.577071] wlan1: associate with 50:6a:03:e6:9a:9c (try 1/3)
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.578590] wlan1: RX ReassocResp from 50:6a:03:e6:9a:9c (capab=0x1 status=0 aid=2)
Wed Oct 19 01:40:21 2022 kern.warn kernel: [  122.582953] ath10k_ahb a800000.wifi: pdev param 0 not supported by firmware
Wed Oct 19 01:40:21 2022 kern.warn kernel: [  122.589670] ath10k_ahb a800000.wifi: failed to enable peer stats info: -95
Wed Oct 19 01:40:21 2022 kern.info kernel: [  122.596699] wlan1: associated
Wed Oct 19 01:40:21 2022 daemon.notice wpa_supplicant[1036]: wlan1: Associated with 50:6a:03:e6:9a:9c
Wed Oct 19 01:40:21 2022 daemon.notice netifd: Network device 'wlan1' link is up
Wed Oct 19 01:40:21 2022 daemon.notice wpa_supplicant[1036]: wlan1: CTRL-EVENT-CONNECTED - Connection to 50:6a:03:e6:9a:9c completed [id=0 id_str=]
Wed Oct 19 01:40:21 2022 daemon.notice netifd: Interface 'camelot' has link connectivity
Wed Oct 19 01:40:21 2022 daemon.notice netifd: Interface 'camelot' is setting up now
Wed Oct 19 01:40:21 2022 daemon.notice wpa_supplicant[1036]: wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Wed Oct 19 01:40:21 2022 daemon.notice netifd: camelot (2646): udhcpc: started, v1.35.0
Wed Oct 19 01:40:21 2022 daemon.notice netifd: camelot (2646): udhcpc: broadcasting discover
[... cut ...]
[  119.138008] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
[  122.555656] wlan1: authenticate with 50:6a:03:e6:9a:9c
[  122.563658] wlan1: send auth to 50:6a:03:e6:9a:9c (try 1/3)
[  122.565619] wlan1: authenticated
[  122.577071] wlan1: associate with 50:6a:03:e6:9a:9c (try 1/3)
[  122.578590] wlan1: RX ReassocResp from 50:6a:03:e6:9a:9c (capab=0x1 status=0 aid=2)
[  122.582953] ath10k_ahb a800000.wifi: pdev param 0 not supported by firmware
[  122.589670] ath10k_ahb a800000.wifi: failed to enable peer stats info: -95
[  122.596699] wlan1: associated

tracnunya, 2022-10-20:
Additional Info
If I go to LUCI - NETWORK - WIRELESS - EDIT BUTTON
Then
Interface Configuration
General Setup tab
ESSID
and remove the name of my client "camelot" to "test", a non existent wireless entity for me
Then the scan starts working again.
However, when the scan completes it returns NO DATA instead of showing actual wireless entities.
To get a new scan to work, you must delete the existing definition completely and click ADD on radio.

tracnunya, 2022-10-21:
This functionality breaks in
OpenWrt 22.03.0 r19685-512e76967f  

----------------------------------------------

The functions work in....
OpenWrt 19.07.10 r11427-9ce6aa9d8d 
OpenWrt 21.02.0 r16279-5cc0535800 
OpenWrt 21.02.1 r16325-88151b8303 
OpenWrt 21.02.5 r16688-fa9a932fdb 

Scan works over and over on these release. nothing is added to either log file following multiple scans and joins, with no save and accept.
---------------------------------
Breaks first time on
OpenWrt 22.03.0 r19685-512e76967f 

Nothing is added to the system or kernel log after clicking the second scan, system hangs forever and web will timeout session, dismissing it will eventually allow you to regain control prior to the timeout

The only way to perform a second scan, or to use STATUS - CHANNEL ANALYSIS is to remove any existing radio interface definitions, making OpenWrt 22.03.0 r19685-512e76967f  and above unusable for Netgear EX6150 V2

tracnunya, 2022-11-02:
For anyone trying to work around this issue, I am running a batman mesh with a dir-2660 and twenty ex6150v2 units. It works flawlessly, I am running 21.02.5 r16688-fa9a932fdb on all equipment. This is as high a version that will work.

tracnunya, 2022-11-02:
ok, i have a major update on this issue
I flashed all 6 versions to the current oct 2022 release
all of them have this issue, you can only scan and connect to one radio on each band, 2.4 and 5g
after that the luci interface will hang on subsequent scans

I then took the latest version 22.03.2 which also has this issue
Then I took the backup from one of my ex6150v2 mesh units and restored it to this virgin flash
The unit came up and appears to work, which requires two radio configurations on the 5g

BUT MORE IMPORTANTLY, THE SCAN NOW WORKS AGAIN ON THE LATEST RELEASE ONCE THESE TWO RADIOS WERE DEFINED VIA RESTORED FROM A BACKUP FROM A LOWER PROVEN WORKING RELEASE 21.02.5 r16688-fa9a932fdb

What does this APPEAR to mean, 
1- You can probably manually define the second radio even a fake one to get past the issue of "once a scan is joined no more scans can be done
2- You can definitely work around by using a restore with the two or more radio definitions from a lower release.

tracnunya, 2022-11-05:
TRICK IS
Do NOT attach the radios to the LAN, if you do, then you get bridge not allowed in logs and radios go down
With my batman mesh, the radios are not directly attached to the lan

tracnunya, 2022-11-05:
LAST THREE LINES IN LOG FILE AFTER START UP ON A FULLY RESET 22.03.2

Fri Oct 14 22:45:37 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Oct 14 22:45:37 2022 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 1 addresses
Fri Oct 14 22:45:37 2022 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses

CLICKED SCAN JOIN CREATED A CLIENT RADIO, LEFT ALL AT DEFAULT, IT CREATED WWAN WITH A FIREWALL
RADIO IS CONNECTED TO SYSTEM NEWLY CREATED "WWAN" INSIDE THE EXISTING "WAN"
10 MINUTES LATER RADIO IS STILL UP AND ASSOCIATED

Fri Oct 14 22:48:04 2022 kern.warn kernel: [  215.033039] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Fri Oct 14 22:48:04 2022 kern.warn kernel: [  215.033094] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
Fri Oct 14 22:48:04 2022 kern.info kernel: [  215.069308] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Fri Oct 14 22:48:04 2022 kern.info kernel: [  215.070270] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
Fri Oct 14 22:48:04 2022 kern.info kernel: [  215.353830] ath10k_ahb a800000.wifi: rts threshold -1
Fri Oct 14 22:48:04 2022 kern.warn kernel: [  215.359759] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Fri Oct 14 22:48:05 2022 kern.info kernel: [  215.834514] ath10k_ahb a800000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
Fri Oct 14 22:48:40 2022 kern.warn kernel: [  250.653600] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Fri Oct 14 22:48:40 2022 kern.warn kernel: [  250.653667] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
Fri Oct 14 22:48:40 2022 kern.info kernel: [  250.689470] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Fri Oct 14 22:48:40 2022 kern.info kernel: [  250.690552] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
Fri Oct 14 22:48:40 2022 kern.info kernel: [  250.973603] ath10k_ahb a800000.wifi: rts threshold -1
Fri Oct 14 22:48:40 2022 kern.warn kernel: [  250.980646] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Fri Oct 14 22:48:40 2022 daemon.notice netifd: Wireless device 'radio1' is now up
Fri Oct 14 22:48:40 2022 daemon.notice netifd: Interface 'wwan' is enabled
Fri Oct 14 22:48:44 2022 daemon.notice wpa_supplicant[1286]: wlan1: SME: Trying to authenticate with d8:ec:5e:3e:85:e0 (SSID='camelot' freq=5745 MHz)
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.570542] wlan1: authenticate with d8:ec:5e:3e:85:e0
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.761654] wlan1: send auth to d8:ec:5e:3e:85:e0 (try 1/3)
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.764472] wlan1: authenticated
Fri Oct 14 22:48:44 2022 daemon.notice wpa_supplicant[1286]: wlan1: Trying to associate with d8:ec:5e:3e:85:e0 (SSID='camelot' freq=5745 MHz)
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.777062] wlan1: associate with d8:ec:5e:3e:85:e0 (try 1/3)
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.778468] wlan1: RX AssocResp from d8:ec:5e:3e:85:e0 (capab=0x1 status=0 aid=3)
Fri Oct 14 22:48:44 2022 kern.warn kernel: [  254.783158] ath10k_ahb a800000.wifi: pdev param 0 not supported by firmware
Fri Oct 14 22:48:44 2022 kern.warn kernel: [  254.789408] ath10k_ahb a800000.wifi: failed to enable peer stats info: -95
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.796100] wlan1: associated
Fri Oct 14 22:48:44 2022 daemon.notice netifd: Network device 'wlan1' link is up
Fri Oct 14 22:48:44 2022 daemon.notice netifd: Interface 'wwan' has link connectivity
Fri Oct 14 22:48:44 2022 daemon.notice netifd: Interface 'wwan' is setting up now
Fri Oct 14 22:48:44 2022 daemon.notice wpa_supplicant[1286]: wlan1: Associated with d8:ec:5e:3e:85:e0
Fri Oct 14 22:48:44 2022 daemon.notice wpa_supplicant[1286]: wlan1: CTRL-EVENT-CONNECTED - Connection to d8:ec:5e:3e:85:e0 completed [id=0 id_str=]
Fri Oct 14 22:48:44 2022 kern.info kernel: [  254.805374] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Fri Oct 14 22:48:44 2022 daemon.notice wpa_supplicant[1286]: wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Fri Oct 14 22:48:44 2022 kern.warn kernel: [  254.828915] ath10k_ahb a800000.wifi: Invalid peer id 0 or peer stats buffer, peer: 00000000  sta: 00000000
Fri Oct 14 22:48:44 2022 daemon.notice netifd: wwan (3807): udhcpc: started, v1.35.0
Fri Oct 14 22:48:44 2022 daemon.notice netifd: wwan (3807): udhcpc: broadcasting discover
Fri Oct 14 22:48:48 2022 daemon.notice netifd: wwan (3807): udhcpc: broadcasting discover
Fri Oct 14 22:48:48 2022 daemon.notice netifd: wwan (3807): udhcpc: broadcasting select for 192.168.1.66, server 192.168.1.254
Fri Oct 14 22:48:48 2022 daemon.notice netifd: wwan (3807): udhcpc: lease of 192.168.1.66 obtained from 192.168.1.254, lease time 86400
Fri Oct 14 22:48:48 2022 daemon.notice netifd: Interface 'wwan' is now up
Fri Oct 14 22:48:48 2022 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Fri Oct 14 22:48:48 2022 daemon.info dnsmasq[1]: using nameserver 192.168.1.254#53
Fri Oct 14 22:48:48 2022 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Oct 14 22:48:48 2022 daemon.info dnsmasq[1]: using only locally-known addresses for onion
[... cut ...]
Fri Oct 14 23:01:33 2022 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Oct 14 23:01:33 2022 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Oct 14 23:01:33 2022 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Oct 14 23:01:33 2022 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Oct 14 23:01:33 2022 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Oct 14 23:01:33 2022 user.notice firewall: Reloading firewall due to ifup of wwan (wlan1)

tracnunya, 2022-11-14:
There are two separate issues, that if not being taken into consideration, will prevent us from seeing that there are in fact two different issues at play here.

1- This only has to do with MODE = ACCESS POINT

2- When testing do NOT set NETWORK=LAN, instead, type a new fake name in and hit enter, creating a new fake interface, otherwise, the radio will disable masking further diagnostics.

I have a 17 station batman mesh using ex6150v2 and a dlink dir-2660
All are running flawlessly, all are running the latest release openwrt

There are two radio definitions on the 5g radio, one for the mesh, and one as a client.

If I MANUALLY add a ACCESS POINT, or two or three, there is no issue they will work, each attached to a new fake network interface TEST 1 TEST2 TEST3 etc

The issue has to do with a SCAN that is then used to create an ACCESS POINT, the scan will stop working after the SAVE
important and different than previously described ----- if the new access point network is fake the radio DOES NOT DISABLE, in the previous diagnostics the new thing was attached to the lan and that takes the radio down regardless of what or how the new radio definition is.

So, after creating this new access point with a fake net using scan, it takes twice as long to save, the access point comes right up, but now LUCI is non responsive and the scan will not work nor will the graphing of the radios, for some time, then they will work again.

Cut and dry, there is no big issue here anymore. the graphic display of channels will work and the radios work just fine.

tracnunya, 2022-11-15:
https://youtu.be/s5HPTCyvbKE