openwrt/openwrt #11000: ipq806x - NBG6817 bootloop with current master

koerhen, 2022-10-18:
With the current master (r20964-b58f3c573d) NBG6817 (only luci added) enters into a bootloop. Wifi LEDS are flashing yellow and then reboot. I have no serial cable / connectors the hardware, so I cannot provide a bootlog. 
To unbrick the router, tftp via 192.168.1.99 was required with an older working image.

Config.Buildinfo
---------------------
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_zyxel_nbg6817=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y

Note: A previous master version on 15.05 works on the  NBG6817 (when it was still a testing kernel):
Hostname	OpenWrt
Model	ZyXEL NBG6817
Architecture	ARMv7 Processor rev 0 (v7l)
Target Platform	ipq806x/generic
Firmware Version	OpenWrt SNAPSHOT r20263-f443e9de70 / LuCI Master git-22.213.35850-abd9125
Kernel Version	5.15.59
Local Time	2022-10-18 10:43:46
Uptime	0h 13m 44s
Load Average	0.04, 0.14, 0.18

ynezz, 2022-10-19:
> I have no serial cable / connectors the hardware, so I cannot provide a bootlog.

Since you're compiling the firmware yourself, would you be able to [`git bisect`](https://git-scm.com/docs/git-bisect) the issue? It should be roughly 9 following steps (Bisecting: 350 revisions left to test after this (roughly 9 steps)):

1. you start with `git bisect start b58f3c573d f443e9de70`
2. compile the image and try to boot it
3a. if it boots, continue with `git bisect good` and 2.
3b. if it doesn't boot, continue with `git bisect bad` and 2.

At some point you should find out which commit did broke your device, then paste here output of `git bisect log` command.

koerhen, 2022-10-20:
Nice feature of git albeit a somewhat painful exercise... The culprit seems to be part of the battery of changes when switching ipq806x to the 5.15 kernel by default. 

I ran into build issues within the battery of these commits because I guess these commits (e.g. dts changes) are done in one shot and some of them are interdependent. So the bisect approach lead to non compiling "interim" versions. 

I assumed "git bisect bad" in order to continue and verified with one iteration on top (the first commit after the battery of changes) that after the bunch of changes the bootloop issue exists : 
HEAD is now at cd93980abb ipq40xx: Fix wrong GPIO for internal status LED on ZTE MF289

 
Build issue,e.g. build error 900542698c
=============================================
  DTC     arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb
Error: arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dts:140.1-11 Label or path nand_pins not found
FATAL ERROR: Syntax error parsing input tree
make[6]: *** [scripts/Makefile.lib:347: arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb] Error 1

  
git bisect log
==============
# bad: [b58f3c573d9850e6da12bb0dec9445f20a55745d] arm-trusted-firmware-mvebu: add Methode eDPU support
# good: [f443e9de7003c00a935b9ea12f168e09e83b48cd] zlib: backport null dereference fix
git bisect start 'b58f3c573d' 'f443e9de70'
# good: [49763d907b0d87769c3e5885f1c11b5b62ccfa2b] kernel: add support for mtdsplit-fit offset
git bisect good 49763d907b0d87769c3e5885f1c11b5b62ccfa2b
# good: [eb07020de2b4a5f89579f09f5060d4b9f070a356] mac80211: fix decap offload for stations on AP_VLAN interfaces
git bisect good eb07020de2b4a5f89579f09f5060d4b9f070a356
# good: [f38276c9be5e235cb6fbd222e2d86e108f55eb1e] mediatek: filogic: enable thermal, I2C and PWM of the BPi-R3
git bisect good f38276c9be5e235cb6fbd222e2d86e108f55eb1e
# bad: [cd93980abb0c6452fd3e9c1213caaf26d221fcd4] ipq40xx: Fix wrong GPIO for internal status LED on ZTE MF289F
git bisect bad cd93980abb0c6452fd3e9c1213caaf26d221fcd4
# good: [faa79f97b6040fca7f5a72b7b86c6f10d03b51f8] ipq806x: split files dir to 5.10 and 5.15
git bisect good faa79f97b6040fca7f5a72b7b86c6f10d03b51f8
# bad: [900542698c34ec70669bd524c410733771087fee] ipq806x: 5.15: remove useless spm patch
git bisect bad 900542698c34ec70669bd524c410733771087fee
# bad: [cd619eeff2cb18907b69c26ca496f558575de9ba] ipq806x: 5.15: replace gcc fixes with upstream version
git bisect bad cd619eeff2cb18907b69c26ca496f558575de9ba
# good: [bd02eb75020a5cb73e1ddda91e38f98668cddb4f] ipq806x: 5.15: remove PCI_DOMAINS patch not needed anymore
git bisect good bd02eb75020a5cb73e1ddda91e38f98668cddb4f
# bad: [61a2bedf7902c64a47a42a3d8df3350f1b67ee2f] ipq806x: 5.15: replace lcc patch with upstream version
git bisect bad 61a2bedf7902c64a47a42a3d8df3350f1b67ee2f
# bad: [88bf6525251f7ef3d4f0b63e2cf77b42078101a4] ipq806x: 5.15: replace dtsi patches with upstream version
git bisect bad 88bf6525251f7ef3d4f0b63e2cf77b42078101a4
# first bad commit: [88bf6525251f7ef3d4f0b63e2cf77b42078101a4] ipq806x: 5.15: replace dtsi patches with upstream version

ynezz, 2022-10-20:
> first bad commit: [https://github.com/openwrt/openwrt/commit/88bf6525251f7ef3d4f0b63e2cf77b42078101a4] ipq806x: 5.15: replace dtsi patches with upstream version

@Ansuel any idea what might cause the issue?

Ansuel, 2022-10-20:
@ynezz i'm waiting a log from another user so I was aware of that... I will try to repro this on a r7800... I have a theory.

Ansuel, 2022-10-23:
the problem for now seems to be related to smem and for nbg sddc1... a bit problematic to repro since on my r7800 smem table is wrong and by using a squashfs image i will end up writing in the artbak partition... so i'm a bit stuck with smem partition detected in initramfs but i can't really test with squashfs.

pkgadd, 2022-10-24:
Btw. I did have v5.15 running on the nbg6817 previously (r19838-acbe1d4d36/ v5.15.48, with DSA patches and several other ones), successfully and for many weeks (not quite sure if I can track down the exact patch stack though).

closed 2022-10-25 by openwrt-bot

digitalcircuit, 2022-10-27:
Much belated update - I can confirm the latest prebuilt snapshot ([`r21112-eb456aedfe`](https://github.com/openwrt/openwrt/commit/eb456aedfe24a076a4f53743ad3c090ae447329e )) works on my NBG6817.

Thank you!  I really appreciate the effort to keep this device alive and modern.

(Pardon the delay - I had this on my ToDo list since I was pinged in OFTC/`#openwrt-devel` several days ago, but I had to set aside some time since I had moved my serial-console-equipped NBG6817 back to being my primary router.)

commit 1c514f05 (2022-10-25): ipq806x: 5.15: drop mmc-ddr-1_8v from sdcc1 node

Zyxel NGB6817 is the only router that use mmc for rootfs. Upstream
kernel dtsi have mmc-ddr-1_8v enabled for sddc1. This is wrong as mmc on
ipq806x is supplied by a fixed 3.3v regulator and can't operate at 1.8v.
This cause the sddc1 to malfunction and cause kernel panic.
In old 5.15 version this was disabled but it was put in addition to many
other changes so it was dropped silently. Restore this patch to fix
working condition of such router.

Fixes: 88bf652 ("ipq806x: 5.15: replace dtsi patches with upstream version")
Fixes: #11000
Tested-by: [... cut]
PR #11032 (2022-10-25): ipq806x: 5.15: multiple fixup related to smem and others
When 5.15 kernel was refreshed, a new implementation for smem was used, but I didn't notice that this was supported only with kernel 5.16. A backport patch should fix that.

Also backport one mtd parser patch that mute a confusing warning and revert unwanted DSA conversion for asrock G10.

Fixes: #11000

---

I would love some testing from the smem based device if this fix the bootloop... I tested this on my r7800 using smem and partition gots correctly detected with this.