openwrt/openwrt #11001: Add r8168 NIC driver.

lj0307222, 2022-10-18:
For example, please add r8168 network card driver support

timcoop3r, 2022-10-22:
you can just install r8168 or r8169 with opkg, either once openwrt is running, or using the image builder to pre-package it into an image for your device.

brada4, 2022-10-28:
Typically you need both:
```
kmod-r8169 - 5.10.146-1 - Kernel modules for RealTek RTL-8169 PCI Gigabit Ethernet adapters
r8169-firmware - 20220411-1 - RealTek RTL8169 firmware
```
Most cards are on cheap end and do not have enough ROM for firmware by themselves, or the firmware is old at best.
If that does not work, please post PCI device numbers from `lspci -n -v` (`lspci -v` would show bus connection point for the device, but driver will need the XXXX:YYYY number pair)

981213, 2022-12-09:
You can use kmod-r8169 for realtek PCIE NIC as explained in other comments. :)

closed 2022-12-09 by 981213