openwrt/openwrt #11002: TP-Link TL-WDR4900 v1 (mpc85xx/p1010) wifi config set to firstboot defaults after upgrade to 21.02.5

tim-seoss, 2022-10-18:
On upgrading a  TPLink WRT4900v1  from 21.02.3 to 21.02.5, a default wifi config is appended to `/etc/config/wireless`, leaving the wifi disabled.

```diff
--- etc/config/wireless.21.02.3	2022-10-09 15:28:01.000000000 +0100
+++ etc/config/wireless.21.02.5	2022-10-18 10:04:40.340137205 +0100
@@ -5,14 +5,11 @@
 	option dfs 'true'
 	option path 'ffe09000.pcie/pci9000:00/9000:00:00.0/9000:01:00.0'
 	option log_level '3'
-#	option txpower '20'
 	option htmode 'HT40'
 	option channel '149'
-#	option channel 'auto'
 	option cell_density '0'
 	option band '5g'
 
-
 config wifi-iface 'wifinet0'
 	option device 'radio0'
 	option mode 'ap'
@@ -89,3 +86,34 @@
 	option encryption 'sae-mixed'
 	option key 'XXXXXXXXXX'
 	option disabled '1'
+
+config wifi-device 'radio2'
+	option type 'mac80211'
+	option path 'ffe09000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
+	option channel '36'
+	option band '5g'
+	option htmode 'HT20'
+	option disabled '1'
+
+config wifi-iface 'default_radio2'
+	option device 'radio2'
+	option network 'lan'
+	option mode 'ap'
+	option ssid 'OpenWrt'
+	option encryption 'none'
+
+config wifi-device 'radio3'
+	option type 'mac80211'
+	option path 'ffe0a000.pcie/pci0001:02/0001:02:00.0/0001:03:00.0'
+	option channel '1'
+	option band '2g'
+	option htmode 'HT20'
+	option disabled '1'
+
+config wifi-iface 'default_radio3'
+	option device 'radio3'
+	option network 'lan'
+	option mode 'ap'
+	option ssid 'OpenWrt'
+	option encryption 'none'
+
```
It looks like kernel or device tree changes have resulted in the renumbering of the pci devices for the wifi radios, and the upgrade does not renumber existing entries to suit the new config.  Manually editing the config fixes this.

It looks like there are lines in `target/linux/mpc85xx/base-files/etc/hotplug.d/ieee80211/05-wifi-migrate` which do the opposite of the required fix.  I'll try reversing these.
```
b452af23a86 (Matthias Schiffer 2018-09-04 21:27:27 +0200 56)    migrate_radio 'ffe09000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0' 'ffe09000.pcie/pci9000:00/9000:00:00.0/9000:01:00.0'
b452af23a86 (Matthias Schiffer 2018-09-04 21:27:27 +0200 57)    migrate_radio 'ffe0a000.pcie/pci0001:02/0001:02:00.0/0001:03:00.0' 'ffe0a000.pcie/pcia000:02/a000:02:00.0/a000:03:00.0'
```

Hurricos, 2022-11-01:
Looks like the same backport that we saw in #10554 is being brought back even *further*. You should use the same method as in #10554, i.e. drop the PCI aliases in the device tree so they aren't picked up for PCI device addressing.

Ansuel, 2023-02-08:
Fixed by #11005

closed 2023-02-08 by Ansuel