openwrt/openwrt #11006: dnsmasq catching SIGTERM repeatedly on OpenWRT 22.03.2

clvgt12, 2022-10-18:
Router configuration:
```

Release 22.03.2 is currently running
A current stable release available for download is 22.03.2
A current stable release available for download is 21.02.5
OpenWRT runtime environment: openwrt ath79 generic
Hardware environment:
	Model ID=tplink_archer-a7-v5
	Model Name=tp-link_archer_a7_v5
	Board=tplink_archer-a7-v5
```
Packages installed:
```
#opkg list-installed
at - 3.2.2-1
ath10k-board-qca988x - 20220411-1
ath10k-firmware-qca988x-ct - 2020-11-08-1
base-files - 1494-r19803-9a599fee93
busybox - 1.35.0-4
ca-bundle - 20211016-1
ca-certificates - 20211016-1
cgi-io - 2022-08-10-901b0f04-21
ddns-scripts - 2.8.2-25
ddns-scripts-services - 2.8.2-25
dnsmasq - 2.86-14
dropbear - 2022.82-2
firewall4 - 2022-10-14-4fbf6d75-1
fstools - 2022-06-02-93369be0-2
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2021-08-03-205defb5-2
hostapd-common - 2022-01-16-cff80b4f-13.1
https-dns-proxy - 2022-10-15-1
iw - 5.16-1
iwinfo - 2022-08-19-0dad3e66-1
jansson4 - 2.13.1-2
jq - 1.6-2
jshn - 2022-05-15-d2223ef9-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.10.146-1-72e39b50629eaf33fc9600ab703ee2ab
kmod-ath - 5.10.146+5.15.58-1-1
kmod-ath10k-ct - 5.10.146+2022-05-13-f808496f-1
kmod-ath9k - 5.10.146+5.15.58-1-1
kmod-ath9k-common - 5.10.146+5.15.58-1-1
kmod-cfg80211 - 5.10.146+5.15.58-1-1
kmod-crypto-aead - 5.10.146-1
kmod-crypto-ccm - 5.10.146-1
kmod-crypto-cmac - 5.10.146-1
kmod-crypto-crc32c - 5.10.146-1
kmod-crypto-ctr - 5.10.146-1
kmod-crypto-gcm - 5.10.146-1
kmod-crypto-gf128 - 5.10.146-1
kmod-crypto-ghash - 5.10.146-1
kmod-crypto-hash - 5.10.146-1
kmod-crypto-hmac - 5.10.146-1
kmod-crypto-manager - 5.10.146-1
kmod-crypto-null - 5.10.146-1
kmod-crypto-rng - 5.10.146-1
kmod-crypto-seqiv - 5.10.146-1
kmod-crypto-sha256 - 5.10.146-1
kmod-gpio-button-hotplug - 5.10.146-3
kmod-hwmon-core - 5.10.146-1
kmod-lib-crc-ccitt - 5.10.146-1
kmod-lib-crc32c - 5.10.146-1
kmod-mac80211 - 5.10.146+5.15.58-1-1
kmod-nf-conntrack - 5.10.146-1
kmod-nf-conntrack6 - 5.10.146-1
kmod-nf-flow - 5.10.146-1
kmod-nf-log - 5.10.146-1
kmod-nf-log6 - 5.10.146-1
[... 177 lines cut ...]
Tue Oct 18 10:21:21 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:21:26 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:21:36 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:21:42 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:21:51 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:21:56 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:22:06 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:22:11 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:22:19 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:22:25 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:23:00 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:23:05 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:23:14 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:23:19 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:06 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:11 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:19 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:25 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:35 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:40 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:48 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 10:25:53 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
root@tomato:~/scripts# date
Tue Oct 18 10:26:05 EDT 2022
```

clvgt12, 2022-10-18:
Update,  dnsmasq is also catching SIGTERM repeatedly on Release 22.03.0.   Note the https-dns-proxy package is loaded and configured.

```
# do_sysupgrade.sh 
Usage: /root/scripts/do_sysupgrade.sh [RELEASE_LABEL]
Release 22.03.0 is currently running
A current stable release available for download is 22.03.2
A current stable release available for download is 21.02.5
OpenWRT runtime environment: openwrt ath79 generic
Hardware environment:
	Model ID=tplink_archer-a7-v5
	Model Name=tp-link_archer_a7_v5
	Board=tplink_archer-a7-v5
root@tomato:~/scripts# logread|grep dnsmasq | grep SIGTERM
Tue Oct 18 12:57:17 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:23 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:29 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:34 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:40 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:37 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:43 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
root@tomato:~/scripts# date
Tue Oct 18 12:58:51 EDT 2022
root@tomato:~/scripts# date
Tue Oct 18 12:59:11 EDT 2022
root@tomato:~/scripts# logread|grep dnsmasq | grep SIGTERM
Tue Oct 18 12:57:17 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:23 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:29 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:34 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:40 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:37 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:43 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:52 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:57 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:05 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:10 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
root@tomato:~/scripts# logread|grep dnsmasq | grep SIGTERM
Tue Oct 18 12:57:17 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:23 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:29 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:34 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:57:40 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:37 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:43 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:52 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:58:57 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:05 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:10 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:22 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:27 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:35 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 12:59:40 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
```

ynezz, 2022-10-18:
> Note the https-dns-proxy package is loaded and configured.

This package comes from [packages feed](https://github.com/openwrt/packages), there is [separate issue tracker](https://github.com/openwrt/packages) there.

closed 2022-10-18 by ynezz