openwrt/openwrt #11009: fw4 ipset infinite loop bricks router after upgrading from 22.03.1 to 22.03.2

smferris, 2022-10-18:
My firewall config had been using ipsets with fw3, and I've continued using them with fw4 unchanged.

When I upgraded from 22.03.1 to 22.03.2, the router would stop responding to the network shortly after booting (ping worked for a couple of seconds, then I'd never get a reply again).

After reflashing a factory image over tftp and reloading my config backup except for /etc/config/firewall, I got the router working again.

I experimented with updating the firewall config and testing it with `fw4 print`, and the problem seems to be an infinite loop in fw4.uc parse_setfile.  It's expecting EOF to be indicated by an empty string for the variable line, but it's getting 'null' instead, so it loops forever:

```
Skipping invalid entry 'null' in file '/etc/ssh.ipset' for set 'ssh'
Skipping invalid entry 'null' in file '/etc/ssh.ipset' for set 'ssh'
Skipping invalid entry 'null' in file '/etc/ssh.ipset' for set 'ssh'
...
```

```
                while ((line = fd.read("line")) !== "") {
                        line = trim(line);

                        if (length(line) == 0 || ord(line) == 35)
                                continue;

                        let v = this.parse_ipsetentry(line, set);

                        if (!v) {
                                this.warn(`Skipping invalid entry '${line}' in file '${set.loadfile}' for set '${set.name}'`);
                                continue;
                        }

                        cb(v);

                        count++;
                }
```

smferris, 2022-10-18:
This seems to work for me:
```
diff --git i/root/usr/share/ucode/fw4.uc w/root/usr/share/ucode/fw4.uc
index 2a1e397..63ddfe2 100644
--- i/root/usr/share/ucode/fw4.uc
+++ w/root/usr/share/ucode/fw4.uc
@@ -1787,7 +1787,7 @@ return {

                let line = null, count = 0;

-               while ((line = fd.read("line")) !== "") {
+               while ((line = fd.read("line")) !== null) {
                        line = trim(line);

                        if (length(line) == 0 || ord(line) == 35)
```

jow-, 2022-10-18:
Issue has been fixed in firewall4.git and package updates have been issued for both Snapshots and OpenWrt 22.03

closed 2022-10-18 by jow-