openwrt/openwrt #11010: Devices with MT7530 switches support GPIO LED control, but netdev does not see hw offloaded bridge traffic

alexhorner, 2022-10-18:
Devices with the MT7530 switch chipset can have the chipset's LED pins controlled by OpenWrt as GPIO.

This works perfectly fine when each port on the switch is being used as a separate device in OpenWrt, as netdev can be configured to control the GPIO for link, tx and rx like any other LED.

This does not work properly if two or more ports are configured to be bridged together. Each LED on each port will correctly show the link status, but will only show tx and rx traffic when the traffic passes through to OpenWrt. Traffic which traverses the bridge between devices does not cause LED activity because the job of bridging the ports is offloaded to the switch chipset, leaving netdev unaware of that traffic traversal.

This issue has been opened in reference to discussion on #10829

Djfe, 2022-12-26:
This likely depends on something like to be implemented in upstream
https://lwn.net/ml/linux-kernel/20220503151633.18760-1-ansuelsmth@gmail.com/
If I understand @neheb correctly.
So this issue can be used to track the progress of Ansuel's work regarding switch leds.

neheb, 2022-12-26:
@arinc9 had a PR that removed GPIO control for this reason. I think it was merged. Not sure.

@ansuel 's patchset is a solution that avoids using GPIO.

Djfe, 2022-12-26:
Thanks for clarifying.
Are you talking about arinc9's RGMII-related commits? That was related to GPIO but not LEDs.

anyways, I could remove switch LEDs from all 7621 boards with a PR.
I would look for this to be in a .dts file within the ramips target and then clear up the dts as well as `board.d/01_leds`
```
&switch0 {
	gpio-controller;
	#gpio-cells = <2>;
```
https://github.com/openwrt/openwrt/search?q=%26switch0+%7B+%09gpio-controller%3B (so this, but without the realtek target)

Is this useful/requested?
Are there other targets like Mediatek/MT7622 that need to be adjusted as well? (as far as I can see the Linksys E8450 for example isn't affected by this, is it?)

neheb, 2022-12-26:
I was talking about http://patchwork.ozlabs.org/project/openwrt/list/?series=323510

Djfe, 2022-12-26:
Ah thx, that wasn't merged, yet.
https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_linksys_e7350.dts
The last commit to that file is yours from September.

arinc9, 2022-12-26:
> I was talking about http://patchwork.ozlabs.org/project/openwrt/list/?series=323510

@hauke casual reminder to apply this ;)