openwrt/openwrt #11017: ath79 Netgear WNDAP360 fix serial console speed at 9600

droncheg, 2022-10-19:
Hello. In current master trunk serial console speed changes from 9600 to 115200 after bootconsole disabled (approx after 3sec kernel start). Please change DTS for WNDAP360 to fix serial console at 9600, as it's default speed and indiced near serial console RJ45 port on case.

Proposed changes described here https://forum.openwrt.org/t/wndap360-with-current-22-03-trunk/140029/2

```
--- a/target/linux/ath79/dts/ar7161_netgear_wndap360.dts
+++ b/target/linux/ath79/dts/ar7161_netgear_wndap360.dts
@@ -9,6 +9,10 @@
 	compatible = "netgear,wndap360", "qca,ar7161";
 	model = "Netgear WNDAP360";
 
+	chosen {
+		bootargs = "console=ttyS0,9600";
+	};
+
 	aliases {
 		led-boot = &led_power_orange;
 		led-failsafe = &led_power_orange;
```

Also, please check connectivity via ethernet port after trunk firmware applied - it doesnt work at all.

neheb, 2026-04-08:
cool. https://github.com/openwrt/openwrt/pull/22769 will fix it for master and 25.12 when merged.

neheb, 2026-04-08:
Extra commits needed:

https://github.com/openwrt/openwrt/commit/469c6676b7473802177b2d7300a864f6b6680226
https://github.com/openwrt/openwrt/commit/52d776ac968f59149afcd65f8aa1a1d130343ace
https://github.com/openwrt/openwrt/commit/8d1b2596ee9c9794bd4467871e2746c9479db58c