openwrt/openwrt #11020: X64 master tree can not be completed because of  util-linux bug

openips, 2022-10-20:
commit b1cc4eac1bc82086cb2569deecb1c68678011ab3
X64 master tree can not be completed because of  util-linux 

log below:
```
The Meson build system
Version: 0.61.5
Source dir: /LEDE/openwrt-psw/build_dir/target-x86_64_musl/util-linux-2.38
Build dir: /LEDE/openwrt-psw/build_dir/target-x86_64_musl/util-linux-2.38/openwrt-build
Build type: cross build
Project name: util-linux
Project version: 2.38
C compiler for the host machine: x86_64-openwrt-linux-musl-gcc (gcc 11.3.0 "x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r0+21011-b1cc4eac1b) 11.3.0")
C linker for the host machine: x86_64-openwrt-linux-musl-gcc ld.bfd 2.37
C compiler for the build machine: gcc (gcc 9.4.0 "gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the build machine: gcc ld.bfd 2.34
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: generic
Target machine cpu family: x86_64
Target machine cpu: generic
Dependency bash-completion skipped: feature build-bash-completion disabled
WARNING: You should add the boolean check kwarg to the run_command call.
         It currently defaults to false,
         but it will default to true in future releases of meson.
         See also: https://github.com/mesonbuild/meson/issues/9300
Checking if "wchar_t support" compiles: YES
Library m found: YES
Dependency tinfo skipped: feature tinfo disabled
Found pkg-config: /LEDE/openwrt-psw/staging_dir/host/bin/pkg-config (1.9.3)
Run-time dependency ncursesw found: YES 6.3.20211021
Dependency slang skipped: feature slang disabled
Checking for function "use_default_colors" with dependency ncursesw: YES
Checking for function "resizeterm" with dependency ncursesw: YES
Run-time dependency zlib found: YES 1.2.12
Dependency readline skipped: feature readline disabled
Dependency libuser skipped: feature libuser disabled
Library util skipped: feature libutil disabled
Library utempter skipped: feature libutempter disabled
Dependency systemd skipped: feature systemd disabled
Dependency libsystemd skipped: feature systemd disabled
Dependency libudev skipped: feature systemd disabled
Library crypt found: YES
Library pam found: NO
Dependency libcryptsetup skipped: feature cryptsetup disabled
Checking for function "crypt_activate_by_signed_key" with dependency libcryptsetup: NO
Dependency libcap-ng skipped: feature build-setpriv disabled
Dependency libselinux skipped: feature selinux disabled
Dependency libmagic skipped: feature magic disabled
Found CMake: /LEDE/openwrt-psw/staging_dir/host/bin/cmake (3.24.2)
make[3]: Leaving directory '/LEDE/openwrt-psw/package/utils/util-linux'
time: package/utils/util-linux/compile#2.33#0.67#3.19
    ERROR: package/utils/util-linux failed to build.
```

ynezz, 2022-10-20:
> ERROR: package/utils/util-linux failed to build.

Re-run the build with `make package/util-linux V=s` and paste the error here.

openips, 2022-10-20:
Re-run the build with make package/util-linux V=s 
error log below
```
song@ubantu:/LEDE/openwrt-psw$ make package/util-linux V=s
make[2]: Entering directory '/LEDE/openwrt-psw/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/LEDE/openwrt-psw/scripts/config'
time: target/linux/prereq#0.09#0.03#0.14
make[1]: Entering directory '/LEDE/openwrt-psw'
make[1]: *** No rule to make target 'package/util-linux'.  Stop.
make[1]: Leaving directory '/LEDE/openwrt-psw'
make: *** [/LEDE/openwrt-psw/include/toplevel.mk:231: package/util-linux] Error 2

```

openips, 2022-10-20:
it's my mistake. i set the .config file with wrong permissions 
thks your good idea to let me find my  mistake

ynezz, 2022-10-20:
> make[1]: *** No rule to make target 'package/util-linux'.  Stop.

Sorry, I've provided you with wrong command, it should've been `make package/util-linux/compile V=s`