openwrt/openwrt #11025: ath79 Netgear WNDAP360 fix ethernet port

GrandDixence, 2022-10-20:
An upgrade from OpenWrt version 19.07.9 to version 22.03.1 by `sysupgrade -v -n --force /tmp/openwrt-22.03.1-ath79-generic-netgear_wndap360-squashfs-sysupgrade.bin` breaks connectivity via ethernet port of Netgear WNDAP360. Please fix this issue. I'm not alone with this issue. See ethernet connectivity remarks at:
https://github.com/openwrt/openwrt/issues/11017

Ethernet packet counters shown by `ifconfig -a` run up. But the Netgear WNDAP360 doesn't send any ethernet frames through the ethernet cable. I can't capture any arp replies/responses from Netgear WNDAP360 with Wireshark. I can't capture any arp requests from Netgear WNDAP360 with Wireshark.  

```
# more /etc/config/network

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdef:fa92:ea46::/48'

config interface 'wan'
	option ifname 'eth0.9'
	option proto 'static'
	option ipaddr '192.168.9.50'
	option netmask '255.255.255.0'
	option gateway '192.168.9.1'
	option ip6assign '60'

config interface 'lan6'
	option type 'bridge'
	option ifname 'eth0.6'
	option proto 'static'
	option ip6assign '60'

config interface 'lan8'
        option type 'bridge'
        option ifname 'eth0.8'
        option proto 'static'
        option ip6assign '60'

config interface 'lan11'
        option type 'bridge'
        option ifname 'eth0.11'
        option proto 'static'
        option ip6assign '60'

config interface 'lan12'
        option type 'bridge'
        option ifname 'eth0.12'
        option proto 'static'
        option ip6assign '60'

```

```
# ip link show dev eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 00:03:7f:e0:00:96 brd ff:ff:ff:ff:ff:ff
```

```
# ip link show dev eth0.9
13: eth0.9@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 00:03:7f:e0:00:96 brd ff:ff:ff:ff:ff:ff
```

```
sysctl
net.ipv4.conf.eth0.arp_accept = 0
net.ipv4.conf.eth0.arp_announce = 0
net.ipv4.conf.eth0.arp_filter = 0
net.ipv4.conf.eth0.arp_ignore = 0
net.ipv4.conf.eth0.arp_notify = 0
net.ipv4.conf.eth0.drop_gratuitous_arp = 0
net.ipv4.conf.eth0.proxy_arp = 0
net.ipv4.conf.eth0.proxy_arp_pvlan = 0

net.ipv4.conf.eth0/9.arp_accept = 0
net.ipv4.conf.eth0/9.arp_announce = 0
net.ipv4.conf.eth0/9.arp_filter = 0
net.ipv4.conf.eth0/9.arp_ignore = 1
net.ipv4.conf.eth0/9.arp_notify = 0
net.ipv4.conf.eth0/9.drop_gratuitous_arp = 0
net.ipv4.conf.eth0/9.proxy_arp = 0
net.ipv4.conf.eth0/9.proxy_arp_pvlan = 0

net.ipv4.conf.default.arp_accept = 0
net.ipv4.conf.default.arp_announce = 0
net.ipv4.conf.default.arp_filter = 0
net.ipv4.conf.default.arp_ignore = 1
net.ipv4.conf.default.arp_notify = 0
net.ipv4.conf.default.drop_gratuitous_arp = 0
net.ipv4.conf.default.proxy_arp = 0
net.ipv4.conf.default.proxy_arp_pvlan = 0

net.ipv4.conf.all.arp_accept = 0
net.ipv4.conf.all.arp_announce = 0
net.ipv4.conf.all.arp_filter = 0
net.ipv4.conf.all.arp_ignore = 1
net.ipv4.conf.all.arp_notify = 0
net.ipv4.conf.all.drop_gratuitous_arp = 0
net.ipv4.conf.all.proxy_arp = 0
```

```
# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 00:03:7F:E0:00:96  
          inet6 addr: fe80::203:7fff:fee0:96/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9 errors:0 dropped:0 overruns:0 frame:0
          TX packets:83 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:540 (540.0 B)  TX bytes:4202 (4.1 KiB)
          Interrupt:4 
```

```
# ifconfig eth0.9
eth0.9    Link encap:Ethernet  HWaddr 00:03:7F:E0:00:96  
          inet addr:192.168.9.50  Bcast:192.168.9.255  Mask:255.255.255.0
          inet6 addr: fe80::203:7fff:fee0:96/64 Scope:Link
          inet6 addr: fdef:fa92:ea46:40::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9 errors:0 dropped:0 overruns:0 frame:0
          TX packets:77 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:378 (378.0 B)  TX bytes:3378 (3.2 KiB)
```

It looks like the Netgear WNDAP360 is able to receive ethernet frames. 
```
# arp -a
IP address       HW type     Flags       HW address            Mask     Device
192.168.9.1      0x1         0x0         00:a0:57:1b:9f:6a     *        eth0.9
```

```
# dmesg |grep -i eth0
[    4.376189] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii
[   13.603491] eth0: link up (1000Mbps/Full duplex)
[   13.608140] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   15.634231] eth0: link down
[   38.625556] br-lan11: port 1(eth0.11) entered blocking state
[   38.631225] br-lan11: port 1(eth0.11) entered disabled state
[   38.637251] device eth0.11 entered promiscuous mode
[   38.642130] device eth0 entered promiscuous mode
[   39.001571] br-lan12: port 1(eth0.12) entered blocking state
[   39.007291] br-lan12: port 1(eth0.12) entered disabled state
[   39.013276] device eth0.12 entered promiscuous mode
[   39.226145] br-lan6: port 1(eth0.6) entered blocking state
[   39.231636] br-lan6: port 1(eth0.6) entered disabled state
[   39.281377] device eth0.6 entered promiscuous mode
[   39.490240] br-lan8: port 1(eth0.8) entered blocking state
[   39.495802] br-lan8: port 1(eth0.8) entered disabled state
[   39.501582] device eth0.8 entered promiscuous mode
[   41.763527] eth0: link up (1000Mbps/Full duplex)
[   41.768181] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   41.863136] br-lan6: port 1(eth0.6) entered blocking state
[   41.868629] br-lan6: port 1(eth0.6) entered forwarding state
[   41.874655] br-lan8: port 1(eth0.8) entered blocking state
[   41.880133] br-lan8: port 1(eth0.8) entered forwarding state
[   41.886119] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.9: link becomes ready
[   41.892787] br-lan11: port 1(eth0.11) entered blocking state
[   41.898498] br-lan11: port 1(eth0.11) entered forwarding state
[   41.904716] br-lan12: port 1(eth0.12) entered blocking state
[   41.910370] br-lan12: port 1(eth0.12) entered forwarding state
```

```
# dmesg |grep -i ag71
[    4.366691] ag71xx 19000000.eth: connected to PHY at mdio.0:01 [uid=004dd04e, driver=Generic PHY]
[    4.376189] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii
```

```
# dmesg |grep -i phy
[    4.366689] ag71xx 19000000.eth: connected to PHY at mdio.0:01 [uid=004dd04e, driver=Generic PHY]
[   19.658424] ath: phy0: Ignoring endianness difference in EEPROM magic bytes.
[   19.680878] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   19.682966] ieee80211 phy0: Atheros AR9280 Rev:2 mem=0xb0000000, irq=15
[   19.957874] ath: phy1: Ignoring endianness difference in EEPROM magic bytes.
[   19.982447] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   19.984647] ieee80211 phy1: Atheros AR9280 Rev:2 mem=0xb0010000, irq=16
```

```
# dmesg |grep -i mdio
[    4.366689] ag71xx 19000000.eth: connected to PHY at mdio.0:01 [uid=004dd04e, driver=Generic PHY]
```

```
# dmesg |grep -i 8021q
[    4.421769] 8021q: 802.1Q VLAN Support v1.8
```

walterav1984, 2022-10-23:
Just for reference original [ath79 port](https://github.com/openwrt/openwrt/pull/9685)

neheb, 2022-10-24:
Ping @PolynomialDivision

droncheg, 2022-10-28:
I fixed that with
`pll-data = <0x11110000 0x00001099 0x00991099>;`
in eth0 section of `target/linux/ath79/dts/ar7161_netgear_wndap360.dts`, but there's more - no wlans (wifi) at all. I suppose DTS is incomplete now. I am not qualified to solve this problem.

PolynomialDivision, 2022-12-22:
For me the network works fine? Just tested latest master branch.

droncheg, 2022-12-23:
> For me the network works fine? Just tested latest master branch.

are you tried at gigabit or fast ethernet speed?

PolynomialDivision, 2022-12-23:
> are you tried at gigabit or fast ethernet speed?

I think it was Gigabit. Why?

Can you maybe test:
https://github.com/openwrt/openwrt/pull/11595

droncheg, 2022-12-23:
I tested trunk 21575 with simple [config](https://openwrt.do2.pw/my_builds/firmware/snap_360_tests/360.diff) - eth doesnt work at gigabit (1000 Full), but works at fast (100 Full). Wifi works - that's cool, cause in 22.03.1 wifi also doesnt work without owl package.

Then i applied your patch to this trunk, and got same results. Original images and config [here](https://openwrt.do2.pw/my_builds/firmware/snap_360_tests/), patched [here](https://openwrt.do2.pw/my_builds/firmware/snap_360_tests/patched/).

At 22.03.1 i successfully build and all works when i replaced original dts with [this](https://openwrt.do2.pw/my_builds/firmware/22.03.2-WNDAP360/ar7161_netgear_wndap360.dts) and added manually `CONFIG_PACKAGE_kmod-owl-loader=y` to config (dont know why menuconfig didnt add this). Results are [here](https://openwrt.do2.pw/my_builds/firmware/22.03.2-WNDAP360/).

And please, submit patch to 9600 speed at serial console. It is too hard jumping terminal from 9600 to 115200 and back when debug =) It resolved with
```
chosen {
bootargs = "console=ttyS0,9600";
};
```
in dts

walterav1984, 2023-03-07:
Still need to confirm myself but @eandry77 have you tried installing `ethtool`?

```
opkg update
opkg install ethtool
ethtool -s eth0 autoneg off speed 100 duplex full #put this line in /etc/rc.local if succesfull so it works after boot
```

eandry77, 2023-04-27:
Sorry for the late answer :
"ethtool -s eth0 autoneg off speed 100 duplex full" worked when manually executed, but not in /etc/rc.local.

Does anyone know if there is a proper fix for the gigabit somewhere which can be tested ? Any pending PR  ?

PolynomialDivision, 2023-04-27:
@eandry77 Some people say that the `pll-data = <0x11110000 0x00001099 0x00991099>;` fixes the issue.

walterav1984, 2023-04-28:
@eandry77 

Have you tried running `/usr/sbin/ethtool` from `/etc/rc.local` using the whole path maybe put a `/bin/sleep 10` before it?

ewpa, 2023-06-24:
@eandry77 I found a reliable place for the `ethtool` workaround is in `/etc/hotplug.d/iface`.  Let me know if you would like the script.

droncheg, 2024-10-18:
> Hi do you think you can do the same thinks with the last firmware ?

Why not, it is similar. But I sold my access points so I won't be able to test the functionality of the new firmware

rodin-ia, 2025-05-09:
On release 24.10.1 all the problems described above are relevant + problem with overlayfs_data partition mounting. I.e. for correct AP work it is necessary to use correct dts and install kmod-owl-loader package. Just in case I have attached dts from droncheg to the comment.

[ar7161_netgear_wndap360.dts.txt](https://github.com/user-attachments/files/20125611/ar7161_netgear_wndap360.dts.txt)

neheb, 2026-04-05:
that's weird. netgear,wndap360 is in the ath79 file for ubootenv-tools.

Anyway, next build:

https://mega.nz/file/XJA1XZwC#ZqY1rlml74v-OhEocB9UbSWt3-hHxYgCAtaDncm80iQ

git diff:

```
--- a/target/linux/ath79/image/generic.mk
+++ b/target/linux/ath79/image/generic.mk
@@ -2248,13 +2248,11 @@ define Device/netgear_wndap360
   DEVICE_MODEL := WNDAP360
   DEVICE_PACKAGES := kmod-owl-loader
   IMAGE_SIZE := 7744k
-  BLOCKSIZE := 256k
+  BLOCKSIZE := 64k
   LOADER_TYPE := bin
   KERNEL := kernel-bin | append-dtb | lzma | loader-kernel | uImage none
   KERNEL_INITRAMFS := $$(KERNEL)
   IMAGES := sysupgrade.bin
-  IMAGE/sysupgrade.bin := append-kernel | pad-to 64k | append-rootfs | pad-rootfs | \
-       check-size | append-metadata
 endef
 TARGET_DEVICES += netgear_wndap360
```

changed the block size as well as the sysupgrade image to be the same as other netgear units. Smaller now and probably won't work.

ronnyadsetts, 2026-04-05:
> New build with:
> 
> ```
> --- a/target/linux/ath79/image/generic.mk
> +++ b/target/linux/ath79/image/generic.mk
> @@ -2253,8 +2253,6 @@ define Device/netgear_wndap360
>    KERNEL := kernel-bin | append-dtb | lzma | loader-kernel | uImage none
>    KERNEL_INITRAMFS := $$(KERNEL)
>    IMAGES := sysupgrade.bin
> -  IMAGE/sysupgrade.bin := append-kernel | pad-to 64k | append-rootfs | pad-rootfs | \
> -       check-size | append-metadata
>  endef
>  TARGET_DEVICES += netgear_wndap360
> ```
> 
> https://mega.nz/file/iRoWzZDK#wOpCkIFeV3hPIawhvwfxiO5J0jyAAjhtFtWTuqNP0is

Fails to boot I'm afraid(I had to mess around with baud rate as the initrd image I have for 2.10.3 is 115200 so missed some of the sysupgrade):

```
ar7100> reset

Resetting...

U-Boot 1.1.4 dni-1.07 (Jul  7 2009 - 14:17:28)

AP94 (ar7100) U-boot 0.0.12
DRAM:  b8050000: 0xc0140180
128 MB
Top of RAM usable for U-Boot at: 88000000
Reserving 228k for U-Boot at: 87fc4000
Reserving 192k for malloc() at: 87f94000
Reserving 56 Bytes for Board Info at: 87f93fc8
Reserving 36 Bytes for Global Data at: 87f93fa4
Reserving 128k for boot params() at: 87f73fa4
Stack Pointer at: 87f73f88
Now running in RAM - U-Boot at: 87fc4000
id read 0x100000ff
flash size 8MB, sector count = 128
Flash:  8 MB
In:    serial
Out:   serial
Err:   serial
Net:   ag7100_enet_initialize...
: cfg1 0xf cfg2 0x7014
 VSCXXX Found 0  unit 0:0  phy_addr: 1  id: 004dd04e
PHY:   Atheros AR8021
eth0: 00:03:7f:e0:00:96
eth0 up
eth0
Manu data is valid!
### main_loop entered: bootdelay=4

### main_loop: bootcmd="bootm 0xbf050000"
Enter SPACE to drop into boot loader:  0
## Booting image at bf050000 ...
   Image Name:   MIPS OpenWrt Linux-6.12.80
   Created:      2026-04-05  19:04:32 UTC
   Image Type:   MIPS Linux Unknown Image (uncompressed)
   Data Size:    2782208 Bytes =  2.7 MB
   Load Address: bf070000
   Entry Point:  bf070000
   Verifying Checksum ... OK
Wrong Image Type for bootm command
```

I've now reflashed with the image from https://github.com/openwrt/openwrt/issues/11025#issuecomment-4187935240 which boots.