openwrt/openwrt #11029: MT7622 (Belkin RT3200) ARP fails after roam

clutch2sft, 2022-10-20:
I have 802.11r setup with 802.1x/PEAP and local radius server.  The WiFi roaming stuff appears to work flawlessly.  However, as the client begins arp for its gateway it never sees a response.  This happens roaming both ways IE: if device connects to Router(AP) then roams to AP1 it happens .... if device connect to AP1 then roams to Router(AP) it happens.  Net result is iPhone, Android, and Windows all decide the internet is no longer available after a roam and disconnect from the wifi.

Configuration:

AP1 lan2 <-----> lan2 Router(AP)

```#begin router1
config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'HE80'
        option country 'US'
        option cell_density '0'
        option channel 'auto'
        option disabled '0'
        option txpower '5'

config wifi-iface 'wifinet4'
        option acct_port '1813'
        option acct_secret 'somesecret'
        option acct_server '127.0.0.1'
        option auth_port '1812'
        option auth_secret 'somesecret'
        option auth_server '127.0.0.1'
        option device 'radio1'
        option encryption 'wpa2'
        option ieee80211w '1'
        option mode 'ap'
        option network 'Network1'
        option ssid 'SSID'
        option wpa_disable_eapol_key_retries '1'
        option nasid 'E89F80ECD75A'
        option disassoc_low_ack '0'
        option ieee80211r '1'
        option mobility_domain '6310'
        option ft_over_ds '0'
        option ft_psk_generate_local '1'
        option r1_key_holder 'E89F80ECD75A'
        option pmk_r1_push '1'
        list r0kh 'E8:9F:80:EC:D7:5A,E89F80ECD75A,fef976bb94a0bbac396fd3f5809473b5 E8:9F:80:E9:60:24,E89F80E96024,fef976bb94a0bbac396fd3f5809473b5'
        list r1kh 'E8:9F:80:EC:D7:5A,E8:9F:80:EC:D7:5A,fef976bb94a0bbac396fd3f5809473b5 E8:9F:80:E9:60:24,E8:9F:80:E9:60:24,fef976bb94a0bbac396fd3f5809473b5'

config device 'device1'
        option bridge_empty '1'
        option name 'DotQSwitch'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option type 'bridge'

config bridge-vlan 'bridge_vlan1'
        option device 'DotQSwitch'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'
        option vlan '5'
        
config interface 'Network1'
        option proto 'static'
        option device 'DotQSwitch.5'
        option ipaddr '172.16.5.1'
        option netmask '255.255.255.0'
        option defaultroute '0'

#end router1

#begin AP1

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'HE80'
        option country 'US'
        option cell_density '0'
        option channel 'auto'
        option disabled '0'

config wifi-iface 'wifinet4'
        option acct_port '1813'
        option acct_secret 'somesecret'
        option acct_server '172.16.5.1'
        option auth_port '1812'
        option auth_secret 'somesecret'
        option auth_server '172.16.5.1'
        option device 'radio1'
        option encryption 'wpa2'
        option ieee80211w '1'
        option mode 'ap'
        option network 'Network1'
        option ssid 'SSID'
        option wpa_disable_eapol_key_retries '1'
        option nasid 'E89F80E96024'
        option disassoc_low_ack '0'
        option ieee80211r '1'
        option mobility_domain '6310'
        option ft_over_ds '0'
        option ft_psk_generate_local '1'
        option r1_key_holder 'E89F80E96024'
        option pmk_r1_push '1'
        list r0kh 'E8:9F:80:EC:D7:5A,E89F80ECD75A,fef976bb94a0bbac396fd3f5809473b5 E8:9F:80:E9:60:24,E89F80E96024,fef976bb94a0bbac396fd3f5809473b5'
        list r1kh 'E8:9F:80:EC:D7:5A,E8:9F:80:EC:D7:5A,fef976bb94a0bbac396fd3f5809473b5 E8:9F:80:E9:60:24,E8:9F:80:E9:60:24,fef976bb94a0bbac396fd3f5809473b5'

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'

config device 'device1'
        option name 'DotQSwitch'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option type 'bridge'

config bridge-vlan 'bridge_vlan1'
        option device 'DotQSwitch'
        option vlan '5'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'
        list ports 'wan:u*'

config interface 'Network1'
        option device 'DotQSwitch.5'
        option proto 'dhcp'  #dhcp serves static address (reservved)

#end AP1```

Capture attached from the Windows client with the roam happening at 09:23:24.284769.  I am at the extent of my understanding to troubleshoot further.

Happy to keep this setup around and gather any additional requested data.

cmonroe, 2022-10-22:
I had a very similar issue and can't recall what the fix is on kernel 5.10 / OpenWrt 22.03. If I remember correctly it is no longer an issue in kernel 5.15 / OpenWrt master due to upstream DSA driver changes. If possible you may want to test a snapshot build and see if the behavior persists.

If you find it is in fact fixed I can dig thru my tree a bit more to see if I happen to have a fix for this on kernel 5.10.

From what I recall, the MAC entry gets stuck on the MT7531 switch LAN port feeding your extender.. when you roam back to the root AP it will receive your packets from WiFi but forward responses out the LAN port where the MAC entry is stuck.

Another way to verify this is indeed what's happening is to setup WiFi back-haul between your two nodes instead of wired which will take the switch out of the picture. I would start by testing a master build though.. it's quite stable on Mediatek targets these days.

clutch2sft, 2022-10-25:
@cmonroe   Thank you for the insights.  I was suspecting a stuck mac somewhere but not familar enough with the architecture yet to know where to look.

I will compile from Master and report back the results - though I would like to stay on a mainline for my use case ....sometimes you have to do what you have to do.

clutch2sft, 2022-10-27:
@cmonroe I was able to fight through some bad luck (pulled master when https://github.com/openwrt/packages/pull/19701 happend) and then jumped on the struggle bus for awhile because acme renamed one their scripts and broke my compile script.  But things are on:
OpenWrt SNAPSHOT r21112-eb456aedfe / LuCI Master git-22.299.48611-81f204d 
Kernel 5.15.74

from today.   The bad and the uggly is that the problem persists on this version.  Next up - Wireless backhaul but that will be when I recover sometime tomorrow from all the things I didn't do today that I was supposed to.  :-)

ggg

clutch2sft, 2022-10-28:
@cmonroe I now stand corrected on my comment from two days ago.  After much time I find that roaming works fine for Apple and Windows devices but on this snapshot I am getting a MIC error from the client for all of my Android devices with my current settings.  Will play with that but net-net is the ARP problem does not exist with:

OpenWrt SNAPSHOT r21112-eb456aedfe / LuCI Master git-22.299.48611-81f204d
Kernel 5.15.74

closed 2022-10-28 by clutch2sft

schuettecarsten, 2022-11-17:
I come back to this issue and I can still reproduce the issue on todays version (21306).
[LuCI Master (git-22.314.78547-2cbfd64)](https://github.com/openwrt/luci) /[ OpenWrt SNAPSHOT (r21306-49bbfd9968)](https://openwrt.org/)

I can see that the client connects to the router using 2,4 GHz Wifi (MT7622 chipset) and is shown correctly in the list of connected clients in the AP "Wireless" Luci page. But then, Wifi connection is shown with "exclamation mark" (Android) and seconds later, it is completely lost.