openwrt/openwrt #11034: OpenWRT Build fail with utils/cryptsetup

Chsuma09, 2022-10-21:
Hi

Recent feeds from openwrt broken the clean build. Do you have any fix for this.

patching file rules.mk
Hunk #1 succeeded at 99 (offset -4 lines).
patching file target/linux/Makefile
Hunk #1 succeeded at 7 (offset -2 lines).
patching file package/utils/busybox/Config-defaults.in
patching file package/system/fstools/patches/0100-automount.patch
patching file package/system/fstools/patches/0102-mount-options.patch
patching file package/system/fstools/patches/0103-mtk-ntfs-mount-by-ufsd.patch
patching file package/utils/mtd-utils/Makefile
patching file scripts/make-squashfs-hashed.sh
patching file scripts/prepare-dm-verity-uboot-script.sh
patching file tools/ar-tool/Makefile
patching file tools/ar-tool/src/Makefile
patching file tools/ar-tool/src/ar-tool.py
patching file tools/Makefile
patching file feeds/packages/libs/libaio/Makefile
patching file feeds/packages/utils/cryptsetup/Makefile
Hunk #1 FAILED at 8.
Hunk #2 succeeded at 25 (offset 1 line).
Hunk #3 succeeded at 81 (offset 2 lines).
Hunk #4 succeeded at 111 (offset 2 lines).
1 out of 4 hunks FAILED -- saving rejects to file feeds/packages/utils/cryptsetup/Makefile.rej
patching file feeds/packages/utils/cryptsetup/patches/0001-dont-use-c89.patch
patching file feeds/packages/utils/cryptsetup/patches/0100-add-configure-arg-to-link-with-tiny-libiconv-in-host-build.patch
patching file feeds/packages/utils/lvm2/Makefile
patching file feeds/packages/utils/lvm2/patches/100-change-linker-search-dir-order.patch
patching file package/Makefile
Hunk #1 succeeded at 61 with fuzz 1 (offset 1 line).
patching file package/libs/popt/Makefile
patching file package/libs/popt/patches/0001-add-configure-arg-to-link-with-tiny-libiconv-in-host-build.patch

**Branch is: 2102.**

Rgds
Chandra

ynezz, 2022-10-21:
Check https://openwrt.org/bugs#issue_trackers, this bug report looks like it should've been filled in https://github.com/openwrt/packages/issues.

ynezz, 2022-10-21:
> this bug report looks like it should've been filled in https://github.com/openwrt/packages/issues.

I stand corrected, this:

>  patching file feeds/packages/utils/cryptsetup/patches/0100-add-configure-arg-to-link-with-tiny-libiconv-in-host-build.patch

Doesn't look like OpenWrt, so you should report it at the place where you've got this from.

closed 2022-10-21 by ynezz