openwrt/openwrt #11068: Asus rt-ax53u led support

lukjod, 2022-10-23:
AS for now it supports only power led and 3 lan and 1 wan led in newest 22.03.snapshot
Is this possible to add support for additional 2 wlan and usb led?

DragonBluep, 2022-10-23:
I wonder if the power led will blink when the system is booting?

lukjod, 2022-10-23:
No it just light up when sysstem is fully booted

DragonBluep, 2022-10-23:
What is the color of the usb and wlan LEDs? I will try to make a patch.

lukjod, 2022-10-23:
they are all blue including power lan wan
thx in advance

DragonBluep, 2022-10-23:
After about an hour, you can download the test firmware here:
https://github.com/DragonBluep/openwrt/actions/runs/3307541575

lukjod, 2022-10-23:
So after first test of first firmware, power led blinks during startup and stays on on full boot.
usb led is working well, unfortunatelly wlan leds don't work no in netdev bor always on.
will try second firmware and report back

DragonBluep, 2022-10-23:
The second on is released now. Please run `cat /sys/class/leds/blue:power/trigger` in SSH and show me the output.
And please also try https://github.com/DragonBluep/openwrt/actions/runs/3307541575 an hour later and run `fw_printenv` in SSH.

lukjod, 2022-10-23:
ok thx for very quick support
first ssh command`cat /sys/class/leds/blue:power/trigger
[none] timer heartbeat default-on netdev mt7530-0:00:link mt7530-0:00:1Gbps mt7530-0:00:100Mbps mt7530-0:00:10Mbps mt7530-0:01:link mt7530-0:01:1Gbps mt7530-0:01:100Mbps mt7530-0:01:10Mbps mt7530-0:02:link mt7530-0:02:1Gbps mt7530-0:02:100Mbps mt7530-0:02:10Mbps mt7530-0:03:link mt7530-0:03:1Gbps mt7530-0:03:100Mbps mt7530-0:03:10Mbps usbport phy0rx phy0tx phy0assoc phy0radio phy1rx phy1tx phy1assoc phy1radio
`
should i test https://github.com/DragonBluep/openwrt/actions/runs/3307622040 or wait for new first firmware?

DragonBluep, 2022-10-23:
> should i test https://github.com/DragonBluep/openwrt/actions/runs/3307622040 or wait for new first firmware?

Please try this one first.  

> unfortunatelly wlan leds don't work no in netdev bor always on.

You mean wlan2g/5g LED is alway on? If so, 3307622040 should fix it.

lukjod, 2022-10-23:
On second firmware the same usb light can be controlled , wlan leds dont work at all, they are completely off no matter what i choose

DragonBluep, 2022-10-23:
Can you try to find the GPIO of the wlan led? First you need to run `ls /sys/class/gpio`, you'll get three strings gpiochipNUM1, gpiochipNUM3, gpiochipNUM3. And then, you need to download gpio-test.sh and change `GPIOCHIP=0` to `GPIOCHIP=NUM1`, `GPIOCHIP=NUM2`, `GPIOCHIP=NUM3` respectively. Finally push this file to /tmp directory and run `chmod +x /tmp/gpio-test.sh` `/tmp/gpio-test.sh`.
https://openwrt.org/docs/guide-developer/add.new.device#gpio_leds
If successful, you can get the GPIO. If you use Windows OS, winscp is a good software to push files.

lukjod, 2022-10-23:
ok, tried this but unfortunately leds dont light up on those gpios
export       gpiochip416  gpiochip448  gpiochip480  unexport
what else can be done?

DragonBluep, 2022-10-23:
I see, WLAN LEDs are controlled by mt7915 wireless chip. It's a pity but it seems that we can't control both of them now. #10400
```
GPIO PIN MAP in ASUS RT-AX53U:
WLAN2G -- MT7905 GPIO18
WLAN5G -- MT7905 GPIO26
```
Or maybe you can create an issue on [openwrt/mt76](https://github.com/openwrt/mt76) and provide the information above.

lukjod, 2022-10-23:
Can You please add support for usb led as it is working well to main branch of openwrt, power led also works now ok

DragonBluep, 2022-10-24:
> Can You please add support for usb led as it is working well to main branch of openwrt, power led also works now ok

@rozhuk-im is working on this so let's just wait. https://github.com/openwrt/openwrt/pull/10400

DragonBluep, 2022-12-11:
It seems like LEDs work on snapshots version now.
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.bin

lukjod, 2022-12-11:
hmm it doesn't work for me, any additional setup is needed?

DragonBluep, 2022-12-11:
No, not need.
I saw that some patches were pushed to the master branch yesterday. https://github.com/openwrt/openwrt/commit/a75a7981621a14adb3a11d1af66b72320b13c5ef
```
551201379efe wifi: mt76: move leds field in leds struct
14fbb6d6e85e wifi: mt76: move leds struct in mt76_phy
81edc468fc62 wifi: mt76: mt7915: enable per-phy led support
bbad827e447f wifi: mt76: mt7615: enable per-phy led support
```
If it doesn't work, maybe you can contact the maintainer. https://github.com/openwrt/mt76/issues/714

rozhuk-im, 2022-12-12:
> It seems like Wi-Fi LEDs work on snapshots version now.

Nice to hear!
WiFi leds was inaccessible via GPIO.
I hope to done with RT-AX53U on new year holidays.

e9hack, 2022-12-18:
> No, not need. I saw that some patches were pushed to the master branch yesterday. [a75a798](https://github.com/openwrt/openwrt/commit/a75a7981621a14adb3a11d1af66b72320b13c5ef)
> 
> ```
> 551201379efe wifi: mt76: move leds field in leds struct
> 14fbb6d6e85e wifi: mt76: move leds struct in mt76_phy
> 81edc468fc62 wifi: mt76: mt7915: enable per-phy led support
> bbad827e447f wifi: mt76: mt7615: enable per-phy led support
> ```
To get wifi leds working, the '-DCONFIG_MT76_LEDS' in the Makefile must be replaced by '-DCONFIG_MT76_LEDS=1'.
 ```
--- a/Makefile	2022-12-10 13:11:05.000000000 +0100
+++ b/Makefile	2022-12-18 12:26:45.617821410 +0100
@@ -1,5 +1,5 @@
 # SPDX-License-Identifier: GPL-2.0-only
-EXTRA_CFLAGS += -Werror -DCONFIG_MT76_LEDS
+EXTRA_CFLAGS += -Werror -DCONFIG_MT76_LEDS=1
 obj-m := mt76.o
 obj-$(CONFIG_MT76_USB) += mt76-usb.o
 obj-$(CONFIG_MT76_SDIO) += mt76-sdio.o
 ```

rozhuk-im, 2022-12-19:
> To get wifi leds working, the '-DCONFIG_MT76_LEDS' in the Makefile must be replaced by '-DCONFIG_MT76_LEDS=1'.

I convert it into openwrt repo patch
```
diff --git a/package/kernel/mt76/patches/003-wifi_leds.patch b/package/kernel/mt76/patches/003-wifi_leds.patch
new file mode 100644
index 0000000000..100ee7a56c
--- /dev/null
+++ b/package/kernel/mt76/patches/003-wifi_leds.patch
@@ -0,0 +1,9 @@
+--- a/Makefile	2022-12-10 13:11:05.000000000 +0100
++++ b/Makefile	2022-12-18 12:26:45.617821410 +0100
+@@ -1,5 +1,5 @@
+ # SPDX-License-Identifier: GPL-2.0-only
+-EXTRA_CFLAGS += -Werror -DCONFIG_MT76_LEDS
++EXTRA_CFLAGS += -Werror -DCONFIG_MT76_LEDS=1
+ obj-m := mt76.o
+ obj-$(CONFIG_MT76_USB) += mt76-usb.o
+ obj-$(CONFIG_MT76_SDIO) += mt76-sdio.o
```
rebuild and reinstall.
Leds not work, probably I miss something?

e9hack, 2022-12-19:
> rebuild and reinstall. Leds not work, probably I miss something?

Sorry, I did test to much in parallel. This was the wrong patch.
```
diff --git a/package/kernel/mt76/Makefile b/package/kernel/mt76/Makefile
index c71c7c52de..cdc8ae54ac 100644
--- a/package/kernel/mt76/Makefile
+++ b/package/kernel/mt76/Makefile
@@ -360,6 +360,8 @@ ifdef CONFIG_PACKAGE_kmod-mt7663u
 endif
 ifdef CONFIG_PACKAGE_kmod-mt7915e
   PKG_MAKE_FLAGS += CONFIG_MT7915E=m
+  PKG_MAKE_FLAGS += CONFIG_MT76_LEDS=y
+  NOSTDINC_FLAGS += -DCONFIG_MT76_LEDS
   ifdef CONFIG_TARGET_mediatek_filogic
     PKG_MAKE_FLAGS += CONFIG_MT7986_WMAC=y
     NOSTDINC_FLAGS += -DCONFIG_MT7986_WMAC

```
And for the usb led:
```
diff --git a/target/linux/ramips/dts/mt7621_asus_rt-ax53u.dts b/target/linux/ramips/dts/mt7621_asus_rt-ax53u.dts
index aec8a0494d..90004b0426 100644
--- a/target/linux/ramips/dts/mt7621_asus_rt-ax53u.dts
+++ b/target/linux/ramips/dts/mt7621_asus_rt-ax53u.dts
@@ -27,10 +27,20 @@
 		compatible = "gpio-leds";
 
 		led_power: led-0 {
+			label = "blue:power";
 			color = <LED_COLOR_ID_BLUE>;
-                        function = LED_FUNCTION_POWER;
+			function = LED_FUNCTION_POWER;
 			gpios = <&gpio 13 GPIO_ACTIVE_LOW>;
 		};
+
+		usb: led-7 {
+			label = "blue:usb";
+			color = <LED_COLOR_ID_BLUE>;
+			function = LED_FUNCTION_USB;
+			gpios = <&gpio 14 GPIO_ACTIVE_LOW>;
+			trigger-sources = <&ehci_port2>;
+			linux,default-trigger = "usbport";
+		};
 	};
 
 	keys {
diff --git a/target/linux/ramips/image/mt7621.mk b/target/linux/ramips/image/mt7621.mk
index 75022f9708..a0347c61b7 100644
--- a/target/linux/ramips/image/mt7621.mk
+++ b/target/linux/ramips/image/mt7621.mk
@@ -307,7 +307,7 @@ define Device/asus_rt-ax53u
   IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
   IMAGE/factory.bin := append-kernel | pad-to $$(KERNEL_SIZE) | append-ubi | \
 	check-size
-  DEVICE_PACKAGES := kmod-mt7915e kmod-usb3 uboot-envtools
+  DEVICE_PACKAGES := kmod-mt7915e kmod-usb3 uboot-envtools kmod-usb-ledtrig-usbport
 endef
 TARGET_DEVICES += asus_rt-ax53u
 

```

lukjod, 2022-12-19:
Can You please include this to snapshot?

rozhuk-im, 2022-12-20:
@e9hack thanks!
https://github.com/rozhuk-im/openwrt/commit/d3d9210b57fb2c7015dcc4e0005ace12fbb6957b with that changes wifi leds work and all other leds I can control via WebUI.

Is any chance to take control for wifi leds?

Djfe, 2022-12-21:
You could edit the description of the commit, since you don't have to remove lan4 anymore. (was done already in the main branch)

Are you planning on adding and testing uboot-envtools support? else you might want to defer this to someone else and another pull request so led support gets added faster.

you could update your pr #10400 with rozhuk-im@d3d9210

Djfe, 2022-12-23:
I just tested the patch @Znevna created on my Asus RT-AX53U. (I git applied his patch over the current openwrt master branch)
I can confirm, that all LEDs are working (power, lan1-3, wan, wifi 2.4, wifi 5 and usb).
and I can confirm @Znevna's observation regarding the 5GHz LED shining (slightly dimmed) during the boot process (after the bootloader) while power led is flashing. It stops right when WAN starts lighting up for the first time.

What I noticed:
WiFi LED's don't always represent the state the radio is in.
After boot up:
I enable wifi with luci -> led lights up (this is equally the case for both radios)
I disable wifi with luci -> led light stays on

I enable wifi with luci -> led lights up
I reboot
radio's light is up again
I disable wifi with luci -> led turns off (successfully)
I enable wifi with luci -> led lights up
I disable wifi with luci -> led light stays on

I need to reboot to update the led light.

By "Led lights up/stay on" I mean that it's in a state of blinking (not fully lit all the time), even if the radio is turned off. So I'm not sure the blinking is actually linked to WiFi traffic. The blinking is not static (it switches between blinking and being lit for a short while and then blinking again; the number of blinks is different most of the time).
It seems like the wifi led's can change state once to off right after a reboot and if turned on for the first time they stay on no matter the state of your wifi radio.

Should I report this to the mt76 repo as a new issue?

Djfe, 2022-12-23:
it's not like we could clone your repo for that branch anyways 😅
your github is private

But we can still get it as a patch file to git apply
https://github.com/Znevna/openwrt/commit/5cd1821c26c14fa17ffbdcfe3c45fc43beab2d85.patch

lgitano, 2023-01-04:
> Sorry, I did test to much in parallel. This was the wrong patch.
>
> ```
> diff --git a/package/kernel/mt76/Makefile b/package/kernel/mt76/Makefile
> index c71c7c52de..cdc8ae54ac 100644
> --- [...]

Thank You very much! Have been sucessfully patching the Makefile for the Filogic 830 (MT7986) on the Bananapi BPi-R3, both the W2G and W5G  LED where  not working in Snapshot as well. USB LED is not abailable on this device, afaik. Would it possible to include this patch as well in the Snapshot branch? Kinda annyoing compiling OpenWRT again and again just for this feature, everything else so far seems to work fine on this device :)

Djfe, 2023-01-04:
@hauke already suggested activating this build flag for all devices using the MT76 driver. This supports that idea, but we should still wait for the response of the maintainers of mt76 on the matter.
https://github.com/openwrt/openwrt/pull/11601#discussion_r1059124071

Znevna, 2023-01-04:
Well, I've reverted those changes and submitted a PR in mt76 instead https://github.com/openwrt/mt76/pull/725

lgitano, 2023-01-04:
Thank You very much!

Znevna, 2023-01-04:
@Djfe Also regarding the wifi LEDs, they are working properly apparently, just the radios don't get completely turned off for some reason from LuCI.
However, `wifi down` & `wifi up` from CLI does the job.

Djfe, 2023-01-04:
@Znevna I literally stumbled upon this fact a couple of days ago ^^
But I wasn't able to confirm it, yet. (no access to my device atm.)
https://github.com/openwrt/openwrt/pull/11220#issuecomment-1336263591
(this issue was linked to from an issue in the mt76 repo)
It's literally the behaviour I've been describing a couple posts ago:
https://github.com/openwrt/openwrt/issues/11068#issuecomment-1363612808

Thanks for informing me about it though :)
I'm wondering if this is a bug (wifi not actively being disabled by LuCI) and if it's in OpenWRT or MT76.
And why are they blinking as if the wifi where active (when no device is able to connect)?

Znevna, 2023-01-04:
Well, when the "disable" button doesn't work right in LuCI I'm left with the radios still broadcasting, the SSID is visible, but nothing can connect to them. The LEDs don't lie, I don't know what's going on here.

Djfe, 2023-01-04:
I might compile release with the led flag to compare whether this is a regression in the snapshots.
Maybe I'll just install compiled releases, if I find a way to see whether the device is broadcasting without looking at the led's ^^
Though I can't test release 21 with our device for obvious reasons 😅

dfateyev, 2023-01-04:
> I might compile release with the led flag to compare whether this is a regression in the snapshots. Maybe I'll just install compiled releases, if I find a way to see whether the device is broadcasting without looking at the led's ^^ 

You can try pressing "Restart" (Wifi interface) right before pressing "Disable". This way, you will probably be able to disable radio interface completely. At least, it forks for my MT7621 devices.
I doubt that it is a regression in master snapshots, probably was introduced earlier — but haven't traced specifically.

Znevna, 2023-01-04:
Confirmed, pressing restart before pressing disable makes the "disable" action work properly.
These are logs from a "working" disable:
```
Wed Jan  4 03:21:20 2023 kern.info kernel: [12797.346347] device phy0-ap0 left promiscuous mode
Wed Jan  4 03:21:20 2023 kern.info kernel: [12797.351418] br-lan: port 4(phy0-ap0) entered disabled state
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: Remove interface 'phy0-ap0'
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Wed Jan  4 03:21:20 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Wed Jan  4 03:21:20 2023 daemon.notice netifd: Wireless device 'radio0' is now down
```
And these are logs from a "broken" disable attempt:
```
Wed Jan  4 03:21:53 2023 kern.info kernel: [12830.766311] device phy0-ap0 left promiscuous mode
Wed Jan  4 03:21:53 2023 kern.info kernel: [12830.771339] br-lan: port 4(phy0-ap0) entered disabled state
Wed Jan  4 03:21:53 2023 daemon.notice netifd: radio0 (15966): WARNING: Variable 'data' does not exist or is not an array/object
Wed Jan  4 03:21:53 2023 daemon.notice netifd: radio0 (15966): Bug: PHY is undefined for device 'radio0'
Wed Jan  4 03:21:53 2023 daemon.notice netifd: Wireless device 'radio0' is now down
```

Djfe, 2023-01-04:
nice! the PR was pushed, now the only thing remaining are the wifi leds

closed 2023-01-04 by openwrt-bot

lgitano, 2023-01-06:
> nice! the PR was pushed, now the only thing remaining are the wifi leds

@Djfe: am fairly new to the github process, i am experiencing the same behaviour with the wifi LEDs for Target System Mediatek Raling Arm -> Subtarget Filogic 830 (MT7986) -> Target Profile (Bananapi BPi-R3).  Do i have to open up a new thread/ patch request for this device or will this issue eventually get fixed for all devices equipped with this chipset after the last request here? Thank You very much!

Znevna, 2023-01-07:
As mentioned in my PR, the wifi LEDs depend on a mt76 PR too, that got accepted a few hours ago, it will make it into master OpenWrt at the next mt76 version bump, so we need to wait just a little longer.

Djfe, 2023-01-08:
Here you go @lukjod
[openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/10367511/openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.zip)

Djfe, 2023-01-08:
> You can try pressing "Restart" (Wifi interface) right before pressing "Disable". This way, you will probably be able to disable radio interface completely. At least, it works for my MT7621 devices. I doubt that it is a regression in master snapshots, probably was introduced earlier — but haven't traced specifically.

weird, that doesn't work for me at all
all my attempts:
```
[  577.912142] br-lan: port 4(phy0-ap0) entered blocking state
[  577.917845] br-lan: port 4(phy0-ap0) entered disabled state
[  577.924097] device phy0-ap0 entered promiscuous mode
[  577.929898] br-lan: port 4(phy0-ap0) entered blocking state
[  577.935515] br-lan: port 4(phy0-ap0) entered forwarding state
[  577.943486] br-lan: port 4(phy0-ap0) entered disabled state
[  578.054179] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  578.061334] br-lan: port 4(phy0-ap0) entered blocking state
[  578.067450] br-lan: port 4(phy0-ap0) entered forwarding state
[  592.158379] device phy0-ap0 left promiscuous mode
[  592.163437] br-lan: port 4(phy0-ap0) entered disabled state
[  745.876081] br-lan: port 4(phy0-ap0) entered blocking state
[  745.881661] br-lan: port 4(phy0-ap0) entered disabled state
[  745.888171] device phy0-ap0 entered promiscuous mode
[  745.893573] br-lan: port 4(phy0-ap0) entered blocking state
[  745.899674] br-lan: port 4(phy0-ap0) entered forwarding state
[  770.323659] device phy0-ap0 left promiscuous mode
[  770.328691] br-lan: port 4(phy0-ap0) entered disabled state
[  806.328838] br-lan: port 4(phy0-ap0) entered blocking state
[  806.334434] br-lan: port 4(phy0-ap0) entered disabled state
[  806.340905] device phy0-ap0 entered promiscuous mode
[  806.346556] br-lan: port 4(phy0-ap0) entered blocking state
[  806.352152] br-lan: port 4(phy0-ap0) entered forwarding state
[  819.537343] device phy0-ap0 left promiscuous mode
[  819.542403] br-lan: port 4(phy0-ap0) entered disabled state
[  867.126674] br-lan: port 4(phy0-ap0) entered blocking state
[  867.132289] br-lan: port 4(phy0-ap0) entered disabled state
[  867.138850] device phy0-ap0 entered promiscuous mode
[  867.144145] br-lan: port 4(phy0-ap0) entered blocking state
[  867.150212] br-lan: port 4(phy0-ap0) entered forwarding state
[  876.568233] device phy0-ap0 left promiscuous mode
[  876.573315] br-lan: port 4(phy0-ap0) entered disabled state
[ 1029.314903] br-lan: port 4(phy0-ap0) entered blocking state
[ 1029.320502] br-lan: port 4(phy0-ap0) entered disabled state
[ 1029.327010] device phy0-ap0 entered promiscuous mode
[ 1029.332356] br-lan: port 4(phy0-ap0) entered blocking state
[ 1029.338505] br-lan: port 4(phy0-ap0) entered forwarding state
[ 1052.764022] device phy0-ap0 left promiscuous mode
[ 1052.769060] br-lan: port 4(phy0-ap0) entered disabled state
```

I just discovered the following old issue, it sounds similar, so maybe this issue is fairly old. I might be able to do a regression test with a D-Link DIR-860L
https://github.com/openwrt/openwrt/issues/7485

Djfe, 2023-01-08:
This is the commit that added the error string, when it fails:
https://github.com/openwrt/openwrt/pull/4314
(so it doesn't fail silently anymore)

I'm probably going to open a separate issue for this to figure out why these errors come up and what needs to be fixed.

Djfe, 2023-01-08:
@lukjod here's your build:
[openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/10367731/openwrt-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.zip)
(the first time I forgot to include LuCI for you ^^)

Djfe, 2023-01-08:
@Igitano
good question, I'm not entirely sure. (I'm new to developing on openwrt)
Support seems to be there though, so the only thing missing is likely the build flag.
https://github.com/openwrt/mt76/blob/7fae1de12ae7832a6095fd2df198f41fabd5223d/mt7915/init.c#L501

I could build a new image for your device with the change that's about to land in master and you could install the image, to test if it works.
On that note: which storage variant are you using? (there are four or five)

If it doesn't work then we might need to add a kmod package for your wifi chip to the mk file and add entries for the wifi led's into the 01_board file.
The latter is useful regardless, in case you want to disable led's for example.

lukjod, 2023-01-08:
@Djfe Tried it but unfortunatelly it doesnt work on my router. It was not booting at all and come into bootloop. Had to reinstall snapshot from openwrt to put it on work. What can be wrong?

Djfe, 2023-01-08:
that's unfortunate 🙈
I'll check again later.
did you reset the config on updating? (sysupgrade -n filename)

Djfe, 2023-01-10:
> @Djfe Tried it but unfortunatelly it doesnt work on my router. It was not booting at all and come into bootloop. Had to reinstall snapshot from openwrt to put it on work. What can be wrong?

@lukjod Could you pls check whether the bin file has the same checksum?
`245dd8430e296b1c9bdafdfda9a8018657678b469f3bcd8cc78d18443f78ebce`
You need to extract the sysupgrade.bin from the zipfile first.

How did you install the upgrade? LuCI?
Did you keep the config or not?

Djfe, 2023-01-10:
I'm kinda confused, because I tested the image on my side, before uploading it here to avoid something like this.
I just downloaded it. Unzipped it with 7zip. And then flashed it over LuCI (while selecting to not keep the settings/config) and it boots just fine for me with a working LuCI out of the box.

Maybe your upload to the box failed/had an issue.
Pls use LuCI and compare the sha256 from LuCI with the one I posted above.
[image]

lukjod, 2023-01-10:
@Djfe now it worked but when try to restore packages and settings system says it is read only. Will go back to snapshot and wait for newer driver there, For me its too complicated and above my knowledge. Thanks anyway for Yours work

Djfe, 2023-01-10:
understandable, oh well.
I'll remove the image zip above once the change lands in snapshot.

Djfe, 2023-02-03:
@lukjod the MT76 WiFi driver just received an update on master:
https://github.com/openwrt/openwrt/commit/521efb62ebb55dcc67dc618757aa2e3ccbdf3774

LEDs are working on Snapshot now :)
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=asus_rt-ax53u

here is how to install luci via SSH (section 1 and 2):
https://openwrt.org/docs/guide-user/luci/luci.essentials

lukjod, 2023-02-03:
Thank You very much for info will update asap

lgitano, 2023-02-08:
Hello, thanks for the information :)

Can confirm MT76 on Filogic/ BPi R3, both LED's are working now when either Wifi Chip is enabled in Luci, cool :) However, when i switch it off again the according LED remains switched on and will get disabled once system is rebooted. 

Cheers!

dfateyev, 2023-02-08:
> However, when i switch it off again the according LED remains switched on and will get disabled once system is rebooted.

You can check if the [workaround](https://github.com/openwrt/openwrt/issues/11068#issuecomment-1371285982) above works for you. It works for ramips, probably for other platforms, too.

Djfe, 2023-02-08:
> However, when i switch it off again the according LED remains switched on and will get disabled once system is rebooted.

That's actually a bug on master branch, that has nothing to do with the driver nor the leds.
The leds are working fine. But if you disable wifi in LuCI, then that will fail and the wifi chip will stay online while connection to the wifi network fails. So the leds just show that wifi is still online.
Using `wifi down` in ssh works fine though. (`wifi up` to enable wifi again)

I discovered the root of the issue and reported it here:
https://github.com/openwrt/openwrt/issues/11725#issuecomment-1403270202

lgitano, 2023-02-08:
Heya, found some time to  play around with this neither restarting the devices before disabling nor toggling the device with wifi up/down seems to  work, here. But anyways, i'm happy that these turn on, finally :) Time to move past the testing stage here and get a production system up and running, i guess!