openwrt/openwrt #11074: plink -m ... on 22.03.2 gives FATAL ERROR: Remote side unexpectedly closed network connection

lllaffer, 2022-10-24:
After applying

openwrt-22.03.2-ath79-generic-tplink_archer-c5-v1-squashfs-sysupgrade.bin

plink -v -pw pass root@router.ip -m init.sh    

which  had been working before ( ... with same 'plink') - now gives

```
Looking up host "router.ip" for SSH connection                                    
Connecting to fdbb:f389:49fe::1 port 22                                           
We claim version: SSH-2.0-PuTTY_Prerelease_0.77:20220525.01d8561                  
Connected to fdbb:f389:49fe::1                                                    
Remote version: SSH-2.0-dropbear                                                  
Using SSH protocol version 2                                                      
No GSSAPI security context available                                              
Doing ECDH key exchange with curve Curve25519 and hash SHA-256 (unaccelerated)    
Server also has ssh-ed25519 host key, but we don't know it                        
Host key fingerprint is:                                                          
ssh-rsa 2048 SHA256:7m7cvSNF08ay6J0hjtGW32kwq7AnJ43dw12kBnU9cOs                   
Initialised AES-256 SDCTR (unaccelerated) outbound encryption                     
Initialised HMAC-SHA-256 (unaccelerated) outbound MAC algorithm                   
Initialised AES-256 SDCTR (unaccelerated) inbound encryption                      
Initialised HMAC-SHA-256 (unaccelerated) inbound MAC algorithm                    
Using username "root".                                                            
Sent password                                                                     
Access granted                                                                    
Opening main session channel                                                      
Opened main channel                                                               
Remote side unexpectedly closed network connection                                
FATAL ERROR: Remote side unexpectedly closed network connection
```

Transferring the file with pscp and executing it with plink does work.

brada4, 2022-10-24:
Do you ask help to trace a script?
Most likely you dont get strdin or anything stty like password prompts from plink

brada4, 2022-10-24:
Post line at a time to openwrt to find which disconects like restarts firewall or network or dropbear service.
Another is reading or setting up nonexistent terminal in plink. For openssh that would be -t to force terminal to be allocated.