openwrt/openwrt #11077: Turris Omnia Broken Switch Configuration

drujd, 2022-10-24:
At least since 22.03 and definitely in 22.03.2, Turris Omnia switch is broken in OpenWrt (although in ways average user may not notice at first glance). Even with OpenWrt stock, anything sent from a switch port to router (possibly to other scenarios) gets broadcasted TO ALL switch ports. This has obvious performance implications, e.g. having a 10baseT device with flow control connected effectively gimps ALL communication from switch-connected devices to router/internet to 10 Mb/s.

This is effectively this bug on turrisos: https://gitlab.nic.cz/turris/os/build/-/issues/336

The Turris team fixed this in their patched kernel 5.15, I guess they didn't even bother to fix this upstream / in older kernel versions... :(

I would say this is quite a blocke. You should remove Omnia as a supported device from 22.03 until this gets fixed...

ynezz, 2022-10-24:
Probably duplicate of https://github.com/openwrt/openwrt/issues/10997 and https://github.com/openwrt/openwrt/issues/10628.

closed 2022-10-24 by ynezz

Borromini, 2022-12-01:
That's just lovely. @BKPepe Any chance you could lend a hand with this? The Turris issue does not reference any specific patch, just that one of their 5.15 updates fixed it...

BKPepe, 2022-12-01:
That's being discussed in one of those duplicated issues, no need to discuss it in every single issue. What helps is to switch to kernel 5.15, because this is fixed in one of those DSA fixes in Linux kernel and the difference between 5.10 and 5.15 is huge... You would need to backport almost the entirely DSA subsystem or use the newer kernel, the later one was better for the maintenance and that's what we did.

hauke, 2023-01-09:
There is a mailing list discussion about this topic: https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg63367.html
It was also reported to upstream kernel: https://bugzilla.kernel.org/show_bug.cgi?id=209487
The problem is fixed in OpenWrt master with kernel 5.15. OpenWrt 21.02 is not affected, only kernel 5.10 is affected.

champtar, 2023-01-09:
@hauke if you use bridge vlan filtering everything works fine with 5.10 and 5.15 is broken ... see my sum up here https://github.com/openwrt/openwrt/issues/10997#issuecomment-1364286664.
Can't investigate more as I'll not have access to my Omnia for another ~6 weeks

cowwoc, 2023-01-10:
> We've discussed those issues in the developer meeting yesterday and the conclusion was to disable support for those broken devices in 22.03 release. So [I've just sent out such proposal in the patch form for a review](https://lists.infradead.org/pipermail/openwrt-devel/2022-December/039921.html).

When you say "disable support for those broken devices" do you mean that as a WRT1900ACS user I cannot upgrade the firmware from 22.03.2 to 22.03.3? Or do you mean that I can upgrade but some feature will be disabled?

cowwoc, 2023-01-10:
I just realized that you removed complete support for my device.

When can we expect support to come back? Will version 22.03.4 upgrade to Kernel 5.15 which fixes this?

champtar, 2023-01-10:
@cowwoc can you share in #10997 if you have any of the reported issues and what does your network config look like

cowwoc, 2023-01-10:
@champtar I am not personally impacted by this bug because I am only using one switch port. Performance is not a problem because this is a home router with low traffic. The only way I am impacted is losing out on updates because my device is no longer supported :(

mdsharpe, 2023-01-12:
wrt3200acm user here. Agree with the above; not sure it makes sense to withdraw support at this stage. In fact, in I'm reading things right, the latest supported release is apparently a broken one, and you've blocked the release with the fix?

tojestzart, 2023-01-12:
qestion is how they scale their own releases comparing to kernel.  like 22.03.x vs kernel. Its too much diging to find it but 
22.03.3 --> Update Linux kernel from 5.10.146 to 5.10.161
and https://openwrt.org/releases/22.03/start

Developers didnt state correctly which release is affected *jump to DSA) whole, or just updates.
Stop doing dymy. @rmilecki

zorxd, 2023-01-12:
> I just realized that you removed complete support for my device.
> 
> When can we expect support to come back? Will version 22.03.4 upgrade to Kernel 5.15 which fixes this?

From what I understand, the kernel won't be updated to > 5.10 (likely 5.15) until the next major OpenWRT release (23.xx?)

I too don't think it's a good idea to block releases. People will just install 22.03.2 instead of the more secure 22.03.3 and future updates. There is no regression when going from 22.03.2 to 22.03.3 when using the problematic switch.

And is downgrading to 21.02 a good idea anyways? Is that version still getting security updates?

nullr0ute, 2023-01-13:
> From what I understand, the kernel won't be updated to > 5.10 (likely 5.15) until the next major OpenWRT release (23.xx?)

Is there a roadmap to the next major release? Is it going to be 5.15 or the next LTS (probably 6.1)?

TDFKAOlli, 2023-01-14:
@champtar Like @cowwoc I'm not impacted by the bug, but impacted by the fact that for my ACS there is no firware update available anymore. 21.02.05 didn't got an update with security fixes and anyway a downgrade would be a pain due to switch from swconfig to DSA for mvebu. I'm now aware of the issue (which already is in 22.03.02 and former 22.03 versions) so I should be able to decide by myself. Blocking 22.03.3 for those devices isn't a good idea in MPOV as it expose those devices to security vulnerabilites. A disclaimer can be given for those devices giving the note about the bug (which has already been done in the release note of this version) and then people can decide themselves.

thecode, 2023-01-14:
I believe no one is tracking a closed issue, but I agree that these releases shouldn't being blocked. I am on `23.03.0`, while I am pretty sure I am effected by this bug I never noticed it, I already use packages & config which are not backward compatible and now I am stuck on `23.03.0`.
In my opinion this is a strange decision, the WRT series had a WIFI bug for a long time but it didn't make anyone block them from future releases. The release notes have a section for [Known issues](https://openwrt.org/releases/22.03/notes-22.03.3#known_issues) so unless the issue is critical and may break the device I don't see any reason excluding it from the release.

BKPepe, 2023-01-14:
It would be nice to get a reply from @drujd , because he was quite proactive from my point of view (https://github.com/openwrt/openwrt/issues/10997#issuecomment-1289292990, https://github.com/openwrt/openwrt/issues/10628#issuecomment-1333679157 and in the latter one we discussed it and it did not seem to help much), who complained about it, too much and because of that it was disabled. I need to admit that having radio silence is not a good solution. 

However, since a few people complained, OpenWrt put this topic on their internal meeting and decided to disable it. You can take a look here about it: https://github.com/openwrt/openwrt/issues/10628#issuecomment-1333531751

There are also other people, as you can see on the OpenWrt forum, who does not have the issue, which was reported three times:
- https://forum.openwrt.org/t/22-03-x-no-release-builds-for-turris-omnia-linksys-wrt-clearfog/147955

I am against suggesting users use the snapshot for their devices, which they are using on a daily basis, as it can be broken sometimes, but right now,  there are these options:
- stay on OpenWrt 22.03.2 (which could have some vulnerable packages...)
   - use even older stable release, which is still supported
- compile their own build by using ImageBuilder or copy and compile OpenWrt from the source code
- complain about it on the mailing list to let OpenWrt core members know about it

Overall, it was not a good idea to disable devices using the affected Marvell switch for 22.03.2, as it happened since OpenWrt 22.03.0.

drujd, 2023-01-14:
> It would be nice to get a reply from @drujd , because he was quite proactive from my point of view (https://github.com/openwrt/openwrt/issues/10997#issuecomment-1289292990, [...]

What kind of reply do you expect? (Salty much?)

IIRC (correct me if I'm wrong, obviously you read my comments way more times than I did and remember them well) I did not suggest to necessarily stop building the images, just to clearly mark the devices as unsupported/problematic and warn the users.

> I need to admit that having radio silence is not a good solution.

And yet it kind of seems like you think it is:

Case 1: This "look at what you've done!" comment

Case 2: You have known about this bug for a long time but (seemingly?) didn't even bother to do the bare minimum for your upstream and communicate with OpenWrt devs to avoid this happening in the first place? When I saw the same exact issue with OpenWrt a year (or so) later after TurrisOS, the deja vu really made me question how Turris-OpenWrt communication works...

In ideal world, 22.03 would never claim to support any router with the switch - this issue has been known before even 22.03 rc1 was released. Had I found out about the issue reappearing on OpenWrt sooner, I would report it sooner. Too bad...

AgentOak, 2023-01-14:
I believe I am also not affected by this bug on my Linksys WRT1200AC, as I see all port LEDs blinking individually, and now stuck on 22.03.2 (other than upgrading to a snapshot). Are the LEDs a reliable way to tell, or is there another test I should do to confirm the bug?

TDFKAOlli, 2023-01-14:
@thecode Yes, you are right, this ticket is closed. I just ended up here because of the link on the release notes page. Commenting in the forum is the better idea, I'll do so. BTW: You can still download 22.03.2, just go here https://downloads.openwrt.org/releases/22.03.2/ and the images are still available for that version. Just for 22.03.3 they are missing from the download folders.

drujd, 2023-01-14:
> Are the LEDs a reliable way to tell, or is there another test I should do to confirm the bug?

I certainly would not rely on LEDs to diagnose the issue.

Connect two devices to the switch. Run e.g. ping to router on the first device, then enable promiscuous mode on the second one and watch traffic (Wireshark...) - if you see the ping packets (well, in one direction anyway), that's the issue.

thecode, 2023-01-14:
As far as I understand if you have only a single port on the LAN and a single port on the WAN this is not an issue? or there is a mix between the WAN & LAN? I have a downstream switch which has only a single connection to the router.

drujd, 2023-01-14:
> As far as I understand if you have only a single port on the LAN and a single port on the WAN this is not an issue?

Correct.

champtar, 2023-01-14:
For the record, Turris Omnia works just fine with kernel 5.10 (22.03 or master) and bridge vlan filtering (non default config), but this break with 5.15 :(

drujd, 2023-01-14:
> For the record, Turris Omnia works just fine with kernel 5.10 (22.03 or master) and bridge vlan filtering (non default config), but this break with 5.15 :(

I read about that several times. I do not currently have time to actually test that, but IIRC half a year ago when it manifested in testing TurrisOS builds, just enabling vlan filtering itself did NOT fix that issue.

That being said, if there is some kind of actual workaround involving enabling VLANs (just create some vlan configuration that is functionally equivalent to the default).

NullDev42, 2023-01-14:
> > As far as I understand if you have only a single port on the LAN and a single port on the WAN this is not an issue?
> 
> Correct.

Is this true across the board or does it depend upon the internal hardware configuration?  For instance, if the LAN and WAN are both located on the switch (rather than tied to a port on the CPU), would there be a potential issue?

drujd, 2023-01-14:
> Is this true across the board or does it depend upon the internal hardware configuration?  For instance, if the LAN and WAN are both located on the switch (rather than tied to a port on the CPU), would there be a potential issue?

Hard to tell, according to other people, the issue does not occur with vlan filtering enabled (which you would use for the separation of LAN and WAN on the switch), but I cannot confirm that.

Falcon-118, 2023-01-14:
No statement, no suggestions.  Just FYI from personal observation for confirmation/information on some of the speculation:  
I use my (4) wrt3200acm as Dumb APs, Openwrt 23.03.2.  

I bond the wan port to the br-lan device and use the wan port as the backhaul on all Dumb APa.  I switched to Openwrt specifically for DSA and make moderate use of vlan filtering for both wireless and ethernet... primary interface and 2 additional.   I just saw the notifications for 23.03.3 and have been reading through the chatter on the "Broken MV88E6176 switch" issue.  I've done some quick ping testing (A>B, B>A, swapping modes) and not seeing any crosstalk. I haven't had any issues.  I don't think it matters, but my primary router is pfSense on a netgate appliance.   

GOpenWRT:~# bridge v
port              vlan-id
lan4              20 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan1              1 PVID Egress Untagged
wan               1 PVID Egress Untagged
                  10
                  20
br-lan            1
                  10
                  20
wlan0             1 PVID Egress Untagged
wlan1             1 PVID Egress Untagged
wlan1-1           10 PVID Egress Untagged
wlan1-2           20 PVID Egress Untagged

I think I'll sit on 23.03.2 until this gets fleshed out.  Since it's a known issue, hopefully it will get a quick fix.

I've got a couple of spare wrt3200acm and will play with the snapshots to see what susses out.

champtar, 2023-01-14:
@Falcon-118 with kernel 5.15 I see tagged traffic exiting on access port, remember to use `-e` with tcpdump (`tcpdump -nne -i ...`)
see https://github.com/openwrt/openwrt/issues/10997#issuecomment-1362350456 for my config and my tests

Falcon-118, 2023-01-15:
Thanks for the link to your other comment..  Good read, especially the patches link.   I haven't moved beyond 5.10 (yet) and yes, I did use the same parms on tcpdump.   Saw nothing unexpected with my config, either direction.

sur1v, 2023-01-15:
Another solution for WRT users is [Divested-WRT](https://divested.dev/unofficial-openwrt-builds/mvebu-linksys/) firmware, totally from SNAPSHOT and right now on kernel 5.15.86

TDFKAOlli, 2023-01-15:
O.k. so correct test to verify would be?

    ping in local lan from device 1 to device 2 both being connected on the lan ports of the router
    tcpdump on openWrt device on WAN interface (tcpdump -nne -i wan net 192.168.1.0/24) and check if ping traffic can be seen
    Would that be the correct way to verify if a device is affected?

Pinging the internet from device 1 shows incoming packets on the device 2 on my device, so the problem on the LAN switch acting as a hub is seen. Looking at the above way of verifying LAN to WAN I cannot see internal packets on the WAN interface. (Linksys WRT1900ACS). But not sure if the way of testing it is correct.

champtar, 2023-01-15:
@TDFKAOlli in my case with 5.15 + bridge vlan filtering, just having devices connected to wifi, I see tagged frame exiting 'access port' on the switch (wifi bridged to vlan 2 and 3, access port vlan 2, tagged frame with vlan id 2 and 3 instead of untagged frame).
I would do the tcpdump from another device, always.

champtar, 2023-01-15:
@TDFKAOlli run tcpdump on another device, always

AgentOak, 2023-01-16:
I just tested by pinging the router from a device on LAN1 and had another computer connected directly to LAN3 running `tcpdump -nne -i <interface>` . I do not see any of the ICMP packets. I also confirmed that tcpdump does work by running some DNS lookups on the same device and they show up, as well as ARP packets. Otherwise there's complete radio silence despite the other ports being quite busy.

I can't test with default configuration right now because a downtime would affect several people, but it looks like it's fine with my configuration. I have every LAN port connected to a different interface in OpenWRT so maybe that's the reason. Quite unfortunate that they decided to block the update for my device...

champtar, 2023-01-17:
@AgentOak can you share your network config (with the IPs / psk / ...) ?

AgentOak, 2023-01-17:
@champtar /etc/config/network: https://gist.github.com/AgentOak/040650b3e49473394d396df131e5033c
Device is Linksys WRT1200AC, official OpenWRT 22.03.2 image. SQM is installed on wan interface, flow offloading is disabled

chealy, 2023-01-21:
In support of debugging issues with the Marvell switches like this, there is a tool that Andrew Lunn wrote here:  https://github.com/lunn/mv88e6xxx_dump

This tool supports dumping internal switch register data and the switches MAC database in a human readable format.  Kernel support for exposing this data was added over several kernel versions.  5.10 has at least this though:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/dsa/mv88e6xxx?h=v5.10&id=bfb255428966e2ab2c406cf6c71d95e9e63241e4
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/dsa/mv88e6xxx?h=v5.10&id=b71a8d60252111d89dccba92ded7470faef16460

With 5.11 kernel, support was also added for viewing the VTU (vlan translation unit) here:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/dsa/mv88e6xxx?h=v5.11&id=ca4d632aef031c7946c42a6eb873d24ab6474f17

It might be useful to integrate this tool or some other similar tool that can access the switch registers via the devlink region interface for debugging purposes.

porroso, 2023-01-21:
my Linksys WRT32X show _switch 0x3520 detected: Marvell 88E6352, revision 1,_ not the one affected (_88E6176_)

ynezz, 2023-01-22:
BTW @kkudielka [has posted on the list](https://lists.infradead.org/pipermail/openwrt-devel/2022-December/040018.html) probably some information worth looking into:

> I believe this is the same bug I reported upstream in October '20 (but nobody was able to reproduce): https://bugzilla.kernel.org/show_bug.cgi?id=209487
> Quoting myself (Comment 11):
> > The issue seems to be finally solved by acceptance of the "RX filtering inDSA" patch series (thanks to Tobias Waldekranz & Vladimir Oltean).
> > https://lore.kernel.org/netdev/20210629140658.2510288-1-olteanv@gmail.com/
> > I tested this with 5.14-rc3 on a Turris Omnia. Traffic from a DSA switch port, addressed to the bridge, is not flooded anymore to other DSA switch ports - even if unicast flooding is turned on.----
> So, I guess this would be the series to backport.

kkudielka, 2023-01-22:
I tried to back-port "RX filtering in DSA" to OpenWrt 5.10, but eventually gave up. Too many dependencies.

For reference, the workaround I used with pre-5.14 mainline kernels was:
```
for port in lan0 lan1 ...; do
    bridge link set dev $port flood off
done
```
This effectively disables flooding of unknown unicast traffic from the specified slave ports. It worked in my setup, but... The whole story, why unicast flooding may be needed in some use cases, has been explained by Russell King in this [commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c138806344855cd7c094abbe7507832107e8171e).

The "RX filtering in DSA" series eventually makes the switch **learn** the addresses behind the CPU port, including that of the master bridge. Only then, I'd say it is reasonable to have unicast flooding enabled.

JamesWWesby, 2023-01-22:
> This effectively disables flooding of unknown unicast traffic from the specified slave ports. It worked in my setup, but... The whole story, why unicast flooding may be needed in some use cases, has [...]

So far so good it works with kernel 5.15 on my router. But one thing with unicast and multicast frames might still be wrong: The switch sends vlan tagged unicast/mutlicast frames also to ports not belonging to that vlan. That looks in my eyes wrong.

NullDev42, 2023-01-23:
> my Linksys WRT32X show _switch 0x3520 detected: Marvell 88E6352, revision 1,_ not the one affected (_88E6176_)

I believe this is within the affected grouping of switch devices (mv88e6xxx).  This same chip was part of [this bug report](https://github.com/openwrt/openwrt/issues/10628).

tojestzart, 2023-04-16:
22.03.**04** they bumped kernel up a little is that problem solved or not? Nothing mentioned in changelog

drujd, 2023-04-16:
In all likelihood this will never get fixed for the 5.10 kernels / 22.03 images. We'll have to wait for 5.15+ / next major OpenWrt release.

thecode, 2023-04-16:
I was at least expecting that WRT images will be built for `22.03.4` as not everyone is affected by this so people will be able to choose if they want to install it or not. Anyhow I guess commenting on a closed issue won't do much, it might be better to create a discussion about it.

bjoernv, 2023-04-24:
If I read the [Release Notes for 22.03.4](https://github.com/openwrt/openwrt/releases/tag/v22.03.4) correctly, all users which are affected by this bug and stuck at 22.03.2 have 11 open CVE's on their routers. What is the recommendation for these users regarding security?

sbocinec, 2023-04-24:
`mvebu` routers are affected by the 7 security CVEs from `22.03.3` and 11 security CVEs from `22.03.4`, but only some are affected by the performance implications of this bug. I'm not sure how OpenWRT maintainers prioritize fixes but I'd also love if the security implications were considered having much higher impact and thus priority, than this switch configuration issue. I feel, releases for `mvebu` should be unblocked as it's going to cause much more harm than benefits.

Falcon-118, 2023-04-24:
> `mvebu` routers are now affected by 7 security CVEs from `22.03.3` and 11 security CVEs from `22.03.4`, 

I'm not sure how OpenWRT maintainers prioritize fixes but I'd love if the security fixes had much higher impact and thus priority, than this switch configuration issue. I feel, releases for `mvebu` should be unblocked as it's going to cause much more harm than benefits.

Agreed.   It's become apparent that "Stable" in terms of their Stable Releases from the devs has to do with their processes and NOT their product.

Falcon-118, 2023-04-24:
> `mvebu` routers are now affected by 7 security CVEs from `22.03.3` and 11 security CVEs from `22.03.4`,  I'm not sure how OpenWRT maintainers prioritize fixes but I'd love if the security fixes had [...]

Agreed. It's become apparent that "Stable" in terms of their Stable Releases from the devs has to do with their processes and NOT their product.

jebrossard, 2023-05-02:
MEVBU router owners should either:
1. Roll back to latest version of 21.xx.x (Which I did on my WRT1900ACS v1)
2. Install SNAPSHOT (And LuCI if desired.)

rezad1393, 2023-06-10:
> MEVBU router owners should either:
> 
>     1. Roll back to latest version of 21.xx.x (Which I did on my WRT1900ACS v1)
> 
>     2. Install SNAPSHOT (And LuCI if desired.)

is this fixed in :
https://github.com/openwrt/openwrt/releases/tag/v23.05.0-rc1

rezad1393, 2023-06-10:
> If I read the [Release Notes for 22.03.4](https://github.com/openwrt/openwrt/releases/tag/v22.03.4) correctly, all users which are affected by this bug and stuck at 22.03.2 have 11 open CVE's on their routers. What is the recommendation for these users regarding security?

use imagebuilder from https://downloads.openwrt.org/releases/22.03.5/targets/mvebu/cortexa9/ and create image for your router.
it downloads compiled packages and creates an image for your router even though official releases for that version are not there.
or maybe use [v23.05.0-rc1](https://github.com/openwrt/openwrt/releases/tag/v23.05.0-rc1)

thecode, 2023-06-10:
I am using [v23.05.0-rc1](https://github.com/openwrt/openwrt/releases/tag/v23.05.0-rc1) and everything works perfect

rezad1393, 2023-06-10:
> I am using [v23.05.0-rc1](https://github.com/openwrt/openwrt/releases/tag/v23.05.0-rc1) and everything works perfect

you mean the issue is fixed? or you just mean in general it works?

thecode, 2023-06-10:
> > I am using [v23.05.0-rc1](https://github.com/openwrt/openwrt/releases/tag/v23.05.0-rc1) and everything works perfect
> 
> you mean the issue is fixed? or you just mean in general it works?

The issue was already reported in the past to be fixed in newer kernels and snapshots, so yes it should be fixed. I had never experienced the issue on a `wrt32x` so I can't say anything about it.

I also assume that if builds are up again for this target it is because the issue is fixed.

rezad1393, 2023-06-10:
> Looks like 6.1 is eventually going to help.
> 
> #11011

I asked about that rc-1 because of this.
it made it seem that issue will be fixed in 6.1 and not 5.15.

zaventh, 2023-06-10:
> > Looks like 6.1 is eventually going to help.
> > #11011
> 
> I asked about that rc-1 because of this. it made it seem that issue will be fixed in 6.1 and not 5.15.

I'm not an official source, but this release is based on kernel `5.15.114`. I can say from personal experience this switch configuration issue is indeed resolved in this kernel.

However, at least one other issue exists where the switch crashes under certain circumstances that appears to have not made it in, which is tracked under #12375.

nmschulte, 2023-06-15:
Don't forget the original/duplicated https://github.com/openwrt/openwrt/issues/10997.

anton000, 2023-07-08:
WRT1900ACS can find the 23.05.0-rc2 img here. Be sure not to flash the snapshot, it doesnt have Luci.

https://firmware-selector.openwrt.org/?version=23.05.0-rc2&target=mvebu%2Fcortexa9&id=linksys_wrt1900acs

catscarlet, 2023-08-10:
I'm still using Linksys WRT32X, which is using mv88e6176 switch.
What should I do for now? Waiting for **23.05 Release** ?

(I'm using WRT32X in home so the security issue is not a issue to me, althrough it's kinda silly when tcpdump/wireshark, packages everywhere)

[8 more replies]