openwrt/openwrt #11101: firewall4: Mandatory source zone prevents creating the proper helper rule for PPTP client

vgaetera, 2022-10-28:
Mandatory source zone prevents creating the proper helper rule for PPTP client:
```bash
# service firewall restart
Section pptp (Allow-PPTP-Client) must specify a source zone for target 'helper'
```

The existing PPTP helper rule:
```bash
# uci show firewall.pptp
firewall.pptp=rule
firewall.pptp.name='Allow-PPTP-Client'
firewall.pptp.dest_port='1723'
firewall.pptp.proto='tcp'
firewall.pptp.dest='wan'
firewall.pptp.set_helper='pptp'
firewall.pptp.target='HELPER'
```

This is supposed to add a helper rule in the output chain for the WAN interface.
But the firewall limitation for mandatory source zone prevents creating the rule.
See the error message listed in the beginning.

Meanwhile, the manually added rule works just fine:
```bash
# nft insert rule inet fw4 output_wan meta nfproto ipv4 tcp dport 1723 ct helper set pptp
# nft list chain inet fw4 output_wan
table inet fw4 {
	chain output_wan {
		meta nfproto ipv4 tcp dport 1723 ct helper set "pptp"
		jump accept_to_wan
	}
}
```

Therefore, the mandatory source zone for firewall helper rules is incorrect.
Please remove this limitation.

The issue affects both OpenWrt SNAPSHOT and OpenWrt  22.03.2.

Reproducing the issue:
```bash
opkg update
opkg install kmod-nf-nathelper-extra
uci add firewall rule
uci set firewall.@rule[-1].name="Allow-PPTP-Client"
uci set firewall.@rule[-1].dest_port="1723"
uci set firewall.@rule[-1].proto="tcp"
uci set firewall.@rule[-1].dest="wan"
uci set firewall.@rule[-1].set_helper="pptp"
uci set firewall.@rule[-1].target="HELPER"
uci commit firewall
/etc/init.d/firewall restart
```

camelcamro, 2023-09-20:
sorry, but when can this changes commited ? now, since 1 year pending ?

vgaetera, 2024-07-17:
I guess this should be re-opened here:
https://github.com/openwrt/firewall4/issues

closed 2024-07-17 by vgaetera